Dunedan/mbp-2016-linux #108: 13,2 to 14,3 Audio Codecs

marc-git, 2019-07-31:
Not much has been completed but there is some activity. Was thinking of adding this somewhere:

A bit of activity happening here:  https://github.com/davidjo/snd_hda_macbookpro

See also: https://bugzilla.kernel.org/show_bug.cgi?id=195671

Audio drivers to 13,2 / 13,3 probably same as 14,3.  It seems this will not be straightforward at all. From bugzilla : 

> The 14,3 from what I can see is the same chips as 13,2 and 13,3.
>
>   Cirrus Logic CS8409     (HDA Codec)
> Cirrus Logic CS42L83A   (Mic+HP ADC/DAC)
>  Maxim MAX98706          (Speaker Amps)

Encouraging is that @davidjo appears to actually hear sound!

davidjo, 2019-08-04:
I was about to send an update to this and upload an improved implementation to davidjo.
I can confirm I am playing sound on speakers on my Macbook Pro 14,3.
(Ive ignored the headphone setup so far).
There is also a putative 14,1 setup (this uses the SSM3515 amps which require different setup
to the MAXs)
It is complicated because essentially the HDA has become pointless - its just transforming the digital PCI audio into a TDM stream for sending to the amps where the real DACs are.
All this setup involves totally undocumented coef indexes of the vendor node.
The I2C setup of the amps is also done by coef indexes of the vendor node (there are multi-1000s of hda verb calls involved.
This was all done from a log of hda verbs sent from OSX.

marc-git, 2019-08-04:
Thanks for writing here. I also have the 14,3. Then I'm probably worth nothing to you for testing a setup.  Good work anyway! This machine isn't far off being fully operational! So you are modifing a replay from OSX and driving the audio with it? Super weird that there is such a difference between windows and mac versions. 
What do you mean with putative 14,1 setup ? according to who, or what does this word mean in this context?

davidjo, 2019-08-05:
I dont have a 14,1 machine but I have been corresponding with someone who does and is on one of the kernel bugs for sound.
This is from a log from his machine - but I cant test the code so dont know if it works.
Im pretty certain it should but Im still identifying the minor changes - there are gpio mask differences and obviously the amp i2c calls are different.

marc-git, 2019-11-09:
Hi there.. Any progress on this?

davidjo, 2019-11-09:
I uploaded a new version around Oct 25th which is much improved. It now handles some headsets with mike plus no mike headsets. I also think it is working on 14,1, 14,2 and 14,3 (which seems to mean some 13 series machines also - basically anything with a 8409 HDA and either MAX98706 or SSM3515 amps plus CS42L83A chip) - at least other users have said this. So I think I have a working kernel module - the questions now are how the interface with the rest of the linux audio chain works - pulse seems to assign some strange labels if at all - which I dont really know anything about. The kernel module only handles the formats Apple defines so its reliant on higher level subsystems  to do the required audio conversions.
It likely still has some timing issues if you do things fast eg plug/unplug headphones.

devalanche, 2019-12-07:
@davidjo thanks a lot for your work on this. after a little hassle with meeuw's fork I got my mbp 14,3 speakers speaking on ubuntu 19.10 kernel 5.3.0.24. first goal achieved.
Unfortunately the built-in mic is not working, same prob with the jack. 
I am pretty newb to kernel or driver coding, so I can't promise any help beyond testing stuff, but are you still on it? Do you see any chance to make your process with headsets work on kernels 5.3.* work as well?

davidjo, 2019-12-07:
Its looks as tho meeuws fork is still on my first attempt and hasnt been updated with the commits I made around Oct 15th. But if meeuws fork works on kernel 5.3 for you its highly likely my updated version will work as well. So far I havent tried switching kernels as most of my machines are on 18.04 LTS - but there will be a new LTS version next Spring I think when I will upgrade which will go to a ver 5 kernel. As to the mike the code that meeuws fork is based didnt attempt to do anything with the CS42L83A chip so not surprising nothings working - my new version for headsets has a much better understanding of that chip but I still didnt attempt any tests of mikes - either internal or external.

devalanche, 2019-12-08:
ok so, let me know if I can test anything on the ver 5 kernel. until then I stick to meeow‘s fork since it is not too long time left to the next LTS release

Meetem, 2020-01-02:
I'am having MBP 14,3 too, but just using 5.3.0-26 kernel is not working with sound (although wifi is working fine after setting TX Power to 10 (there is a guide how to do that). 

Sound is not working on my machine at all. Only bluetooth headset is working. I'am newbie to linux, but can provide some help with testing on real machine too. 
MBP 2017 14,3 Radeon Pro 560
Ubuntu 19.10 5.3.0-26 kernel

Feel free to contact me, as @devalanche said, this computer is actually near to fully working, looks like only sound is missing.

devalanche, 2020-01-02:
> I'am having MBP 14,3 too, but just using 5.3.0-26 kernel is not working with sound (although wifi is working fine after setting TX Power to 10 (there is a guide how to do that).
>
> Sound is not [...]

bro, I got speakers and jack working with this:
https://github.com/leifliddy/snd_hda_macbookpro
mic is still missing though...

Meetem, 2020-01-02:
@devalanche 

Almost not working for me. Just tested and once after switching and clicking test Front Left is worked out with "Front Left" test voice sequence. Since this no music playing and audio test is not working too. Sad. 
I can provide some logs if needed, but I don't know where they are

devalanche, 2020-01-03:
you could try to reinstall the original kernel modules if you had tried to modify some in the past and then apply the patch.
Alternatively you could to try the patch on a fresh installation on another partition/had or smth.
That way you could ensure if the driver is correctly installed and see if it does a better job

davidjo, 2020-01-03:
@Meetem so are you saying you have sound working with the leifliddy snd_hda_macbookpro (which   is based on my updates from October)?
As I understand the 14,3 it has 4 speakers 2 on each side - but I was unable to find any suggestion the 8409 HDA/ MAX amps are doing any signal processing - as the OSX driver has items suggesting some DSP operations are being done what I guess is that in creating the 4 channel audio Apple essentially does some signal processing to boost bass for one of the channels of each pair - for linux you need to look for pulse modules/alsa modules that can apply such processing

Meetem, 2020-01-04:
@davidjo yes, it's working. May be sound quality correlates with the fact that only "Front Left" test sound sequence is working, rear ones and front right doesn't produce any sound. But both left and right speakers are working when sound is played by some applications (of course you can't tweak left-right front-rear balance).
I can try some bass-boosting plugins, but not sure will it work if OS is seeing only Front Left speaker. 
There are also Windows drivers for this sound card (installed by bootcamp), may be there is a way to see how it's working, I can dump them if you need.

What about other sound outputs: I also have an A2DP (on linux, AAC on mac) bluetooth headset there, and sound quality drops over time and sometimes even lagging while on macOS it's working fine. Looks like there is bluetooth tx power problem, cause on mac/android I can listen to music on about 5m distance, and on linux distance is about 1m.

Meetem, 2020-01-04:
@davidjo bass-boosting doesn't helps, while bass is heard from left-right speakers it doesn't "feel" like it should. No vibrations at all, there is subwoofer for sure, and it's not working (also I can test it easily cause I have some coarse dust in it)

marc-git, 2020-01-04:
would be great if we had an apple contact who would just tell us.  Is this plugin 'safe' in the sense of not blowing out the speakers yet?

Meetem, 2020-01-04:
@marc-git since bass-boosting doesn't work right, cause doesn't using subwoofer that doesn't matter. And if you are trying to use it _on linux_ you should know what are you doing, safety is on you now.

MrAsimov, 2020-06-21:
@marc-git did you got the internal audio and microphone working by applying the Cirrus 8409 HDA chip driver patch from https://github.com/davidjo/snd_hda_macbookpro ? If you are running MBP >= 13.1 except for 13.3 (information already provided) can you provide me the dmesg and hwinfo information for that model(s) so I can update PR https://github.com/Dunedan/mbp-2016-linux/pull/149 with related README.md and updated/new dmesg and hwinfo information for those models?

marc-git, 2020-06-22:
thanks for organising. I didn't install since I did not want to blow my speakers. I don't need sound when running linux.

MrAsimov, 2020-06-22:
@marc-git not a problem. Dunno if the volume control is still an issue. I have not seen any.

andyholst, 2020-07-26:
This issue is bit outdated since built-in audio has been verified and is fully  1working for MBP models 13,3 and 14,3.

@marc-git You should rename the title to only include MBP model 13,2 since it has not been verified.

marc-git, 2020-08-06:
will try it out this week on 14,3 and get back to you. Or possibly this issue should be closed now since I don't have a 13,2?

andyholst, 2020-08-07:
@marc-git read the README.md and test the latest commit of the driver patch it is working on Linux kernel version 5.5+, I am running version 5.7.10.

sg0wtham, 2020-08-08:
I am using MBP 13,2 and with the patch (https://github.com/davidjo/snd_hda_macbookpro) I can confirm that audio output through speakers and headphone is working. But no luck with the internal mic. 

I'm using kernel 5.4.0 at the moment.

andyholst, 2020-08-08:
@sg0wtham test it with kernel version 5.5+, before version 5.5 internal audio and microphone is not working. Can be easily installed for Ubuntu/Mint/Debian via deb package manager https://kernel.ubuntu.com/~kernel-ppa/mainline/ (amd 64bit architecture) should be easy to install as well for Fedora/Gentoo. Or if you prefer to compile it from source.

sg0wtham, 2020-09-01:
@andyholst Really sorry for the delayed updates since my Mac was in Apple care. 

As you mentioned I had installed kernel v5.5.19. Now I can confirm that both speakers and mic are working seamlessly. Thank you very much for that.

But there is one thing I have noticed, when I'm using wired headset, internal Mic does not seem to work. But with wireless Bluetooth headphones internal Mic is working perfectly.

andyholst, 2020-09-06:
@sg0wtham I can't speak about the wired headset issue since I have never tried it, but I assume you did make sure the right input was selected (wired microphone instead of the built-in microphone), however you should make a PR and update the README.md file accordingly to update the input and output audio support for MBP 13,2 as working.

sansyrox, 2020-11-12:
By any chance do you all happen to know a fix for Manjaro based OS. The solution by @davidjo works only for debian based platforms :(