Dunedan/mbp-2016-linux #109: Recent wifi issue after kernel upgrade (5.2.5)

mfauvain, 2019-08-01:
After recent upgrade to 5.2.x on macbook 14,1 wifi has issues not seen before and log shows :
'08:18:32 kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4350/5 wl0: Nov 26 2015 03:48:57 version 7.35.180.133 (r602372) FWID 01-c45b39d6'
'08:18:32 kernel: brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4350/5 wl0: Nov 26 2015' '03:48:57 version 7.35.180.133 (r602372) FWID 01-c45b39d6'
'08:18:32 kernel: brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available'
'08:18:32 kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip BCM4350/5'
'08:18:32 kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip BCM4350/5'

Not 100% sure those were not here before but symptoms are : at work wifi connects but DNS is (or seems) broken, wpa_supplicant shows PMKSA-CACHE-ADDED and immediately PMKSA-CACHE-DELETED for all BSSID (corresponding to same SSID).
At home, wifi works ok (multiple BSSID as well)

Tried every possible combination of systemd-resolved, resolvconf etc etc.. can't get it to ping google.com
Fibnally gave up and bought a wifi dongle (realtek) and wifi works perfect, so seems dns resolving was not the issue.

anyone seen the same behaviour after kernel/linux firmware upgrade?

mfauvain, 2019-08-01:
for completeness, smeg boot log shows : 
`[    3.357288] usbcore: registered new interface driver brcmfmac`
`[    3.469905] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip BCM4350/5`
`[    3.471303] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBookPro14,1.txt failed with error -2`
`[    3.471311] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.txt failed with error -2`
`[    3.772694] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4350c2-pcie for chip BCM4350/5`
`[    3.772709] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available`
`[    3.772979] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4350/5 wl0: Nov 26 2015 03:48:57 version 7.35.180.133 (r602372) FWID 01-c45b39d6`
`[    3.799142] brcmfmac 0000:02:00.0 wlp2s0: renamed from wlan0`

mfauvain, 2019-08-12:
tried many kernels (previous) and linux-firmware (however brcmfmac4350c2 is identical) and no luck. Can't figure ou what changed so that now wifi at work doesn't work. Connection to router is made but impossible to ping anything.
Just to lift a doubt I have, anyone else seeing the same log as mine, esp the part saying
`brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBookPro14,1.txt failed with error -2`
Am I missing this .txt file?

hashimda, 2019-08-15:
@mfauvain I tried out an Arch live iso to test out the 5.2.5 kernel on my own Macbook Pro 14,1,  and it doesn't seem to have any issues there with connecting to wifi, neither does the openSUSE Tumbleweed net installer. Though openSUSE Tumbleweed doesn't boot after installation, however that is irrelevant to this thread, but just a word of caution if you decide to install it. My point is that my MBP 14,1 seems to be working just fine with wifi on the 5.2.x kernel, so it may be a different issue.

mfauvain, 2019-08-15:
Thx for feedback, I also have wifi working most of the time, at home etc.. but at work wifi uses WPA-EAP and even though it does connect, I can't get to ping anything. log shows a lot of PMKSA-Cache issues. Not sure if it comes from wifi firmware+kernel upgrade, what I can tell is wifi at work (same WPA-EAP) does connect and ping with a wifi dongle (realtek firmware) and kernel 5.2.x
I am pretty sure the issue came with broadcom brcm firmware upgrade. I will try to downgrade the whole system to arch archive from a month ago and see if I can get wifi working again.

hstaub, 2019-08-15:
Maybe 5.2.8 works better, it contains a wifi related fix. See:

https://www.spinics.net/lists/linux-wireless/msg188475.html
https://lkml.org/lkml/2019/8/8/901
https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.2.8

mfauvain, 2019-08-15:
I tried that, no better unfortunately. Issue appeared around mid july and that's when 5.2's kernels started in arch repos. So will try to revert system to early july and see.

mfauvain, 2019-09-02:
Back at work, after a few more upgrades I tried again to connect to wifi, seems to connect but can't ping. running kernel 5.2.11 now and wpa_supplicant 2:2.9-1
I did not try to revert to older version, I will try to find some time to do so. Not sure if issue comes from brcmfmac driver or wpa_supplicant but log still shows many  PMKSA-CACHE-ADDED and immediately PMKSA-CACHE-DELETED for all BSSID it sees.

Dunedan, 2020-07-22:
@mfauvain Do you have any updates regarding this issue?

mfauvain, 2020-07-22:
No luck unfortunately. Up until March I still had the issue and reverting to earlier kernel or wpa_supplicant version did not help (I did not try all possible combination though).... Home office since then and wifi (that is WPA2, not WPA-EAP) works well (however I mainly use wired connection).

Dunedan, 2020-12-08:
As the problem is gone for now, I'm closing this now. I suspect it's also not related to a specific kernel update, but to the known firmware-related wifi problems (see #47 for example). In case there is any more to discuss regarding this issue, I'd be happy to re-open it of course.

closed 2020-12-08 by Dunedan