Dunedan/mbp-2016-linux #110: MacBook Pro 15+: iBridge Support

aunali1, 2019-08-06:
Discussion and updates regarding iBridge related support. Parent thread is #71.

roadrunner2, 2019-08-09:
If folks could test the touchbar and light-sensor drivers on MacBookPro15,*, that would be very much appreciated. The driver is located at [roadrunner2/macbook12-spi-driver](https://github.com/roadrunner2/macbook12-spi-driver/tree/mbp15) (in the `mbp15` branch). See the README at that link for build and install instructions.

Once installed, could you please test the following:
* Touch Bar:
  * does it turn on and show ESC+brightness+multimedia keys?
  * does pressing and releasing the `fn` key reliably toggle between the above and the function-keys?
  * does the above toggling still work after enabling alt mode?
    ```sh
    echo 1 | sudo tee /sys/bus/hid/devices/*/alt
    ```
  * does dimming work?
    ```sh
    # dim
    echo 0 | sudo tee /sys/bus/hid/devices/*/dim_timeout
    # full brightness again
    echo -2 | sudo tee /sys/bus/hid/devices/*/dim_timeout
    ```
  * does turning off the display work?
    ```sh
    # off
    echo 0 | sudo tee /sys/bus/hid/devices/*/idle_timeout
    # on again
    echo 300 | sudo tee /sys/bus/hid/devices/*/idle_timeout
    ```
  * you can also test automatic dimming/off by setting a short timeout; e.g.
    ```sh
    echo 10 | sudo tee /sys/bus/hid/devices/*/idle_timeout
    ```
    The Touch Bar should then dim after about 7 seconds of inactivity, and then turn off after another 3 seconds; touching the Touch Bar or trackpad or hitting any key should wake it to full brightness again.
* Ambient Light Sensor:
  The light sensor is located at the top of the lid next to the camera. The easiest way to test is by installing the `iio-sensor-proxy` package (should be available on most distros) and then running the `monitor-sensor` command - it should say that there is a light sensor, and changing the brightness (e.g. by covering the sensor with your hand) should result in light-changed events.

  Alternatively, the light values can also be watched with some simple like:
  ```sh
  watch -n 1 cat /sys/bus/iio/devices/iio:device0/in_illuminance_input
  ```

wedgeV, 2019-08-10:
> does it turn on and show ESC+brightness+multimedia keys?

yes

> does pressing and releasing the fn key reliably toggle between the above and the function-keys?

no, just stays ESC+brightness+multimedia

> does the above toggling still work after enabling alt mode?

no, just stays ESC+brightness+multimedia

> does dimming work?

yes

>does turning off the display work?

yes

> you can also test automatic dimming/off by setting a short timeout;

yes

> Ambient Light Sensor

couldn't get iio-sensor-proxy to work, so couldn't check

iggy, 2019-08-10:
Can confirm the same results as above.

In addition, I tested the light sensor using the watch command and could see changes as I covered the sensor.

roadrunner2, 2019-08-10:
Thanks to both of you for testing!

Regarding the fn key issue: I think I found the problem and just pushed an update, so please re-pull. If it still doesn't work, then could you do two things:
* check if there are any errors in `dmesg`
* try manually changing the mode:
  ```sh
  # function keys only
  echo 0 | sudo tee /sys/bus/hid/devices/*/fnmode
  # special keys only
  echo 3 | sudo tee /sys/bus/hid/devices/*/fnmode
  # esc key only
  echo 4 | sudo tee /sys/bus/hid/devices/*/fnmode
  ```

Regarding the ALS: if the watch command works, then basically the driver is working, so that's good; `iio-sensor-proxy` is used by at least Gnome and KDE to get the sensor readings, i.e. if that's working then you should be able to configure things like automatic screen brightness. Also, if the ALS is working (irrespective of whether `iio-sensor-proxy` is installed or not) then the touchbar's brightness should automatically change as the sensor's reading changes.

wedgeV, 2019-08-10:
@roadrunner2 just updated and tested. fn key still doesn't do anything. no errors in dmesg and the manual mode changes seem to work, not always but i'd say >95% of the time they work

roadrunner2, 2019-08-10:
@wedgeV Let's first look at the fn key issue then, as having that working will make testing the mode changes much easier. So somehow I must have have the input-device id matching wrong - could you dump me some info on the input devices on your system?
```sh
for inp in /sys/class/input/input* ; do cat $inp/name; grep . $inp/id/*; cat $inp/modalias; cat $inp/capabilities/key; echo ---; done
```

wedgeV, 2019-08-11:
```Lid Switch
/sys/class/input/input0/id/bustype:0019
/sys/class/input/input0/id/product:0005
/sys/class/input/input0/id/vendor:0000
/sys/class/input/input0/id/version:0000
input:b0019v0000p0005e0000-e0,5,kramlsfw0,
0
---
Power Button
/sys/class/input/input1/id/bustype:0019
/sys/class/input/input1/id/product:0001
/sys/class/input/input1/id/vendor:0000
/sys/class/input/input1/id/version:0000
input:b0019v0000p0001e0000-e0,1,k74,ramlsfw
10000000000000 0
---
HDA Intel PCH HDMI/DP,pcm=10
/sys/class/input/input10/id/bustype:0000
/sys/class/input/input10/id/product:0000
/sys/class/input/input10/id/vendor:0000
/sys/class/input/input10/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
Apple Headset
/sys/class/input/input11/id/bustype:0003
/sys/class/input/input11/id/product:5043
/sys/class/input/input11/id/vendor:05ac
/sys/class/input/input11/id/version:0101
input:b0003v05ACp5043e0101-e0,1,4,k72,73,A4,ram4,lsfw
1000000000 c000000000000 0
---
Apple Inc. Apple Internal Keyboard / Trackpad Keyboard
/sys/class/input/input12/id/bustype:0003
/sys/class/input/input12/id/product:027b
/sys/class/input/input12/id/vendor:05ac
/sys/class/input/input12/id/version:0101
input:b0003v05ACp027Be0101-e0,1,4,11,14,k71,72,73,74,75,77,79,7A,7B,7C,7D,7E,7F,80,81,82,83,84,85,86,87,88,89,8A,8C,8E,96,98,9E,9F,A1,A3,A4,A5,A6,AD,B0,B1,B2,B3,B4,B7,B8,B9,BA,BB,BC,BD,BE,BF,C0,C1,C2,F0,ram4,l0,1,2,3,4,sfw
1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
---
Apple Inc. Apple Internal Keyboard / Trackpad Consumer Control
/sys/class/input/input13/id/bustype:0003
/sys/class/input/input13/id/product:027b
/sys/class/input/input13/id/vendor:05ac
/sys/class/input/input13/id/version:0101
input:b0003v05ACp027Be0101-e0,1,4,kA3,A4,A5,A8,D0,ram4,lsfw
10000 13800000000 0 0
---
Apple Inc. Apple Internal Keyboard / Trackpad Mouse
/sys/class/input/input16/id/bustype:0003
/sys/class/input/input16/id/product:027b
/sys/class/input/input16/id/vendor:05ac
/sys/class/input/input16/id/version:0101
input:b0003v05ACp027Be0101-e0,1,2,4,k110,111,112,r0,1,am4,lsfw
70000 0 0 0 0
---
Apple Inc. iBridge Display
/sys/class/input/input19/id/bustype:0003
/sys/class/input/input19/id/product:8302
/sys/class/input/input19/id/vendor:05ac
/sys/class/input/input19/id/version:0101
input:b0003v05ACp8302e0101-e0,1,4,11,14,k71,72,73,74,75,77,79,7A,7B,7C,7D,7E,7F,80,81,82,83,84,85,86,87,88,89,8A,8C,8E,96,98,9E,9F,A1,A3,A4,A5,A6,AD,B0,B1,B2,B3,B4,B7,B8,B9,BA,BB,BC,BD,BE,BF,C0,C1,C2,F0,ram4,l0,1,2,3,4,sfw
1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
---
Sleep Button
/sys/class/input/input2/id/bustype:0019
/sys/class/input/input2/id/product:0003
/sys/class/input/input2/id/vendor:0000
/sys/class/input/input2/id/version:0000
input:b0019v0000p0003e0000-e0,1,k8E,ramlsfw
4000 0 0
---
Power Button
/sys/class/input/input3/id/bustype:0019
/sys/class/input/input3/id/product:0001
/sys/class/input/input3/id/vendor:0000
/sys/class/input/input3/id/version:0000
input:b0019v0000p0001e0000-e0,1,k74,ramlsfw
10000000000000 0
---
PC Speaker
/sys/class/input/input4/id/bustype:0010
/sys/class/input/input4/id/product:0001
/sys/class/input/input4/id/vendor:001f
/sys/class/input/input4/id/version:0100
input:b0010v001Fp0001e0100-e0,12,kramls1,2,fw
0
---
Video Bus
/sys/class/input/input5/id/bustype:0019
/sys/class/input/input5/id/product:0006
/sys/class/input/input5/id/vendor:0000
/sys/class/input/input5/id/version:0000
input:b0019v0000p0006e0000-e0,1,kE0,E1,E3,F1,F2,F3,F4,F5,ramlsfw
3e000b00000000 0 0 0
---
HDA Intel PCH HDMI/DP,pcm=3
/sys/class/input/input6/id/bustype:0000
/sys/class/input/input6/id/product:0000
/sys/class/input/input6/id/vendor:0000
/sys/class/input/input6/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
HDA Intel PCH HDMI/DP,pcm=7
/sys/class/input/input7/id/bustype:0000
/sys/class/input/input7/id/product:0000
/sys/class/input/input7/id/vendor:0000
/sys/class/input/input7/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
HDA Intel PCH HDMI/DP,pcm=8
/sys/class/input/input8/id/bustype:0000
/sys/class/input/input8/id/product:0000
/sys/class/input/input8/id/vendor:0000
/sys/class/input/input8/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
HDA Intel PCH HDMI/DP,pcm=9
/sys/class/input/input9/id/bustype:0000
/sys/class/input/input9/id/product:0000
/sys/class/input/input9/id/vendor:0000
/sys/class/input/input9/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
```

wedgeV, 2019-08-11:
```Lid Switch
/sys/class/input/input0/id/bustype:0019
/sys/class/input/input0/id/product:0005
/sys/class/input/input0/id/vendor:0000
/sys/class/input/input0/id/version:0000
input:b0019v0000p0005e0000-e0,5,kramlsfw0,
0
---
Power Button
/sys/class/input/input1/id/bustype:0019
/sys/class/input/input1/id/product:0001
/sys/class/input/input1/id/vendor:0000
/sys/class/input/input1/id/version:0000
input:b0019v0000p0001e0000-e0,1,k74,ramlsfw
10000000000000 0
---
HDA Intel PCH HDMI/DP,pcm=10
/sys/class/input/input10/id/bustype:0000
/sys/class/input/input10/id/product:0000
/sys/class/input/input10/id/vendor:0000
/sys/class/input/input10/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
Apple Headset
/sys/class/input/input11/id/bustype:0003
/sys/class/input/input11/id/product:5043
/sys/class/input/input11/id/vendor:05ac
/sys/class/input/input11/id/version:0101
input:b0003v05ACp5043e0101-e0,1,4,k72,73,A4,ram4,lsfw
1000000000 c000000000000 0
---
Apple Inc. Apple Internal Keyboard / Trackpad Keyboard
/sys/class/input/input12/id/bustype:0003
/sys/class/input/input12/id/product:027b
/sys/class/input/input12/id/vendor:05ac
/sys/class/input/input12/id/version:0101
input:b0003v05ACp027Be0101-e0,1,4,11,14,k71,72,73,74,75,77,79,7A,7B,7C,7D,7E,7F,80,81,82,83,84,85,86,87,88,89,8A,8C,8E,96,98,9E,9F,A1,A3,A4,A5,A6,AD,B0,B1,B2,B3,B4,B7,B8,B9,BA,BB,BC,BD,BE,BF,C0,C1,C2,F0,ram4,l0,1,2,3,4,sfw
1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
---
Apple Inc. Apple Internal Keyboard / Trackpad Consumer Control
/sys/class/input/input13/id/bustype:0003
/sys/class/input/input13/id/product:027b
/sys/class/input/input13/id/vendor:05ac
/sys/class/input/input13/id/version:0101
input:b0003v05ACp027Be0101-e0,1,4,kA3,A4,A5,A8,D0,ram4,lsfw
10000 13800000000 0 0
---
Apple Inc. Apple Internal Keyboard / Trackpad Mouse
/sys/class/input/input16/id/bustype:0003
/sys/class/input/input16/id/product:027b
/sys/class/input/input16/id/vendor:05ac
/sys/class/input/input16/id/version:0101
input:b0003v05ACp027Be0101-e0,1,2,4,k110,111,112,r0,1,am4,lsfw
70000 0 0 0 0
---
Sleep Button
/sys/class/input/input2/id/bustype:0019
/sys/class/input/input2/id/product:0003
/sys/class/input/input2/id/vendor:0000
/sys/class/input/input2/id/version:0000
input:b0019v0000p0003e0000-e0,1,k8E,ramlsfw
4000 0 0
---
Apple Inc. iBridge Display
/sys/class/input/input20/id/bustype:0003
/sys/class/input/input20/id/product:8302
/sys/class/input/input20/id/vendor:05ac
/sys/class/input/input20/id/version:0101
input:b0003v05ACp8302e0101-e0,1,4,14,k71,72,73,78,A3,A4,A5,CC,E0,E1,E5,E6,F0,ram4,lsfw
1006300001000 3800000000 10e00000180001f f800000000000002
---
Power Button
/sys/class/input/input3/id/bustype:0019
/sys/class/input/input3/id/product:0001
/sys/class/input/input3/id/vendor:0000
/sys/class/input/input3/id/version:0000
input:b0019v0000p0001e0000-e0,1,k74,ramlsfw
10000000000000 0
---
PC Speaker
/sys/class/input/input4/id/bustype:0010
/sys/class/input/input4/id/product:0001
/sys/class/input/input4/id/vendor:001f
/sys/class/input/input4/id/version:0100
input:b0010v001Fp0001e0100-e0,12,kramls1,2,fw
0
---
Video Bus
/sys/class/input/input5/id/bustype:0019
/sys/class/input/input5/id/product:0006
/sys/class/input/input5/id/vendor:0000
/sys/class/input/input5/id/version:0000
input:b0019v0000p0006e0000-e0,1,kE0,E1,E3,F1,F2,F3,F4,F5,ramlsfw
3e000b00000000 0 0 0
---
HDA Intel PCH HDMI/DP,pcm=3
/sys/class/input/input6/id/bustype:0000
/sys/class/input/input6/id/product:0000
/sys/class/input/input6/id/vendor:0000
/sys/class/input/input6/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
HDA Intel PCH HDMI/DP,pcm=7
/sys/class/input/input7/id/bustype:0000
/sys/class/input/input7/id/product:0000
/sys/class/input/input7/id/vendor:0000
/sys/class/input/input7/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
HDA Intel PCH HDMI/DP,pcm=8
/sys/class/input/input8/id/bustype:0000
/sys/class/input/input8/id/product:0000
/sys/class/input/input8/id/vendor:0000
/sys/class/input/input8/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
HDA Intel PCH HDMI/DP,pcm=9
/sys/class/input/input9/id/bustype:0000
/sys/class/input/input9/id/product:0000
/sys/class/input/input9/id/vendor:0000
/sys/class/input/input9/id/version:0000
input:b0000v0000p0000e0000-e0,5,kramlsfw6,8,
0
---
```

roadrunner2, 2019-08-13:
Thanks, this clarifies things: the keybits are wonky, and that's because the relevant drivers don't know about the new usb id's, and hence aren't being loaded/attached. You can (temporarily) register the new id's as follows:
```sh
sudo modprobe hid-apple
echo 3 05ac 027b 4 | sudo tee /sys/bus/hid/drivers/apple/new_id

sudo modprobe bcm5974
echo 05ac 027b | sudo tee /sys/bus/usb/drivers/bcm5974/new_id
```
The touchbar driver may need to be reloaded after this, not sure.

ppaulweber, 2019-08-16:
@wedgeV and @roadrunner2: I'm using the same patch https://github.com/Dunedan/mbp-2016-linux/issues/71#issuecomment-517208901 on my MacBookAir8,1 (2018 model with T2) and it works very well! If you have not compiled `bcm5974` into the kernel, ensure that you load the `bcm5974` module before the `bce` driver of @MCMrARM, then everything is setup correctly and you do not need to call a drivers `new_id` interface.

ppaulweber, 2019-08-16:
@wedgeV and @roadrunner2: I just fixed the FN issue on my MacBookAir8,1 keyboard by applying the following kernel patch:

<details><summary>Linux 5.1.5 based patch to resolve keyboard FN key issues for <code>MacBookAir8,1</code></summary>
<p>

```diff
diff --git a/drivers/hid/hid-ids.h b/drivers/hid/hid-ids.h
index adce58f24..9e46d7c16 100644
--- a/drivers/hid/hid-ids.h
+++ b/drivers/hid/hid-ids.h
@@ -170,6 +170,7 @@
 #define USB_DEVICE_ID_APPLE_WELLSPRING9_ANSI   0x0272
 #define USB_DEVICE_ID_APPLE_WELLSPRING9_ISO            0x0273
 #define USB_DEVICE_ID_APPLE_WELLSPRING9_JIS            0x0274
+#define USB_DEVICE_ID_APPLE_MBA81              0x027a
 #define USB_DEVICE_ID_APPLE_FOUNTAIN_TP_ONLY   0x030a
 #define USB_DEVICE_ID_APPLE_GEYSER1_TP_ONLY    0x030b
 #define USB_DEVICE_ID_APPLE_IRCONTROL  0x8240
diff --git a/drivers/hid/hid-quirks.c b/drivers/hid/hid-quirks.c
index 77ffba48c..a87bfb174 100644
--- a/drivers/hid/hid-quirks.c
+++ b/drivers/hid/hid-quirks.c
@@ -269,6 +269,7 @@ static const struct hid_device_id hid_have_special_driver[] = {
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ANSI) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ISO) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_JIS) },
+       { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MBA81) },
        { HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_ALU_WIRELESS_2009_ANSI) },
        { HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_ALU_WIRELESS_2009_ISO) },
        { HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_ALU_WIRELESS_2009_JIS) },
@@ -916,6 +917,7 @@ static const struct hid_device_id hid_mouse_ignore_list[] = {
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ANSI) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ISO) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_JIS) },
+       { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MBA81) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_FOUNTAIN_TP_ONLY) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_GEYSER1_TP_ONLY) },
        { }
```

</p>
</details>

To apply this to an `MacBookPro15,x` please modify or add the `027b` device. I've prepared another patch for you MBP guys, please let me know if it works out for you:

<details><summary>Linux 5.1.5 based patch to resolve keyboard FN issues for <code>MacBookPro15,x</code></summary>
<p>

```diff
diff --git a/drivers/hid/hid-ids.h b/drivers/hid/hid-ids.h
index adce58f24..9e46d7c16 100644
--- a/drivers/hid/hid-ids.h
+++ b/drivers/hid/hid-ids.h
@@ -170,6 +170,7 @@
 #define USB_DEVICE_ID_APPLE_WELLSPRING9_ANSI   0x0272
 #define USB_DEVICE_ID_APPLE_WELLSPRING9_ISO            0x0273
 #define USB_DEVICE_ID_APPLE_WELLSPRING9_JIS            0x0274
+#define USB_DEVICE_ID_APPLE_MBP15              0x027b
 #define USB_DEVICE_ID_APPLE_FOUNTAIN_TP_ONLY   0x030a
 #define USB_DEVICE_ID_APPLE_GEYSER1_TP_ONLY    0x030b
 #define USB_DEVICE_ID_APPLE_IRCONTROL  0x8240
diff --git a/drivers/hid/hid-quirks.c b/drivers/hid/hid-quirks.c
index 77ffba48c..a87bfb174 100644
--- a/drivers/hid/hid-quirks.c
+++ b/drivers/hid/hid-quirks.c
@@ -269,6 +269,7 @@ static const struct hid_device_id hid_have_special_driver[] = {
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ANSI) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ISO) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_JIS) },
+       { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MBP15) },
        { HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_ALU_WIRELESS_2009_ANSI) },
        { HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_ALU_WIRELESS_2009_ISO) },
        { HID_BLUETOOTH_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_ALU_WIRELESS_2009_JIS) },
@@ -916,6 +917,7 @@ static const struct hid_device_id hid_mouse_ignore_list[] = {
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ANSI) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_ISO) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_WELLSPRING9_JIS) },
+       { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_MBP15) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_FOUNTAIN_TP_ONLY) },
        { HID_USB_DEVICE(USB_VENDOR_ID_APPLE, USB_DEVICE_ID_APPLE_GEYSER1_TP_ONLY) },
        { }
```

</p>
</details>

Another remark to my previous comment: Due to this change I've updated my (arch) default loaded modules chain like this:

```bash
$ cat /etc/modules-load.d/bce.conf 
# Load bce.ko at boot
hid-apple
bcm5974
bce
```

And everything regarding `bce`-based keyboard and touchpad works just perfectly!

wedgeV, 2019-08-16:
@ppaulweber cool, i'll try that! do you have two finger scrolling etc. working with the touchpad?

ppaulweber, 2019-08-16:
@wedgeV you're welcome

> do you have two finger scrolling etc. working with the touchpad?

Yes, everything works fine! You just really have to ensure that `bce` is loaded after the `bcm5974` driver which is the Apple touchpad driver, and when than the `bce` registers the new discovered USB devices, it assigns correctly the built-in touchpad with the `hid-apple` driver instead to the `hid-generic` which gives you the single click behavior.

wedgeV, 2019-08-16:
@ppaulweber can you share your linux and linux-headers packages (or a git repo that i can build it from)? i'm also on arch, currently using the packages from the repo someone shared on the main issue

ppaulweber, 2019-08-17:
@wedgeV OK, I'll prepare a GitHub repo containing a small config and build script, which applies the patches provided by @MCMrARM et al. and I'll try to merge the support for both platforms, namely `MacBookAir8,1` and `MacBookPro15,x`.

ppaulweber, 2019-08-17:
@wedgeV here you go https://github.com/ppaulweber/linux-mba, have fun and let me know if it works for you now

jhohisel, 2019-08-17:
Hi everyone! I've been following this thread as I work on getting a USB SSD with linux that works on both of my Macs: MacBookAir8,2 (2019 MacBook Air) and iMac19,1 (2019 iMac). Fortunately the 2019 iMac does _not_ contain a T2 chip, so things work fairly well out of the box.

I'm attempting to get WiFi working on both models. The iMac seems to have the same chipset as the MacBookPro15,x (BCM4364, codename Midway) but the MacBook Air has a BCM4355 chipset (codename Hawaii). I created and applied this patch (based on the work from @MCMrARM): https://github.com/jhohisel/macbookair8-2-linux/blob/master/brcmfmac.patch and copied the firmware from macOS.

However, the firmware fails to load (iMac):
```
[   20.314189] usbcore: registered new interface driver brcmfmac
[   20.330616] brcmfmac 0000:03:00.0: enabling device (0000 -> 0002)
[   20.439205] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[   23.374627] brcmfmac 0000:03:00.0: brcmf_pcie_download_fw_nvram: FW failed to initialize
```

MacBook Air fails with a slightly different error:
```
[   27.958248] brcmfmac 0000:02:00.0: enabling device (0000 -> 0002)
[   28.121706] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4355-pcie for chip BCM4355/12
[   28.126287] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355-pcie.Apple Inc.-MacBookAir8,2.txt failed with error -2
[   31.030139] brcmfmac 0000:02:00.0: brcmf_pcie_download_fw_nvram: FW failed to initialize
```

Top tip: you can discover the firmware filenames that Apple uses by running:
`ioreg -l | grep RequestedFiles`
The output will look something like:
```
"RequestedFiles" = ({"Firmware"="C-4364__s-B2/midway.trx","TxCap"="C-4364__s-B2/midway-X0.txcb","Regulatory"="C-4364__s-B2/midway-X0.clmb","NVRAM"="C-4364__s-B2/P-midway-X0_M-HRPN_V-u__m-7.5.txt"})
```

wedgeV, 2019-08-17:
@ppaulweber thanks so much! works, i now have scrolling/right click working!

ppaulweber, 2019-08-17:
@wedgeV perfect! I'm just putting together all the other guys (@MCMrARM et al.) out there have discovered in there heavy lifting. At this point a really bit thanks and nice work to you all :tada: :1st_place_medal:

GoncaloHit, 2019-08-19:
I would like to test and help, but i have some questions, i have the MBP13.3 2019 i7 Touch bar, currently on windows, i wiped out osx, had to inject drivers in win boot setup, as for linux, does the latest kernel have the SSD drivers ? or do i need to build custom kernel with it, then flash it over an USB ?
Thanks

mikeeq, 2019-08-21:
Is keyboard backlight working for anybody?

MCMrARM, 2019-08-26:
Good news, I got suspend working (sometimes, it's not reliable) and am working on pushing the code onto GitHub.

As for reliability, the whole OS sometimes just doesn't wake up, makes it really hard to debug.

MCMrARM, 2019-08-26:
Also it might be worthwhile to try to reverse engineer some of the XPC comms, so that one could try to take over the shitty fan control as I think it's the powerchime service managing it.

MCMrARM, 2019-08-26:
I have pushed all the suspend related code for VHCI under the suspend branch: https://github.com/MCMrARM/mbp2018-bridge-drv/tree/suspend
Testing welcome and appreciated.
Audio is currently unsupported on that branch.

mikeeq, 2019-08-28:
@roadrunner2 any clue how to make keyboard backlight working? Due to lack of apple smc driver keyboard backlight device is not initialized, should it be done by your spi driver or in @MCMrARM bce one?

```
➜ ls -l /sys/class/leds                  
total 0
lrwxrwxrwx. 1 root root 0 Aug 28 11:11 input15::capslock -> ../../devices/pci0000:00/0000:00:1d.4/0000:02:00.1/bce/bce/bce-vhci/usb7/7-5/7-5:1.1/0003:05AC:027B.0004/input/input15/input15::capslock
lrwxrwxrwx. 1 root root 0 Aug 28 11:11 input15::compose -> ../../devices/pci0000:00/0000:00:1d.4/0000:02:00.1/bce/bce/bce-vhci/usb7/7-5/7-5:1.1/0003:05AC:027B.0004/input/input15/input15::compose
lrwxrwxrwx. 1 root root 0 Aug 28 11:11 input15::kana -> ../../devices/pci0000:00/0000:00:1d.4/0000:02:00.1/bce/bce/bce-vhci/usb7/7-5/7-5:1.1/0003:05AC:027B.0004/input/input15/input15::kana
lrwxrwxrwx. 1 root root 0 Aug 28 11:11 input15::numlock -> ../../devices/pci0000:00/0000:00:1d.4/0000:02:00.1/bce/bce/bce-vhci/usb7/7-5/7-5:1.1/0003:05AC:027B.0004/input/input15/input15::numlock
lrwxrwxrwx. 1 root root 0 Aug 28 11:11 input15::scrolllock -> ../../devices/pci0000:00/0000:00:1d.4/0000:02:00.1/bce/bce/bce-vhci/usb7/7-5/7-5:1.1/0003:05AC:027B.0004/input/input15/input15::scrolllock
```

MCMrARM, 2019-08-28:
well my driver is not providing any other functionality that bringing the USB hub up and providing audio support, so it definitely should not be handled by mine

MCMrARM, 2019-08-28:
HID reports for setting up the backlight: https://mrarm.io/u/Paste%202019-08-28%2013-51-35.txt

mikeeq, 2019-08-29:
@roadrunner2 I've tried to test your latest changes:
https://github.com/roadrunner2/macbook12-spi-driver/tree/d05467440f16d17c2572c5fe3ca59f126758f00f

but apple-ib-tb.ko module is crashing during load time:

```
[   91.756773] bce_vhci_urb_dequeue 81
[   91.764955] input: Apple Inc. iBridge Display as /devices/pci0000:00/0000:00:1d.4/0000:02:00.1/bce/bce/bce-vhci/usb7/7-6/7-6:1.0/0003:05AC:8302.0008/input/input16
[   91.816868] apple-ib-touchbar 0003:05AC:8302.0008: input: USB HID v1.01 Keyboard [Apple Inc. iBridge Display] on usb-bce-vhci-6/input0
[   91.817082] apple-ib-touchbar 0003:05AC:8102.0009: : USB HID v1.01 Device [Apple Inc. iBridge DFR brightness] on usb-bce-vhci-7/input0
[   91.869650] BUG: unable to handle kernel NULL pointer dereference at 0000000000000000
[   91.869653] #PF error: [normal kernel read fault]
[   91.869654] PGD 0 P4D 0 
[   91.869656] Oops: 0000 [#1] SMP PTI
[   91.869658] CPU: 4 PID: 9884 Comm: insmod Tainted: G           OE     5.1.19-300.wifi.patch.fc30.x86_64 #1
[   91.869659] Hardware name: Apple Inc. MacBookPro15,2/Mac-827FB448E656EC26, BIOS 220.270.99.0.0 (iBridge: 16.16.6568.0.0,0) 06/25/2019
[   91.869662] RIP: 0010:appletb_match_internal_device+0x33/0x70 [apple_ib_tb]
[   91.869663] Code: 1c 48 8d 86 18 02 00 00 41 b8 01 00 00 00 75 29 eb 2f 48 8b 50 58 48 85 d2 74 1b 48 8b 32 48 c7 c7 34 75 dc c0 b9 0b 00 00 00 <f3> a6 0f 97 c2 80 da 00 84 d2 74 0f 48 8b 00 48 85 c0 75 d4 45 31
[   91.869664] RSP: 0018:ffffa8d18290ba50 EFLAGS: 00010282
[   91.869665] RAX: ffff999793e67a18 RBX: ffff99977fd98e58 RCX: 000000000000000b
[   91.869666] RDX: ffffffffb0ecb080 RSI: 0000000000000000 RDI: ffffffffc0dc7534
[   91.869666] RBP: ffff999793e67800 R08: 0000000000000001 R09: 0000000000000002
[   91.869667] R10: 0000000000000000 R11: 0000000000000001 R12: ffffffffc0dc78d0
[   91.869667] R13: ffff99977fd98ec0 R14: ffff99977fd98e58 R15: ffffffffc0dc9040
[   91.869668] FS:  00007fee958bb2c0(0000) GS:ffff99979db00000(0000) knlGS:0000000000000000
[   91.869669] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   91.869669] CR2: 0000000000000000 CR3: 00000004345c0004 CR4: 00000000003606e0
[   91.869670] Call Trace:
[   91.869673]  ? input_attach_handler+0x38/0xa0
[   91.869675]  ? input_register_handler+0x8f/0xe0
[   91.869676]  ? appletb_probe+0x23f/0x3c0 [apple_ib_tb]
[   91.869678]  ? hid_device_probe+0x109/0x170
[   91.869681]  ? really_probe+0xf9/0x3a0
[   91.869682]  ? driver_probe_device+0xb6/0x100
[   91.869683]  ? driver_allows_async_probing+0x50/0x50
[   91.869684]  ? bus_for_each_drv+0x7b/0xc0
[   91.869686]  ? __device_attach+0xe4/0x160
[   91.869687]  ? __hid_register_driver+0x80/0x80
[   91.869688]  ? bus_rescan_devices_helper+0x3a/0x80
[   91.869689]  ? device_reprobe+0x18/0x30
[   91.869690]  ? bus_for_each_dev+0x78/0xc0
[   91.869691]  ? hid_destroy_device+0x60/0x60
[   91.869692]  ? __hid_bus_driver_added+0x28/0x30
[   91.869693]  ? bus_for_each_drv+0x7b/0xc0
[   91.869694]  ? __hid_register_driver+0x70/0x80
[   91.869695]  ? appletb_init+0xa8/0x1000 [apple_ib_tb]
[   91.869697]  ? 0xffffffffc0dcc000
[   91.869698]  ? do_one_initcall+0x46/0x1c4
[   91.869700]  ? free_unref_page_commit+0x95/0x110
[   91.869702]  ? _cond_resched+0x15/0x30
[   91.869703]  ? kmem_cache_alloc_trace+0x154/0x1c0
[   91.869705]  ? do_init_module+0x23/0x210
[   91.869706]  ? do_init_module+0x5c/0x210
[   91.869706]  ? load_module+0x23de/0x2910
[   91.869709]  ? ima_post_read_file+0xf0/0x100
[   91.869710]  ? __do_sys_finit_module+0xaa/0x110
[   91.869711]  ? __do_sys_finit_module+0xaa/0x110
[   91.869713]  ? do_syscall_64+0x5b/0x170
[   91.869714]  ? entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   91.869715] Modules linked in: apple_ib_tb(OE+) apple_ibridge(OE) nls_utf8 isofs rfcomm nf_conntrack_netlink xt_addrtype br_netfilter xt_CHECKSUM ipt_MASQUERADE tun bridge stp llc nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 xt_conntrack ebtable_nat ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat iptable_mangle iptable_raw iptable_security nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c ip_set nfnetlink overlay ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter ccm des_generic arc4 bnep cmac sha512_ssse3 sha512_generic md4 sunrpc vfat fat uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev iTCO_wdt mei_hdcp iTCO_vendor_support joydev media snd_soc_skl intel_rapl x86_pkg_temp_thermal intel_powerclamp snd_soc_hdac_hda snd_hda_ext_core snd_soc_skl_ipc coretemp snd_soc_sst_ipc snd_soc_sst_dsp kvm_intel snd_soc_acpi_intel_match
[   91.869733]  snd_soc_acpi snd_soc_core kvm brcmfmac snd_compress snd_hda_codec_hdmi ac97_bus snd_pcm_dmaengine brcmutil irqbypass intel_cstate snd_hda_intel intel_uncore cfg80211 snd_hda_codec intel_rapl_perf snd_hda_core idma64 hci_uart snd_hwdep i2c_i801 btqca btrtl intel_lpss_pci btbcm intel_lpss thunderbolt btintel bluetooth mmc_core mei_me sbs mei ecdh_generic intel_pch_thermal rfkill sbshc acpi_tad pcc_cpufreq apple_bl ip_tables dm_crypt cdc_mbim cdc_wdm nvme cdc_ncm usbnet nvme_core mii i915 uas usb_storage i2c_algo_bit drm_kms_helper crct10dif_pclmul crc32_pclmul crc32c_intel drm ghash_clmulni_intel video pkcs8_key_parser fuse bce(OE) snd_pcm snd_seq snd_timer snd_seq_device snd soundcore bcm5974 hid_apple
[   91.869750] CR2: 0000000000000000
[   91.869751] ---[ end trace a79c0a5ec2a52066 ]---
[... cut ...]
[   91.869756] RDX: ffffffffb0ecb080 RSI: 0000000000000000 RDI: ffffffffc0dc7534
[   91.869756] RBP: ffff999793e67800 R08: 0000000000000001 R09: 0000000000000002
[   91.869757] R10: 0000000000000000 R11: 0000000000000001 R12: ffffffffc0dc78d0
[   91.869757] R13: ffff99977fd98ec0 R14: ffff99977fd98e58 R15: ffffffffc0dc9040
[   91.869758] FS:  00007fee958bb2c0(0000) GS:ffff99979db00000(0000) knlGS:0000000000000000
[   91.869759] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   91.869759] CR2: 0000000000000000 CR3: 00000004345c0004 CR4: 00000000003606e0
```

roadrunner2, 2019-08-29:
@mikeeq Thanks for the report - just pushed a fix.

Btw., what model do you have? I'm trying to understand why it crashed for you but not for others.

mikeeq, 2019-08-29:
@roadrunner2 - Macbook Pro 15,2 13" 2019 i5 TouchBar Z0WQ000AR

I can confirm that patched version is working ;) Will you have a minute to implement keyboard backlight to your ibridge driver?

MCMrARM, 2019-08-29:
I'm working on patching the hid_apple driver to support backlight

MCMrARM, 2019-08-29:
https://github.com/MCMrARM/mbp2018-etc/tree/master/apple-hid

mikeeq, 2019-08-29:
@MCMrARM are you able to change keyboard backlight brightness by pressing buttons on touchbar?

but i can confirm that it's working for me - thanks Man. Good job, once again! 
```
➜ ls -l /sys/class/leds/apple::kbd_backlight/
total 0
-rw-r--r--. 1 root root 4096 Aug 29 17:28 brightness
lrwxrwxrwx. 1 root root    0 Aug 29 17:28 device -> ../../../0003:05AC:027B.0004
-r--r--r--. 1 root root 4096 Aug 29 17:28 max_brightness
drwxr-xr-x. 2 root root    0 Aug 29 17:28 power
lrwxrwxrwx. 1 root root    0 Aug 29 17:28 subsystem -> ../../../../../../../../../../../../../class/leds
-rw-r--r--. 1 root root 4096 Aug 29 17:28 trigger
-rw-r--r--. 1 root root 4096 Aug 29 17:26 uevent
```

One thing @roadrunner2 with the latest version of `apple-ib-tb.ko` fn keys are not showing up when pressing FN key (switching from media keys to fn is not working on default fnmode = 1). It's working ok on module built from this commit - https://github.com/roadrunner2/macbook12-spi-driver/tree/3f01cdb5035c4bc6aceeb2e835ded30699b3c06e

MCMrARM, 2019-08-29:
@mikeeq no, I don't load the touchbar drivers during testing.

But are they supposed to just work? I thought it's supposed to be handled by userspace.

aunali1, 2019-08-30:
@mikeeq Does the following command return anything?
```
for inp in /sys/class/input/input* ; do cat $inp/name; grep . $inp/id/*; cat $inp/modalias; echo ---; done | grep Intel
```

mikeeq, 2019-08-30:
@aunali1 Yes it does.

```
➜ for inp in /sys/class/input/input* ; do cat $inp/name; grep . $inp/id/*; cat $inp/modalias; echo ---; done | grep "Internal Keyboard"

Apple Inc. Apple Internal Keyboard / Trackpad
```

roadrunner2, 2019-08-30:
@MCMrARM nice work on the keyboard backlight driver! Sorry, didn't have time to look at it myself. Looking over the report info you provided, I noticed that the descriptor of the report to set the brightness is not present in the list of report descriptors for that interface, only the descriptor for the brightness config :frowning_face: . 

A couple small things: the keyboard backlight doesn't turn unless the value is at least 32 (see backlight-min-on in the config) - in the applespi driver I therefore took the approach of scaling the given brightness values appropriately so that it turns on at value 1 (i.e. set-value = input-value / (max - min) * max). Something you might want to consider doing too. Also, I noticed that the max-brightness is 512, but in the kernel LED_FULL is defined as 255 - have you tested whether whether things like Gnome will actually scale their range between 0 to max, or assume max is always (at most) 255 (this can be tested by setting the keyboard brightness to max via the appropriate touchbar keys, and then looking at the actual brightness value in sysfs)? I.e. I'm wondering if it's necessary to set `cdev.max_brightness` to LED_FULL instead and scale the input values up to 0 - 512.

> @mikeeq no, I don't load the touchbar drivers during testing.
> 
> But are they supposed to just work? I thought it's supposed to be handled by userspace.

The keyboard brightness keys are indeed just supposed to work; however, due to a [bug in upower and gsd](https://gitlab.freedesktop.org/upower/upower/issues/6) you usually have to restart upower (`sudo systemctl restart upower`) after (re)loading the driver providing the keyboard brightness controls (`hid-apple` in your case, `applespi` in models with T1 chip).

roadrunner2, 2019-08-31:
@mikeeq 
> One thing @roadrunner2 with the latest version of `apple-ib-tb.ko` fn keys are not showing up when pressing FN key (switching from media keys to fn is not working on default fnmode = 1). It's [...]

Hmm, that's odd. Could you run a slightly modified version of what you ran at @aunali1 request above?
```sh
for inp in /sys/class/input/input* ; do cat $inp/name; grep . $inp/id/*; cat $inp/capabilities/key; echo ---; done | grep -A5 "Internal Keyboard"
```
I want to doubly confirm the FN key is being properly "exported" by the keyboard driver.

Next, please apply the following patch to your `apple-ib-tb.c`:
```diff
--- a/apple-ib-tb.c
+++ b/apple-ib-tb.c
@@ -1130,6 +1130,10 @@ static bool appletb_match_internal_device(struct input_handler *handler,
                       strcmp(dev->type->name, "usb_device")))
                dev = dev->parent;
 
+       pr_info("match-internal-dev: bustype=%x vendor=%x product=%x dev=%p usb-prod='%s'\n",
+               inp_dev->id.bustype, inp_dev->id.vendor, inp_dev->id.product,
+               dev, dev ? to_usb_device(dev)->product : "-");
+
        /*
         * Apple labels all their internal keyboards and trackpads as such,
         * instead of maintaining an ever expanding list of product-id's we
```
And then show me the output of `dmesg | grep match-internal` after reloading the module. TIA.

mikeeq, 2019-08-31:
@roadrunner2 

```
➜ for inp in /sys/class/input/input* ; do cat $inp/name; grep . $inp/id/*; cat $inp/capabilities/key; echo ---; done | grep -A5 "Internal Keyboard"

Apple Inc. Apple Internal Keyboard / Trackpad
/sys/class/input/input7/id/bustype:0003
/sys/class/input/input7/id/product:027b
/sys/class/input/input7/id/vendor:05ac
/sys/class/input/input7/id/version:0101
10000 0 0 0 1007b00011007 ff9f217ac14057ff ffbeffdfffefffff fffffffffffffffe

➜ dmesg | grep match-internal
[ 2902.630385] match-internal-dev: bustype=3 vendor=5ac product=27b dev=00000000528c5ebf usb-prod='Apple Internal Keyboard / Trackpad'
[ 2902.630393] match-internal-dev: bustype=3 vendor=5ac product=27b dev=00000000528c5ebf usb-prod='Apple Internal Keyboard / Trackpad'
```

roadrunner2, 2019-08-31:
@mikeeq Thanks. Aaargh: I managed to get the string-search test inverted! :man_facepalming:  Just pushed the fix.

mikeeq, 2019-08-31:
Yep, it's working fine now - GJ @roadrunner2 ! Cheers ;)

MCMrARM, 2019-09-01:
@roadrunner2 ah right, I didn't think of that. As for max_brightness it afaik should be fine, but I could verify it a bit later. Before upstreaming this patch I'll probably also want to get rid of the hardcoded constants in some places.

ppaulweber, 2019-09-06:
@MCMrARM thanks for the keyboard backlight patch, I was able to integrate this one along with the changes for the `MacBookAir8,x` and it just works, well done :tada:

MCMrARM, 2019-09-08:
I have been working on some more of the Touch ID integration stuff....
the protocol is absolutely insane, and gigantic:
I've decided to list all the commands I can find and try to name them (the biometrickitd daemon handling this is written in Objective C... free function names) 
 https://github.com/MCMrARM/bkitd/blob/master/BkKernelCommand.h#L10  this is 61 commands....
Anyways, some basic stuff work (like loading the calibration info), however trying to do enroll fails with `-3`, whatever that means honestly.

imbushuo, 2019-09-09:
@MCMrARM I think SEP (the other PCIe device) is absolutely involved in any enrollment/verification according to my rough inspection in IDA.

MCMrARM, 2019-09-09:
@imbushuo who knows, maybe it is maybe it's not. @aunali1 claimed most of the SEP code is NOP'ed.

Anyways, I have got a bit futher, generally everything not requiring a credential set works, however enroll requires one and I'm damned.

MCMrARM, 2019-09-09:
As for credential set I have yet to figure out how it is obtained...
The `TouchID` binary (sorry, I don't know what directory I got it from right now) calls -[LAContext evaluatePolicy:options:reply:] with policy=APolicyTouchIDEnrollment. Then the credential is taken from -[LAContext externalizedContext]. Then next, I think the `ModuleACM` from LocalAuthentication is used, which turns all requests onto the `AppleCredentialManager` kext. As for what happen later, I have to figure it out.

The credentials are 16 bytes and change sometimes (sometimes are reused), unsure if it changes according to time or according to system reboots.

imbushuo, 2019-09-09:
Regarding Touch Bar, I have an implementation for Windows with media/volume control, "now playing" display, lock computer button and a few more thing (mostly Fn keys). It is possible to just use libusb for the usermode presentation component, which is not possible on Windows due to the design of the Apple VHCI Windows driver.

- Kernel driver: https://github.com/imbushuo/DFRDisplayKm
- Usermode content host: https://github.com/imbushuo/DFRContentHost

Usermode content host is implemented in .NET Core and Avalonia so it is possible to make it cross-platform (just remove some WinRT references and replace the IOCTL part with libusb transfers). C# haters can also implement it in other UI Framework and languages using these examples provided above (but that's out of my scope).

aunali1, 2019-09-09:
@imbushuo I have a functioning DFR display daemon for linux that works using libusb and DBus for communicating system events (Power states, lock screen, etc.) The framework for drawing UI elements is based on Qt for now, not sure if thats the best option. Will be open sourcing it soon.

MCMrARM, 2019-09-10:
I imagine if possible Gtk would be preferred, but I'm not sure if it has a flexible enough layout engine, so I didn't start working on anything. Qt is also ok, though Qt Quick I imagine?

Dunedan, 2020-07-22:
Is my understanding correct that the TouchBar of the 2018+ models is supported fine by https://github.com/roadrunner2/macbook12-spi-driver/ now and the backlight integration for these models got implemented in https://github.com/MCMrARM/mbp2018-etc/tree/master/apple-hid and is working fine as well?

If yes, it'd be worth to point to the patched `hid-apple` driver in the README.

@MCMrARM: It'd be also great if you could get your changes upstreamed, so all users can benefit from them out-of-the-box.

bmd7297, 2021-02-14:
> @imbushuo I have a functioning DFR display daemon for linux that works using libusb and DBus for communicating system events (Power states, lock screen, etc.) The framework for drawing UI elements is based on Qt for now, not sure if thats the best option. Will be open sourcing it soon.

Any progress on open sourcing this?

zfortier, 2021-05-09:
In case anyone finds this thread because they have installed the iBridge driver and got the touchbar working, but the `fn` key doesn't work as expected to switch brightness/multimedia control mode and function-key mode (F1-F12), you can control the way `fn` key work with a udev rule.

As root, create the file `/etc/udev/rules.d/99-apple_ibridge.rules` and add something similar to this:

```
ACTION=="add|change|move"           \
        , ATTRS{idProduct}=="8302"  \
        , ATTRS{idVendor}=="05ac"   \
        , ATTR{fnmode}="1"
```

Make sure the values for `idProduct` and `idVendor` match your hardware setup. Here is a 1-liner to do the whole thing:

```
sudo bash -c  'lsusb | grep "Touch Bar Display" | cut -d" " -f6 | sed '"'"'s/\(.*\):\(.*\)/ACTION=="add|change|move",ATTRS{idProduct}=="\2",ATTRS{idVendor}=="\1",ATTR{fnmode}="1"/'"'"' > /etc/udev/rules.d/99-apple_ibridge.rules'
```

You can change the value of `fnmode` to modify the behavior of the `fn` key:
 - 0 --> function-keys only
 - 1 --> fn key switches from special to function-keys
 - 2 --> inverse of 1
 - 3 --> special keys only
 - 4 --> escape key only

Finally, you can test different settings:
```
echo "<fn mode value to test>" | sudo tee /sys/bus/hid/devices/*/fnmode
```

pklement77, 2021-07-13:
The touchbar sometimes won't show up after reboot, sometimes it does; sometimes it won't come up after wake up from sleep. Once even the keyboard did not work after wake up (had to hard reset). Macbook 15,2 and arch linux. Thanks for your efforts!