Dunedan/mbp-2016-linux #111: MacBook Pro 15+: Audio Support

aunali1, 2019-08-06:
Discussion and updates regarding Broadcom Wifi related support. Parent thread is #71.

MCMrARM, 2019-08-19:
Small note: I fixed the start audio glitch, audio is mostly fine now.

As I stated before, the config has to be written if we want both speaker and headphones to work, and I have no 20 hours to waste. If anyone else is willing to spend the time trying to understand that format, I'd gladly appreciate it.

MCMrARM, 2019-08-19:
Ok I wrote a config:
```
<confdir:pcm/front.conf>
<confdir:pcm/surround40.conf>

AppleT2.pcm.surround40.0 {
	@args [ CARD ]
	@args.CARD {
		type string
	}
	type hw
	card $CARD
	device 1
}
AppleT2.pcm.front.0 {
	@args [ CARD ]
	@args.CARD {
		type string
	}
	type hw
	card $CARD
	device 0
}

```
Put this in /usr/share/alsa/cards/AppleT2.conf
No idea if this can be written in a better way or not, for now this is what we're stuck with.

mikeeq, 2019-08-19:
I can confirm that with ALSA config file mentioned above there are now two configurations in sound settings: 
- Analog Surround 4.0 Output - speakers
- Analog Stereo Output - headphone jack
and one output device. 
Microphone is still not available and audio output configuration are not dynamically changed (dis/connecting headphones). Nevertheless really good job @MCMrARM! 

In the mean time could you please check wifi thread - #112 

I'm pretty sure noone (besides you) wasn't able to make wifi working. Maybe we are missing something. Cheers everyone!

mikeeq, 2019-08-28:
@MCMrARM to make internal mic working do we need to only modify existing ALSA config file or does it need any work on driver side to be done?

MCMrARM, 2019-08-28:
Work on the driver needs to be done.

MCMrARM, 2019-08-28:
Implemented jack detection: https://github.com/MCMrARM/mbp2018-bridge-drv/commit/d3224b861177ed863c492833d17e09c02ed1d53f
No idea how to integrate into alsa ¯\_(ツ)_/¯

MCMrARM, 2019-08-28:
I'll try to get the microphone working tomorrow, no promises though.

MCMrARM, 2019-08-29:
Some good news: I got internal mic working at all, will push code today. Not sure how good are the results. The microphone is apparently a triple-channel device btw. Might be a little bit of a mess to get it working in alsa.

MCMrARM, 2019-08-30:
Ok, this took a while to figure out but I got the mic/external audio input working under Pulse.
https://gist.github.com/MCMrARM/c357291e4e5c18894bea10665dcebffb <- required files
Note that the new commits broke the old audio config, and you MUST switch to the new configuration if you update. Though the mic works now, so nice.

mikeeq, 2019-08-30:
@MCMrARM are there any additional steps needed? I recompiled module, loaded it, copied those 3 config files and there's no audio/input device now - only `dummy output`

MCMrARM, 2019-09-01:
@mikeeq I mean, I don't know, it works for me. Does changing the paths-output to paths-input work for you?

mikeeq, 2019-09-03:
@aunali1 @ppaulweber @jhohisel Has anyone of you guys tried to use new audio driver and is microphone working for you? For me, after changing to `paths-input` microphones are visible in sound settings but they are not working.

jbg, 2020-04-08:
With the driver from latest git and the config files linked above, launching pulseaudio fails on my MBP15,2 with:

```
Apr 08 19:00:58 little kernel: aaudio_pcm_open
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_params
Apr 08 19:00:58 little kernel: aaudio_pcm_open
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_params
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_free
Apr 08 19:00:58 little kernel: aaudio_pcm_close
Apr 08 19:00:58 little kernel: aaudio_pcm_open
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_params
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_free
Apr 08 19:00:58 little kernel: aaudio_pcm_close
Apr 08 19:00:58 little kernel: aaudio_pcm_open
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_params
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_free
Apr 08 19:00:58 little kernel: aaudio_pcm_close
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_free
Apr 08 19:00:58 little kernel: aaudio_pcm_close
Apr 08 19:00:58 little kernel: aaudio_pcm_open
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_params
Apr 08 19:00:58 little kernel: aaudio_pcm_trigger 1
Apr 08 19:00:58 little kernel: aaudio: Backed up the buffer in 72794953ns [16640]
Apr 08 19:00:58 little kernel: aaudio aaudio: Received timestamp update for dev=3a ts=480cfd953e seed=0
Apr 08 19:00:58 little kernel: aaudio: Started the audio device in 97586739ns
Apr 08 19:00:58 little kernel: aaudio_pcm_open
Apr 08 19:00:58 little kernel: aaudio_pcm_hw_params
Apr 08 19:00:58 little kernel: x86/PAT: pulseaudio:1681 map pfn RAM range req uncached-minus for [mem 0x7c440000-0x7c470fff], got write-back
Apr 08 19:00:58 little kernel: aaudio_pcm_trigger 1
Apr 08 19:00:58 little kernel: aaudio: Started the audio device in 3235965ns
Apr 08 19:00:58 little kernel: aaudio_pcm_pointer while not started
Apr 08 19:00:58 little kernel: aaudio aaudio: Received timestamp update for dev=21 ts=4810dd812a seed=0
Apr 08 19:00:59 little systemd[748]: Started Sound Service.
Apr 08 19:00:59 little kernel: aaudio aaudio: Received timestamp update for dev=3a ts=4821a726d0 seed=0
Apr 08 19:00:59 little kernel: aaudio aaudio: Received timestamp update for dev=21 ts=48258774df seed=0
Apr 08 19:00:59 little kernel: aaudio aaudio: Received timestamp update for dev=3a ts=4836508bf2 seed=0
Apr 08 19:00:59 little kernel: aaudio_pcm_trigger 0
Apr 08 19:00:59 little kernel: aaudio_pcm_hw_free
Apr 08 19:00:59 little kernel: aaudio_pcm_close
Apr 08 19:00:59 little kernel: aaudio_pcm_trigger 0
Apr 08 19:00:59 little kernel: aaudio_pcm_hw_free
Apr 08 19:00:59 little kernel: aaudio_pcm_close
Apr 08 19:00:59 little systemd[748]: pulseaudio.service: Main process exited, code=killed, status=9/KILL
```

jbg, 2020-04-08:
Disabling realtime scheduling in pulseaudio fixes the above. It appears that with this driver, pulseaudio spends too much time in realtime blocks and gets killed.

cristianmiranda, 2020-04-20:
Hi guys, you're doing a great job @MCMrARM ! 
I have a quick question, I'm running Ubuntu on a MacbookPro 2017 (no T2 chip), would this approach work for me? Can i try https://github.com/MCMrARM/mbp2018-bridge-drv ?
Thanks!

leifliddy, 2020-04-21:
Have you tried this repo yet?
https://github.com/MCMrARM/mbp2018-bridge-drv

I would just add add an ```install``` command to that ```Makefile```
ie
```
obj-m += bce.o
bce-objs := pci.o mailbox.o queue.o queue_dma.o vhci/vhci.o vhci/queue.o vhci/transfer.o audio/audio.o audio/protocol.o audio/protocol_bce.o audio/pcm.o

MY_CFLAGS += -DWITHOUT_NVME_PATCH
#MY_CFLAGS += -g -DDEBUG
ccflags-y += ${MY_CFLAGS}
CC += ${MY_CFLAGS}

all:
        make -C /lib/modules/$(shell uname -r)/build M=$(PWD) modules

clean:
        make -C /lib/modules/$(shell uname -r)/build M=$(PWD) clean

install:
        mkdir -p /lib/modules/$(shell uname -r)/updates
        cp bce.ko /lib/modules/$(shell uname -r)/updates
        depmod -a
```
**the lines under ```all: clean: and install:``` need to be indented with a ```tab```
then just run, ```make``` + ```make install``` + ```reboot```
Give that a shot and see if it works.

cristianmiranda, 2020-04-29:
Hey, thanks! I haven't installed linux on my T2 macbook, but if I decide to do it I'll definitely check this out. Thanks again!

mushoku-ningen, 2020-07-30:
@leifilddy it doesnt work why

mushoku-ningen, 2020-07-30:
@leifliddy  it doesnt work why
im a simple man who uses 15,4 mac and wants sound in arch linux

mushoku-ningen, 2020-07-30:
@leifliddy i did exactly what you said but no sound in my arch

mushoku-ningen, 2020-07-30:
@leifliddy 
my problem solved
you should add ``` mkinitcpio -p linux ``` at the end

pklement77, 2021-07-12:
Hello, got some kernel panic issues on 15,2 - sometimes it works, sometimes everything stops (mouse not moving, X froze, unable to do anything).. seems not to do these when I don't install the T2 audio drivers (but then I only get speakers, no mic).

Thanks for your all's great work though!

Linux version 5.12.15-1-mbp (linux-mbp@archlinux) (gcc (GCC) 11.1.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Mon, 12 Jul 2021 01:16:58 +0000

Arch distro. Followed the howto on t2linux.org