Dunedan/mbp-2016-linux #112: MacBook Pro 15+: Wifi Support

aunali1, 2019-08-06:
Discussion and updates regarding Broadcom Wifi related support. Parent thread is #71.

jhohisel, 2019-08-17:
Hi everyone! I've been following this thread as I work on getting a USB SSD with linux that works on both of my Macs: MacBookAir8,2 (2019 MacBook Air) and iMac19,1 (2019 iMac). Fortunately the 2019 iMac does _not_ contain a T2 chip, so things work fairly well out of the box.

I'm attempting to get WiFi working on both models. The iMac seems to have the same chipset as the MacBookPro15,x (BCM4364, codename Midway) but the MacBook Air has a BCM4355 chipset (codename Hawaii). I created and applied this patch (based on the work from @MCMrARM): https://github.com/jhohisel/macbookair8-2-linux/blob/master/brcmfmac.patch and copied the firmware from macOS.

However, the firmware fails to load (iMac):
```
[   20.314189] usbcore: registered new interface driver brcmfmac
[   20.330616] brcmfmac 0000:03:00.0: enabling device (0000 -> 0002)
[   20.439205] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[   23.374627] brcmfmac 0000:03:00.0: brcmf_pcie_download_fw_nvram: FW failed to initialize
```

MacBook Air fails with a slightly different error:
```
[   27.958248] brcmfmac 0000:02:00.0: enabling device (0000 -> 0002)
[   28.121706] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4355-pcie for chip BCM4355/12
[   28.126287] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355-pcie.Apple Inc.-MacBookAir8,2.txt failed with error -2
[   31.030139] brcmfmac 0000:02:00.0: brcmf_pcie_download_fw_nvram: FW failed to initialize
```

I observe a different error when the NVRAM file (.txt) is missing, so I know that it is correctly detecting the presence of these files and attempting to load them.

Top tip: you can discover the firmware filenames that Apple uses by running:
`ioreg -l | grep RequestedFiles`
The output will look something like:
```
"RequestedFiles" = ({"Firmware"="C-4364__s-B2/midway.trx","TxCap"="C-4364__s-B2/midway-X0.txcb","Regulatory"="C-4364__s-B2/midway-X0.clmb","NVRAM"="C-4364__s-B2/P-midway-X0_M-HRPN_V-u__m-7.5.txt"})
```

ppaulweber, 2019-08-17:
> Top tip: you can discover the firmware filenames that Apple uses by running:
ioreg -l | grep RequestedFiles

@jhohisel well done and nice tip! 

I'm experimenting with the WLAN (WiFi) support on my `MacBookAir8,1` since a week or so, I've pushed my version of this patch now as well (see reference above). Can you check if your `MacBookAir8,2` platform goes along fine with the provided `bcm5974`, `hid`, `nvme`, and `brcmfmac` patches in https://github.com/ppaulweber/linux-mba? Thx!

jhohisel, 2019-08-17:
I am currently running your `bcm5974` and `hid` patches. Applying them added two-finger scrolling, right-click and function key support.

I'll try the `nvme` patch now.

jhohisel, 2019-08-17:
```
bluehax@bluehax:~$ lsblk
NAME           MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
sda              8:0    0 465.8G  0 disk  
├─sda1           8:1    0   448M  0 part  /boot/efi
├─sda2           8:2    0   4.7G  0 part  /media/bluehax/Debian Linux
├─sda3           8:3    0   4.7G  0 part  /boot
└─sda4           8:4    0   456G  0 part  
  └─sdb4_crypt 254:0    0   456G  0 crypt /
nvme0n1        259:0    0   113G  0 disk  
├─nvme0n1p1    259:1    0   300M  0 part  
└─nvme0n1p2    259:2    0 112.7G  0 part 
```
NVMe Works! I'll have to try installing it to a partition on the laptop now.

One issue I've been unable to fix is getting the keyboard working when I unlock my root partition during boot. It works during the GRUB menu, then stops working when the kernel prompts to unlock the drive for dm-crypt, then functions immediately after when it loads the `bce` module.

I built-in mbp2018-bridge-drv to the kernel by adding a Kconfig entry and declaring the source to the Makefile in `drivers/`. Applied the patches for `bcm5974` and `hid`, then set the module load order in `/etc/modules-load.d/bce.conf`.

ppaulweber, 2019-08-17:
> One issue I've been unable to fix is getting the keyboard working when I unlock my root partition during boot. It works during the GRUB menu, then stops working when the kernel prompts to unlock the drive for dm-crypt, then functions immediately after when it loads the bce module.

If you really need the keyboard in this case you have to take a look into your current `/etc/mkinitcpio.conf` setup. You shall include there the `bce` in the `MODULE=...` variable in order that the `bce` module even gets loaded and integrated into the initramfs creation.

Dunedan, 2019-08-18:
Please focus on wifi in this issue and discuss other topics (like keyboard or NVMe) in separate issues.

mikeeq, 2019-08-19:
Is WiFi (brcmfmac4364) working for anybody (besides @MCMrARM)?

aunali1, 2019-08-20:
@mikeeq What kernel patch are you using for WiFi support?

aunali1, 2019-08-21:
@jhohisel @mikeeq Works perfectly fine for me:

```
[    8.196831] usbcore: registered new interface driver brcmfmac
[    8.212808] brcmfmac 0000:03:00.0: enabling device (0000 -> 0002)
[    8.319800] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[    9.521164] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[    9.840308] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4364/3 wl0: Mar 28 2019 19:17:52 version 9.137.9.0.32.6.34 FWID 01-36f56c94
[    9.933835] brcmfmac 0000:03:00.0 wlp3s0: renamed from wlan0
```

mikeeq, 2019-08-21:
@aunali1 yep, you're right - I've missed one line when I was generating a patch.
https://github.com/mikeeq/mbp-fedora-kernel/commit/d8f5578d4fd9c9ee01c2a125dae30237aefcf391

@ppaulweber could you confirm if that fix is not needed for MacBookAir8,2?

ppaulweber, 2019-08-21:
@aunali1 thanks for the hint and I've experimented for my `MacBookAir8,1` that this `brcm` driver uses the same RAM base as the `brcm` driver of the `MacBookPro15,x`. It now allocates and loads correctly the firmware, and I get the wlan interface.

@jhohisel can you check if this latest patch works for your `MacBookAir8,2` as well, thx!

@mikeeq it does need the same fix, which I've already updated in my repo (see commit reference above)

~~~
[  702.402953] usbcore: registered new interface driver brcmfmac
[  702.512262] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4355-pcie for chip BCM4355/12
[  702.960842] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4355-pcie for chip BCM4355/12
[  702.981704] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4355/12 wl0: Mar 28 2019 19:57:32 version 9.137.9.0.32.6.33 FWID 01-c5e1774f
[  703.109625] brcmfmac 0000:02:00.0 wlp2s0: renamed from wlan0
~~~

jhohisel, 2019-08-22:
@ppaulweber The good news is that now the card is correctly recognized and shows as a WIFI device. The bad news is that I can't seem to connect to my network using it. It repeatedly times out and prompts for the password. I tried switching from wpa_supplicant to iwd as suggested in issue 71, but no difference. I also tried using wicd instead of NetworkManager, it fails to connect saying bad password. I have not yet tried an open network, and yes I am entering the correct password.

Has anyone else experienced this?

mikeeq, 2019-08-22:
@jhohisel try to reload brcmfmac kernel module. I'm experiencing the same issue and after module reload I'm able to connect to WPA2 network ;)

https://github.com/mikeeq/mbp-fedora#known-issues

ppaulweber, 2019-08-22:
@jhohisel I can confirm that for me is `wpa_supplicant`no longer working as well. With `iwd`the WLAN (WiFi) is working as expected. Thx @mikeeq for collecting the known issues in your repo!

jhohisel, 2019-08-23:
I was missing:
```
[device]
wifi.backend=iwd
```
From `/etc/NetworkManager/NetworkManager.conf`

Now I can connect!

macphotoed, 2019-08-24:
Sorry for the silly question all - when I locate the files from mojave, where do I copy them on the linux side?

aunali1, 2019-08-24:
@macphotoed Just copy the *.trx file to /lib/firmware/brcm/brcmfmac4364-pcie.trx, the *.clmb file to /lib/firmware/brcm/brcmfmac4364-pcie.clm_blob and the *.txt file to  /lib/firmware/brcm/brcmfmac4364-pcie.Apple Inc.-MacBookPro15,1.txt

macphotoed, 2019-08-24:
Amazing - thank you again @aunali1 - got the files, thankyou. Back on the linux side of things, copied them over, reboot but there doesn't seem to be any interface for the wifi or device? I followed through the thread and not sure that I missed anything? Running the linux-mba kernel from @ppaulweber currently. I have not installed anything else a yet - is there a module I need to load or something?

MCMrARM, 2019-08-28:
By the way 5Ghz generally works for me, through a full multiband 5ac does not achieve full speeds.

wzygzlm, 2019-08-30:
Sorry. I'd like to ask if I don't have IDA，how can I get the wifi firmware?
Do anybody can share one with me. Mine model is mbp2010 13inch.

aunali1, 2019-08-31:
It seems that with 5.2.11, 5 Ghz works without any problems, and in some cases faster than macOS!

MCMrARM, 2019-09-01:
@wzygzlm I mean, this no longer requires IDA, the firmware is in /usr/share/firmware on OS X.

TRPB, 2019-09-16:
can anyone tell me where the trx and clmb files are? On Arch if it makes any difference.

YungRaj, 2019-09-17:
Can someone post to the original MacBook Pro 15,x thread about how to get the wifi working? It looks like many people have successfully gotten it working. @TRPB has a guide about how to build ArchLinux for these machines but doesn't have the wifi module working. Eventually I plan on writing a guide on how to get it working on more than just one Linux distro collectively.

MCMrARM, 2019-09-17:
TL;Dr with the wifi is to first find your firmware files using `ioreg -l | grep C-4364`, then copy them from `/usr/share/firmware/wifi`. On Linux, the .trx file for your model goes to /lib/firmware/brcm/brcmfmac4364-pcie.bin, the .clmb goes to /lib/firmware/brcm/brcmfmac4364-pcie.clm_blob and the .txt to something like /lib/firmware/brcm/brcmfmac4364-pcie.Apple Inc.-MacBookPro15,2.txt
All of this requires a kernel patch: https://mrarm.io/u/Paste%202019-07-26%2012-30-16.txt

EmmaDSamms, 2019-09-21:
Anyone else not have this working on MBA 8,1(air 2018) it shows up as a ethernet device

EmmaDSamms, 2019-09-21:
Anyone else not have this working on MBA 8,1(air 2018) it shows up as a ethernet device, thanks :)

MCMrARM, 2019-09-21:
@EmmaDSamms sorry, what? Show the output of `lspci`?

caioquirino, 2019-09-23:
Hi people, for me I have the files in /lib/firmware/brcm/ exactly as @MCMrARM mentioned but when I run `modprobe brcm80211` either in boot time or after, my arch-linux just freeze completely. How can I investigate it? because it just hangs and I need to reboot...

loafbox, 2019-10-01:
So I've completely uninstalled MacOS and think I copied over the wrong firmware files before wiping it out.  Would anyone happen to have firmware files for  MacBookPro15,x T2 touchbar model?  
Need...
```
"Firmware"="C-4364__s-B2/kauai.trx"
"imagename"="C-4364__s-B2/kauai-X0.clmb"
"NVRAM"="C-4364__s-B2/P-kauai-X0_M-HRPN_V-u__m-7.5.txt"
```

MCMrARM, 2019-10-03:
I'd suggest against uninstalling MacOS completely because of MacOS being your best firmware update source, so you should keep it if you want to say, update bridgeOS.

miki2o, 2019-10-29:
Is it possible to apply this procedure to MBP 13,3? Taking the .trx, .clmb and .txt from osx and putting them in linux?

cubimon, 2019-12-08:
I can't get it working on my MBP 2019 15,1. I installed `iwd`, `NetworkManager`, enabled the services `iwd` and `NetworkManager`, set `iwd` as default wifi backend in `/etc/NetworkManager/NetworkManager.conf`, copied the three files from OSX to proper names in `/lib/firmware/brcm/`. I have aunali1's linux mbp kernel which contains the wifi patches. What am I missing? `dmesg` says this after boot:
```
[    4.446767] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[    4.479882] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4364/3 wl0: Mar 28 2019 19:17:52 version 9.137.9.0.32.6.34 FWID 01-36f56c94
[    4.498323] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-16
```
When I run `sudo modprobe -r brcmfmac; sudo modprobe brcmfmac`, I see
```
[ 1137.569122] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[ 1137.569355] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
[ 1137.569359] ieee80211 phy0: brcmf_cfg80211_get_channel: chanspec failed (-5)
```
And a bit later
```
[ 1138.193219] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[ 1139.068592] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[ 1139.098725] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4364/3 wl0: Mar 28 2019 19:17:52 version 9.137.9.0.32.6.34 FWID 01-36f56c94
```
But I stilll can't use it. So am I too stupid to use the wifi device or idk.
`ip addr` shows this:
```
4: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether ${macaddress} ff:ff:ff:ff:ff:ff
```
I would be grateful for any kind of help.

geovie, 2019-12-21:
After following @MCMrARM's guide to copy the WIFI firmware files I get the following error when the firmware files are loaded (`dmesg | grep brcm` output):
```
usbcore: registered new interface driver brcmfmac  
brcmfmac 0000:01:00.0: enabling device (0000 -> 0002)  
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3  
bluetooth hci0: Direct firmware load for brcm/BCM.hcd failed with error -2  
Bluetooth: hci0: BCM: Patch brcm/BCM.hcd not found  
brcmfmac 0000:01:00.0: brcmf_pcie_download_fw_nvram: FW failed to initialize  
```
More information here: https://github.com/mikeeq/mbp-fedora-kernel/issues/1
Has anyone an idea why it's failing to initialize?

TIA

Xinkai, 2019-12-24:
I am having the same issue @geovie. This is the dmesg with `brcmfmac.debug=2097150`. (Some omitted).
```
[    4.055976] brcmfmac: brcmfmac_module_init No platform data available.
[    4.055988] brcmfmac: brcmf_usb_register Enter
[    4.056000] usbcore: registered new interface driver brcmfmac
[    4.056000] brcmfmac: brcmf_pcie_register Enter
[    4.056024] brcmfmac: brcmf_pcie_probe Enter 14e4:4464
[    4.056135] brcmfmac 0000:05:00.0: enabling device (0000 -> 0002)
[    4.056320] brcmfmac: brcmf_pcie_get_resource Phys addr : reg space = 000000009656594a base addr 0x00000081400000
[    4.056321] brcmfmac: brcmf_pcie_get_resource Phys addr : mem space = 000000001afaff3b base addr 0x00000081000000 size 0x400000
[    4.056440] brcmfmac: brcmf_chip_recognition found AXI chip: BCM4364/3
[    4.062520] brcmfmac: brcmf_chip_cores_check  [1 ] core 0x800:59 base 0x18000000 wrap 0x18100000
[    4.062522] brcmfmac: brcmf_chip_cores_check  [2 ] core 0x812:58 base 0x18001000 wrap 0x18101000
[    4.062522] brcmfmac: brcmf_chip_cores_check  [3 ] core 0x83e:12 base 0x18002000 wrap 0x18102000
[    4.062523] brcmfmac: brcmf_chip_cores_check  [4 ] core 0x83c:16 base 0x18003000 wrap 0x18103000
[    4.062523] brcmfmac: brcmf_chip_cores_check  [5 ] core 0x812:58 base 0x18004000 wrap 0x18104000
[    4.062524] brcmfmac: brcmf_chip_cores_check  [6 ] core 0x829:22 base 0x18005000 wrap 0x18105000
[    4.062525] brcmfmac: brcmf_chip_cores_check  [7 ] core 0x840:7  base 0x1800a000 wrap 0x00000000
[    4.062525] brcmfmac: brcmf_chip_cores_check  [8 ] core 0x135:0  base 0x00000000 wrap 0x18108000
[    4.062526] brcmfmac: brcmf_chip_cores_check  [9 ] core 0x240:0  base 0x00000000 wrap 0x00000000
[    4.062526] brcmfmac: brcmf_chip_set_passive Enter
[    4.169825] brcmfmac: brcmf_chip_set_passive Enter
[    4.170427] brcmfmac: brcmf_chip_get_raminfo RAM: base=0x160000 size=1310720 (0x140000) sr=0 (0x0)
[    4.170439] brcmfmac: brcmf_chip_setup ccrev=59, pmurev=29, pmucaps=0x3c8c3f1d
[    4.170440] brcmfmac: brcmf_get_module_param Enter, bus=2, chip=17252, rev=3
[    4.170444] brcmfmac: brcmf_alloc Enter
[    4.170451] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364-pcie for chip BCM4364/3
[    4.173974] brcmfmac: brcmf_fw_get_firmwares enter: dev=0000:05:00.0
[    4.175877] brcmfmac: brcmf_fw_complete_request firmware brcm/brcmfmac4364-pcie.bin found
[    4.176362] brcmfmac: brcmf_fw_complete_request firmware brcm/brcmfmac4364-pcie.txt found
[    4.176364] brcmfmac: brcmf_fw_request_nvram_done enter: dev=0000:05:00.0
[    4.176490] brcmfmac: brcmf_fw_request_nvram_done nvram 000000009a836282 len 12896
[    4.176812] brcmfmac: brcmf_chip_get_raminfo RAM: base=0x160000 size=1310720 (0x140000) sr=0 (0x0)
[    4.176813] brcmfmac: brcmf_pcie_download_fw_nvram Halt ARM.
[    4.176814] brcmfmac: brcmf_pcie_download_fw_nvram Download FW brcm/brcmfmac4364-pcie.bin
[    4.466551] brcmfmac: brcmf_pcie_download_fw_nvram Download NVRAM brcm/brcmfmac4364-pcie.txt
[    4.468466] brcmfmac: brcmf_pcie_download_fw_nvram Bring ARM in running state
[    4.468467] brcmfmac: brcmf_chip_set_active Enter
[    4.468810] brcmfmac: brcmf_pcie_download_fw_nvram Wait for FW init
[    6.740083] brcmfmac 0000:05:00.0: brcmf_pcie_download_fw_nvram: FW failed to initialize
[    6.747010] brcmfmac: brcmf_pcie_remove Enter
[    6.747011] brcmfmac: brcmf_detach Enter
[    6.760007] brcmfmac: brcmf_bus_change_state 0 -> 0
[    6.760013] ------------[ cut here ]------------
[    6.760018] WARNING: CPU: 8 PID: 241 at kernel/workqueue.c:3037 __flush_work+0x1c8/0x1e0
[    6.760019] Modules linked in: input_leds hid_apple apple_ib_als(OE) industrialio_triggered_buffer kfifo_buf industrialio apple_ib_tb(OE) apple_ibridge(OE) uvcvideo videobuf2_vmalloc videobuf2_memops joydev videobuf2_v4l2 hid_generic cdc_mbim mousedev videobuf2_common cdc_wdm videodev cdc_ncm usbnet usbhid hid mc mii cmac algif_hash algif_skcipher af_alg intel_rapl_msr intel_rapl_common bnep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul msr ghash_clmulni_intel brcmfmac brcmutil nls_iso8859_1 nls_cp437 snd_hda_codec_hdmi aesni_intel vfat fat snd_hda_intel crypto_simd cfg80211 mei_hdcp hci_uart snd_intel_nhlt 8250_dw cryptd glue_helper intel_cstate btqca intel_uncore btrtl iTCO_wdt btbcm iTCO_vendor_support btintel mei_me bluetooth intel_lpss_pci mmc_core snd_hda_codec intel_rapl_perf mei i2c_i801 pcspkr intel_lpss idma64 snd_hda_core intel_pch_thermal snd_hwdep ecdh_generic sbs rfkill apple_gmux ecc sbshc acpi_tad apple_bl evdev mac_hid
[    6.760049]  ac pkcs8_key_parser crypto_user apple_bce(OE) snd_pcm snd_timer snd soundcore ip_tables x_tables btrfs libcrc32c crc32c_generic xor raid6_pq xhci_pci crc32c_intel xhci_hcd amdgpu gpu_sched i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm agpgart
[    6.760062] CPU: 8 PID: 241 Comm: kworker/8:1 Tainted: G           OE     5.4.6-1-mbp #1
[    6.760063] Hardware name: Apple Inc. MacBookPro15,3/Mac-1E7E29AD0135F9BC, BIOS 1037.60.58.0.0 (iBridge: 17.16.12551.0.0,0) 11/14/2019
[    6.760065] Workqueue: events request_firmware_work_func
[    6.760067] RIP: 0010:__flush_work+0x1c8/0x1e0
[    6.760069] Code: 8a 88 00 e8 fa 40 06 00 45 31 f6 eb 85 8b 0b 48 8b 53 08 83 e1 08 48 0f ba 2b 03 80 c9 f0 e9 4d ff ff ff 0f 0b e9 68 ff ff ff <0f> 0b 45 31 f6 e9 5e ff ff ff e8 e9 03 fe ff 66 0f 1f 84 00 00 00
[    6.760070] RSP: 0018:ffffad64c05cbbe0 EFLAGS: 00010246
[... cut ...]
[    6.760246]  brcmf_pcie_setup+0x38/0xd90 [brcmfmac]
[    6.760255]  brcmf_fw_request_done+0x65/0xe0 [brcmfmac]
[    6.760257]  request_firmware_work_func+0x4f/0x8d
[    6.760259]  process_one_work+0x1e2/0x3b0
[    6.760261]  worker_thread+0x4a/0x3d0
[    6.760263]  kthread+0xfb/0x130
[    6.760264]  ? process_one_work+0x3b0/0x3b0
[    6.760265]  ? kthread_park+0x90/0x90
[    6.760267]  ret_from_fork+0x1f/0x40
[    6.760269] ---[ end trace 83d738c81ce04d6a ]---
[    6.760271] brcmfmac: brcmf_proto_detach Enter
```

Xinkai, 2019-12-24:
From what I understand, the firmware is executed not on the CPU but on the Wifi device. Is there a way to make the Broadcom chip talk back so we can know what's going on? I noticed that there's a `fwsignal.c` file in the kernel tree that suggests it kinda does that, but I don't know how to use it.

Let me know if any more information is needed.

yurnam, 2019-12-30:
similar issue with mbp 15,4   it uses a BCM4377 with a device id: 4488 can't find anything
attempted to implement it like in https://mrarm.io/u/Paste%202019-07-26%2012-30-16.txt but with 4488 instead of 4464 & 4377 instead of 4364, copied the trx, clmb, and txt, from macOS but when i load the brcmfmac module this happens:

```
# modprobe brcmfmac
[24.xxxxx] brcmfmac: brcmf_fw_alloc_request: using brcmfmac4377-pcie for chip 4377/4
[26.xxxxx] brcmfmac: 0000:01:00.0 brcmf_pcie_download_nvram: Invalid shared RAM address 0xfffffff

```
then my mac freezes and after a few seconds it restarts.
this only happens when i have the brcmfmac4377-pcie* files in /lib/firmware/brcm otherwise it obviously just exits with cant find the files.

output of lspci shows 2 Broadcom network controllers:

```
00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation Device 3ea6 (rev 01)
00:12.0 Signal processing controller: Intel Corporation Cannon Point-LP Thermal Controller (rev 30)
00:14.0 USB controller: Intel Corporation Cannon Point-LP USB 3.1 xHCI Controller (rev 30)
00:14.2 RAM memory: Intel Corporation Cannon Point-LP Shared SRAM (rev 30)
00:16.0 Communication controller: Intel Corporation Cannon Point-LP MEI Controller #1 (rev 30)
00:1c.0 PCI bridge: Intel Corporation Device 9dba (rev f0)
00:1c.4 PCI bridge: Intel Corporation Cannon Point-LP PCI Express Root Port #5 (rev f0)
00:1d.0 PCI bridge: Intel Corporation Cannon Point-LP PCI Express Root Port #13 (rev f0)
00:1e.0 Communication controller: Intel Corporation Device 9da8 (rev 30)
00:1f.0 ISA bridge: Intel Corporation Cannon Point-LP LPC Controller (rev 30)
00:1f.3 Audio device: Intel Corporation Cannon Point-LP High Definition Audio Controller (rev 30)
00:1f.4 SMBus: Intel Corporation Cannon Point-LP SMBus Controller (rev 30)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Point-LP SPI Controller (rev 30)
01:00.0 Network controller: Broadcom Inc. and subsidiaries Device 4488 (rev 04)
01:00.1 Network controller: Broadcom Inc. and subsidiaries Device 5fa0 (rev 04)
02:00.0 Mass storage controller: Apple Inc. ANS2 NVMe Controller (rev 01)
02:00.1 Non-VGA unclassified device: Apple Inc. T2 Bridge Controller (rev 01)
02:00.2 Non-VGA unclassified device: Apple Inc. T2 Secure Enclave Processor (rev 01)
02:00.3 Multimedia audio controller: Apple Inc. Apple Audio Device (rev 01)
03:00.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
04:00.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
04:01.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
04:02.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
04:04.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
05:00.0 System peripheral: Intel Corporation JHL7540 Thunderbolt 3 NHI [Titan Ridge 4C 2018] (rev 06)
06:00.0 USB controller: Intel Corporation JHL7540 Thunderbolt 3 USB Controller [Titan Ridge 4C 2018] (rev 06)
```

would be nice if anyone could help me :)

aunali1, 2019-12-31:
@yurnam The patch that you are basing your changes on is incomplete as it does not include the rambase address [diff](https://github.com/aunali1/linux-mbp-arch/blob/master/wifi.patch#L4).

Since this a new chip variant, we can potentially determine the rambase by enabling some more debug info. Please post the dmesg output with the additional kernel argument `brcmfmac.debug=2097150`. I'm particularly interested in the line containing `brcmf_chip_get_raminfo`.

Can you also clone this repo and run`get-info.sh` and PR the results. the 15,4 variant seems to differ quite a bit AFAI can tell.

yurnam, 2019-12-31:
oh sorry, I was using the patch which includes the rambase address 0x160000 
i used this address also for the 4377 chip, which was probably a bad idea.
if i change or remove it, the mac doesnt freeze anymore.

anyway i made a PR to your repo with the mbp15,4 dumps 
thank you!

aunali1, 2020-01-01:
@yurnam Hmm, it seems that the rambase address needs to be found through trial and error. Can you please test through the rambase addresses in [this](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/broadcom/brcm80211/brcmfmac/chip.c?h=v5.4.7#n666) switch case (0x160000 is already known not to work).

yurnam, 2020-01-01:
tried all of them output looks always like this:

```
$ dmesg | grep brcmf

[    0.115781] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.5.0-rc4 root=UUID=967cef72-c47e-4565-9c41-92ef1ac054da ro net.ifnames=0 biosdevname=0 quiet splash brcmfmac.debug=2097150
[  229.970017] brcmfmac: brcmfmac_module_init No platform data available.
[  229.970022] brcmfmac: brcmf_pcie_register Enter
[  229.970038] brcmfmac: brcmf_pcie_probe Enter 14e4:4488
[  229.986615] brcmfmac 0000:01:00.0: enabling device (0000 -> 0002)
[  229.986852] brcmfmac: brcmf_pcie_get_resource Phys addr : reg space = 00000000fbeb7416 base addr 0x00000091800000
[  229.986853] brcmfmac: brcmf_pcie_get_resource Phys addr : mem space = 000000004919b7a6 base addr 0x00000091000000 size 0x400000
[  229.986962] brcmfmac: brcmf_chip_recognition found AXI chip: BCM4377/4
[  229.987331] brcmfmac: brcmf_chip_cores_check  [1 ] core 0x800:64 base 0x18000000 wrap 0x18100000
[  229.987332] brcmfmac: brcmf_chip_cores_check  [2 ] core 0x812:82 base 0x18001000 wrap 0x18101000
[  229.987333] brcmfmac: brcmf_chip_cores_check  [3 ] core 0x83e:15 base 0x18002000 wrap 0x18102000
[  229.987334] brcmfmac: brcmf_chip_cores_check  [4 ] core 0x83c:64 base 0x18003000 wrap 0x18103000
[  229.987334] brcmfmac: brcmf_chip_cores_check  [5 ] core 0x812:82 base 0x18004000 wrap 0x18104000
[  229.987335] brcmfmac: brcmf_chip_cores_check  [6 ] core 0x844:2  base 0x18005000 wrap 0x18105000
[  229.987336] brcmfmac: brcmf_chip_cores_check  [7 ] core 0x840:15 base 0x18010000 wrap 0x00000000
[  229.987336] brcmfmac: brcmf_chip_cores_check  [8 ] core 0x827:35 base 0x18012000 wrap 0x00000000
[  229.987337] brcmfmac: brcmf_chip_cores_check  [9 ] core 0x240:0  base 0x00000000 wrap 0x1810c000
[  229.987338] brcmfmac: brcmf_chip_cores_check  [10] core 0x135:0  base 0x00000000 wrap 0x18112000
[  229.987338] brcmfmac: brcmf_chip_cores_check  [11] core 0x135:0  base 0x00000000 wrap 0x18113000
[  229.987339] brcmfmac: brcmf_chip_set_passive Enter
[  230.093188] brcmfmac: brcmf_chip_set_passive Enter
[  230.093773] brcmfmac: brcmf_chip_get_raminfo RAM: base=0x198000 size=1638400 (0x190000) sr=0 (0x0)
[  230.093784] brcmfmac: brcmf_chip_setup ccrev=64, pmurev=35, pmucaps=0x84565f23
[  230.093786] brcmfmac: brcmf_get_module_param Enter, bus=2, chip=17271, rev=4
[  230.093790] brcmfmac: brcmf_alloc Enter
[  230.093802] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4377-pcie for chip BCM4377/4
[  230.093807] brcmfmac: brcmf_fw_get_firmwares enter: dev=0000:01:00.0
[  230.096520] brcmfmac: brcmf_fw_complete_request firmware brcm/brcmfmac4377-pcie.bin found
[  230.097082] brcmfmac: brcmf_fw_complete_request firmware brcm/brcmfmac4377-pcie.txt found
[  230.097084] brcmfmac: brcmf_fw_request_nvram_done enter: dev=0000:01:00.0
[  230.097298] brcmfmac: brcmf_fw_request_nvram_done nvram 0000000021820ad9 len 4924
[  230.097457] brcmfmac: brcmf_chip_get_raminfo RAM: base=0x198000 size=1638400 (0x190000) sr=0 (0x0)
[  230.097458] brcmfmac: brcmf_pcie_download_fw_nvram Halt ARM.
[  230.097459] brcmfmac: brcmf_pcie_download_fw_nvram Download FW brcm/brcmfmac4377-pcie.bin
[  230.259368] brcmfmac: brcmf_pcie_download_fw_nvram Download NVRAM brcm/brcmfmac4377-pcie.txt
[  230.260366] brcmfmac: brcmf_pcie_download_fw_nvram Bring ARM in running state
[  230.260367] brcmfmac: brcmf_chip_set_active Enter
[  230.260566] brcmfmac: brcmf_pcie_download_fw_nvram Wait for FW init
[  232.533371] brcmfmac 0000:01:00.0: brcmf_pcie_download_fw_nvram: FW failed to initialize
[  232.533394] brcmfmac: brcmf_pcie_remove Enter
[  232.533397] brcmfmac: brcmf_detach Enter
[  232.560082] brcmfmac: brcmf_bus_change_state 0 -> 0
[  232.560111] Modules linked in: brcmfmac brcmutil sd_mod sg uas usb_storage scsi_mod r8188eu(C) lib80211 cfg80211 fuse hid_apple input_leds apple_ib_als(OE) industrialio_triggered_buffer kfifo_buf industrialio apple_ib_tb(OE) apple_ibridge(OE) joydev cdc_mbim cdc_wdm mousedev hid_generic uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev cdc_ncm usbhid usbnet mc hid mii snd_hda_codec_hdmi squashfs rfkill loop intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp snd_sof_pci kvm_intel snd_sof_intel_byt i915 snd_sof_intel_ipc kvm snd_sof_xtensa_dsp snd_sof_intel_hda_common snd_soc_hdac_hda irqbypass snd_sof_intel_hda snd_sof snd_hda_ext_core snd_soc_acpi_intel_match i2c_algo_bit crct10dif_pclmul snd_soc_acpi crc32_pclmul ledtrig_audio drm_kms_helper ghash_clmulni_intel snd_soc_core nls_iso8859_1 snd_compress aesni_intel nls_cp437 drm ac97_bus crypto_simd vfat snd_pcm_dmaengine fat cryptd glue_helper mei_hdcp intel_gtt intel_cstate
[  232.560329]  ? __brcmf_dbg+0x8b/0x110 [brcmfmac]
[  232.560362]  brcmf_fweh_detach+0x15/0x60 [brcmfmac]
[  232.560378]  brcmf_detach+0x72/0xf0 [brcmfmac]
[  232.560395]  brcmf_pcie_remove+0x70/0x110 [brcmfmac]
[  232.560426]  brcmf_pcie_setup+0x38/0xd70 [brcmfmac]
[  232.560446]  brcmf_fw_request_done+0x65/0xe0 [brcmfmac]
[  232.560555] WARNING: CPU: 7 PID: 182 at drivers/net/wireless/broadcom/brcm80211/brcmfmac/fweh.c:309 brcmf_fweh_detach+0x5b/0x60 [brcmfmac]
[  232.560556] Modules linked in: brcmfmac brcmutil sd_mod sg uas usb_storage scsi_mod r8188eu(C) lib80211 cfg80211 fuse hid_apple input_leds apple_ib_als(OE) industrialio_triggered_buffer kfifo_buf industrialio apple_ib_tb(OE) apple_ibridge(OE) joydev cdc_mbim cdc_wdm mousedev hid_generic uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2 [... cut ...] d_soc_core nls_iso8859_1 snd_compress aesni_intel nls_cp437 drm ac97_bus crypto_simd vfat snd_pcm_dmaengine fat cryptd glue_helper mei_hdcp intel_gtt intel_cstate
[  232.560678] RIP: 0010:brcmf_fweh_detach+0x5b/0x60 [brcmfmac]
[  232.560730]  brcmf_detach+0x72/0xf0 [brcmfmac]
[  232.560747]  brcmf_pcie_remove+0x70/0x110 [brcmfmac]
[  232.560784]  brcmf_pcie_setup+0x38/0xd70 [brcmfmac]
[  232.560812]  brcmf_fw_request_done+0x65/0xe0 [brcmfmac]
[  232.560879] brcmfmac: brcmf_proto_detach Enter
```

yurnam, 2020-01-06:
is there any other way finding out what to do;

just trying to bruteforce the rambase address doesn't seem to help

ktb92677, 2020-01-07:
I am having trouble getting my Macbook Air 8,2 to get wifi. Here is what I did:

From MacOS:
ioreg -l | grep RequestedFiles
C-4355__s-C1/hawaii.trx, C-4355__s-C1/hawaii-X0.clmb, C-4355__s-C1/P-hawaii-X0_M-YSBC_V-u__m-4.3.txt

... So I copied these files and I installed fedora.

From Fedora:

I pasted these files as such:

C-4355__s-C1/hawaii.trx -> /lib/firmware/brcm/brcmfmac4355-pcie.bin
C-4355__s-C1/hawaii-X0.clmb -> /lib/firmware/brcm/brcmfmac4355-pcie.clm_blob
C-4355__s-C1/P-hawaii-X0_M-YSBC_V-u__m-4.3.txt -> "/lib/firmware/brcm/brcmfmac4355-pcie.Apple Inc.-MacBookAir8,2.txt"

I definitely did not make any typos when I plugged this in. The wifi does not come online. Did I miss any steps? Any help would be greatly appreciated

ktb92677, 2020-01-07:
Hey @jhohisel you seemed to have success getting the wifi drivers working on a MacBookAir8,2. Any chance you can see anything I am doing wrong with my configuration?

ktb92677, 2020-01-09:
Quick follow up, wanted to also include @ppaulweber into this conversation as you also seemed to have success at loading brcmfmac4355 successfully. Any chance you can think of what I am doing wrong? Also take a look at this thread: https://github.com/mikeeq/mbp-fedora-kernel/issues/3#issuecomment-572193109

ktb92677, 2020-01-11:
Hey @aunali1 I think it might be time to remove the check mark from MacBookAir8,y... I don't think it is loading properly at this point and the only person who claims to have gotten it to work properly has seemingly disappeared. @jhohisel

aunali1, 2020-01-11:
@ktb92677 Can you try using my kernel patches to see if that makes a difference? I also have lingering suspicions that certain revised models have entirely different FW (Ex. 15,1 2018 uses `kauai` whereas 15,1 2019 uses `midway`). Not exactly sure how to proceed, we might have to add functionality found in Apple's BCMWLAN driver that is not present in `brcmfmac`.

ktb92677, 2020-01-11:
@aunali1 Is this the patch you are referring to? https://github.com/aunali1/linux-mbp-arch/blob/master/wifi.patch Because if so it is nearly identical to the patch I am already using: https://github.com/mikeeq/mbp-fedora-kernel/blob/v5.4-f31/patches/wifi.patch

aunali1, 2020-01-12:
@yurnam Others have had success using the firmware archive from macOS Mojave I have hosted here: [https://packages.aunali1.com/apple/wifi-fw/18G2022](https://packages.aunali1.com/apple/wifi-fw/18G2022)

yurnam, 2020-01-12:
Cool, thanks, I'll try this now :)

yurnam, 2020-01-12:
Sadly, no luck with the FW from Mojave.
but with rambase 0x160000 i don't get a system freeze anymore, just a
FW failed to initalize like before.

same with all the rambase addresses (0x160000, 0x180000, 0x198000, 0x200000)

yurnam, 2020-01-13:
Update:
With Rambase address 0x170000 I get:
`# modprobe brcmfmac
$ dmesg -w

[  237.522523] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  237.522689] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  237.545236] brcmfmac: brcmfmac_module_init No platform data available.
[  237.545247] brcmfmac: brcmf_usb_register Enter
[  237.545265] usbcore: registered new interface driver brcmfmac
[  237.545265] brcmfmac: brcmf_pcie_register Enter
[  237.545283] brcmfmac: brcmf_pcie_probe Enter 14e4:4488
[  237.545580] brcmfmac: brcmf_pcie_get_resource Phys addr : reg space = 000000003b8600be base addr 0x00000091800000
[  237.545581] brcmfmac: brcmf_pcie_get_resource Phys addr : mem space = 00000000d6c2949d base addr 0x00000091000000 size 0x400000
[  237.545696] brcmfmac: brcmf_chip_recognition found AXI chip: BCM4377/4
[  237.546156] brcmfmac: brcmf_chip_cores_check  [1 ] core 0x800:64 base 0x18000000 wrap 0x18100000
[  237.546157] brcmfmac: brcmf_chip_cores_check  [2 ] core 0x812:82 base 0x18001000 wrap 0x18101000
[  237.546158] brcmfmac: brcmf_chip_cores_check  [3 ] core 0x83e:15 base 0x18002000 wrap 0x18102000
[  237.546158] brcmfmac: brcmf_chip_cores_check  [4 ] core 0x83c:64 base 0x18003000 wrap 0x18103000
[  237.546159] brcmfmac: brcmf_chip_cores_check  [5 ] core 0x812:82 base 0x18004000 wrap 0x18104000
[  237.546160] brcmfmac: brcmf_chip_cores_check  [6 ] core 0x844:2  base 0x18005000 wrap 0x18105000
[  237.546161] brcmfmac: brcmf_chip_cores_check  [7 ] core 0x840:15 base 0x18010000 wrap 0x00000000
[  237.546161] brcmfmac: brcmf_chip_cores_check  [8 ] core 0x827:35 base 0x18012000 wrap 0x00000000
[  237.546162] brcmfmac: brcmf_chip_cores_check  [9 ] core 0x240:0  base 0x00000000 wrap 0x1810c000
[  237.546164] brcmfmac: brcmf_chip_cores_check  [10] core 0x135:0  base 0x00000000 wrap 0x18112000
[  237.546164] brcmfmac: brcmf_chip_cores_check  [11] core 0x135:0  base 0x00000000 wrap 0x18113000
[  237.546165] brcmfmac: brcmf_chip_set_passive Enter
[  237.653024] brcmfmac: brcmf_chip_set_passive Enter
[  237.653801] brcmfmac: brcmf_chip_get_raminfo RAM: base=0x170000 size=1638400 (0x190000) sr=0 (0x0)
[  237.653815] brcmfmac: brcmf_chip_setup ccrev=64, pmurev=35, pmucaps=0x84565f23
[  237.653820] brcmfmac: brcmf_get_module_param Enter, bus=2, chip=17271, rev=4
[  237.653828] brcmfmac: brcmf_alloc Enter
[  237.653851] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4377-pcie for chip BCM4377/4
[  237.653859] brcmfmac: brcmf_fw_get_firmwares enter: dev=0000:01:00.0
[  237.654720] brcmfmac: brcmf_fw_complete_request firmware brcm/brcmfmac4377-pcie.bin found
[  237.654757] brcmfmac: brcmf_fw_complete_request firmware brcm/brcmfmac4377-pcie.txt found
[  237.654759] brcmfmac: brcmf_fw_request_nvram_done enter: dev=0000:01:00.0
[  237.655106] brcmfmac: brcmf_fw_request_nvram_done nvram 000000003dcb3447 len 4940
[  237.655541] brcmfmac: brcmf_chip_get_raminfo RAM: base=0x170000 size=1638400 (0x190000) sr=0 (0x0)
[  237.655544] brcmfmac: brcmf_pcie_download_fw_nvram Halt ARM.
[  237.655546] brcmfmac: brcmf_pcie_download_fw_nvram Download FW brcm/brcmfmac4377-pcie.bin
[  237.799783] brcmfmac: brcmf_pcie_download_fw_nvram Download NVRAM brcm/brcmfmac4377-pcie.txt
[  237.801348] brcmfmac: brcmf_pcie_download_fw_nvram Bring ARM in running state
[  237.801348] brcmfmac: brcmf_chip_set_active Enter
[  237.801828] brcmfmac: brcmf_pcie_download_fw_nvram Wait for FW init
[  238.196474] brcmfmac: brcmf_pcie_download_fw_nvram Shared RAM addr: 0x001f5de0
[  238.196591] brcmfmac: brcmf_pcie_init_share_ram_info PCIe protocol version 7
[  238.196603] brcmfmac: brcmf_pcie_init_share_ram_info max rx buf post 511, rx dataoffset 0
[  238.196612] brcmfmac: brcmf_pcie_bus_console_init Console: base 2bb850, buf 268650, size 8192
[  238.196658] brcmfmac: brcmf_pcie_init_ringbuffers Using host memory indices
[  238.196706] brcmfmac: brcmf_pcie_init_ringbuffers Nr of flowrings is 40
[  238.196782] brcmfmac: brcmf_pcie_request_irq Enter
[  238.197454] brcmfmac: brcmf_attach Enter
[  238.197460] brcmfmac: brcmf_proto_attach Enter
[  238.197816] brcmfmac: brcmf_proto_msgbuf_attach Feeding buffers, rx data 511, rx event 8, rx ioctl resp 8
[  238.199065] brcmfmac: brcmf_pcie_ring_mb_write_wptr W w_ptr 511 (0), ring 1
[  238.199069] brcmfmac: brcmf_pcie_ring_mb_ring_bell RING !
[  238.199101] brcmfmac: brcmf_pcie_ring_mb_write_wptr W w_ptr 8 (0), ring 0
[  238.199103] brcmfmac: brcmf_pcie_ring_mb_ring_bell RING !
[  238.199130] brcmfmac: brcmf_pcie_ring_mb_write_wptr W w_ptr 16 (0), ring 0
[  238.199132] brcmfmac: brcmf_pcie_ring_mb_ring_bell RING !
[  238.199138] brcmfmac: brcmf_fweh_register event handler registered for PSM_WATCHDOG
[  238.199141] brcmfmac: brcmf_bus_started 
[  238.199145] brcmfmac: brcmf_add_if Enter, bsscfgidx=0, ifidx=0
[  238.199148] brcmfmac: brcmf_add_if allocate netdev interface
[  238.199167] brcmfmac: brcmf_add_if  ==== pid:340, if:wlan%d (00:00:00:00:00:00) created ===
[  238.199171] brcmfmac: brcmf_bus_change_state 0 -> 1
[  238.199177] brcmfmac: brcmf_msgbuf_query_dcmd ifidx=0, cmd=262, len=20
[  238.199181] brcmfmac: brcmf_pcie_ring_mb_write_wptr W w_ptr 17 (0), ring 0
[  238.199184] brcmfmac: brcmf_pcie_ring_mb_ring_bell RING !
[  240.409699] ieee80211 phy0: brcmf_msgbuf_query_dcmd: Timeout on response for query command
[  240.409718] brcmfmac: brcmf_fil_cmd_data Failed: error=-5
[  240.409725] brcmfmac: brcmf_fil_iovar_data_get ifidx=0, name=cur_etheraddr, len=6
[... cut ...]
[  240.409745] ieee80211 phy0: brcmf_bus_started: failed: -5
[  240.409762] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-5
[  240.409767] brcmfmac: brcmf_detach Enter
[  240.437471] brcmfmac: brcmf_bus_change_state 1 -> 0
[  240.437485] brcmfmac: brcmf_proto_detach Enter
[  240.437491] brcmfmac: brcmf_proto_msgbuf_detach Enter
[  240.439910] brcmfmac: brcmf_pcie_remove Enter
[  240.439919] brcmfmac: brcmf_detach Enter
[  240.466308] brcmfmac: brcmf_bus_change_state 0 -> 0
[  240.466313] brcmfmac: brcmf_proto_detach Enter`

thomasjenkins, 2020-03-05:
On my 15,2 wifi spins forever looking for networks unless I boot with wifi off in network-manager, connect to my phone for internet using bluetooth, then enable wifi. Once I've connected once to the phone the list of wifi networks to connect to is populated normally and connecting is no problem.

joker5bb, 2020-04-21:
Hello,

I have the 
03:00.0 Network controller [0280]: Broadcom Inc. and subsidiaries BCM4360 802.11ac Wireless Network Adapter [14e4:43a0] (rev 03)

What are the exact macOS instructions to extract firmware and make it a bin file?

joker5bb, 2020-04-23:
```
ioreg -l | grep BCM43                 
    | |   |       |   "IOFirmwareVersion" = "Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1600.4)"
    | |   |       | |   "IO80211DriverVersion" = "Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1600.4)"
```

npx001, 2020-04-26:
about bcm4377 - looking "inside" apple driver.. is almost identical as bcm4357
if something has to be patched/modded should be around linux driver for 4357 i.e  "brcmsmac"

MoonPadUSer, 2020-05-12:
@yurnam I have a MacBookPro15,4 too, and I am trying to get the WiFi to work, did you manage to make it work?

aunali1, 2020-05-12:
@MoonPadUSer What kernel/distro are you testing?

MoonPadUSer, 2020-05-12:
I'm running your linux-mbp-arch kernel on Manjaro KDE 20.0

yurnam, 2020-05-12:
> @yurnam I have a MacBookPro15,4 too, and I am trying to get the WiFi to work, did you manage to make it work?

No, unfortunately not.

[73 more replies]