Dunedan/mbp-2016-linux #113: libinput and right-click dragging

jboyens, 2019-08-17:
I'm sure this is due to either a recent kernel upgrade or a recent libinput upgrade but I've started to have an issue with right click dragging (which I use to resize windows).

Normally, I'll hold the 'command' key down and then two finger drag, but recently that's being seen as a scroll event and not as a pointer move event. My Google-fu has not been great enough to track down how to disable scrolling in this case.

I do have an awful workaround, if you start the click with two fingers and then lift to have only 1 finger down it goes back to the previous behavior. It's a bit maddening and hard to use. I didn't change anything that I know of RE: configuration, so it's "new" behavior.

Any pointers or help would be appreciated.

jboyens, 2019-08-17:
Kernel 5.2.8; libinput 1.14.0

jboyens, 2019-08-17:
Kernel 5.2.8; libinput 1.14

Downgrading to libinput 1.13.4 fixes the issue.

jboyens, 2019-08-17:
Reported: https://gitlab.freedesktop.org/libinput/libinput/issues/343

Dunedan, 2019-08-23:
Seems this got fixed with https://gitlab.freedesktop.org/libinput/libinput/merge_requests/282 for upcoming libinput versions.

Dunedan, 2020-07-22:
@jboyens Can you confirm that this works again with recent libinput versions, so we can close this issue?!

closed 2020-07-22 by jboyens