Dunedan/mbp-2016-linux #114: Cannot detect external NTFS drives

Ninlives, 2019-08-25:
All my devices formatted in NTFS seems not working properly with linux on mbp13,2, but it used to work out of the box.
The situation is, if the usb device is connected before booting into grub, it can be detected and mounted by the system properly, otherwise, nothing happens after plugging the device: `lsusb` , `lsblk` and  `fdisk -l` don't list the device and `dmesg` outputs nothing related to the device. Hotplugging works fine in MacOS so I think there is nothing wrong with the adapters. I'm on 5.2.8 and I've tested on 5.1 and 5.0 with no luck. Has anyone met the same problem? Do you have any ideas?

ClashTheBunny, 2019-09-28:
Have you tried any of the suggestions in https://github.com/Dunedan/mbp-2016-linux/issues/120?

Dunedan, 2020-07-22:
This sounds like a duplicate of #120. Therefore I'm closing this now. If it isn't a duplicate, please drop me a note and I'll reopen it.

closed 2020-07-22 by Dunedan