Dunedan/mbp-2016-linux #115: Question on upgrading Mac OS and Ubuntu on MBP2016 13,1

syangav, 2019-08-30:
Hi, here is the case: 

I already have the two os: macOS Sierra 10.12.6 and Ubuntu 16.04 running smoothly on my MacbookPro 13,1. Now I have two questions: 

First, is it possible to upgrade the macOS Sierra to macOS Mojave while not interfering the Linux system? 

Second, is it possible to upgrade Ubuntu 16.04 to Ubuntu 18.04 while not interfering the macOS system? 

Thanks in advance!

Dunedan, 2019-08-30:
Upgrading Ubuntu should be no problem at all.
If upgrading macOS works flawlessly depends on how you boot Linux. I use the EFI boot manager to load grub and continue the usual way from there. However when upgrading macOS, the boot manager entries get reset, so macOS is booted instead and I have to manually adjust the boot order again once.

As usual: Make backups before touching something which could result in data loss.

Dunedan, 2020-07-22:
I'm closing this now, as there seems nothing left to be discussed so far.

closed 2020-07-22 by Dunedan