Dunedan/mbp-2016-linux #116: Discussion for the iMac19,1

CuriousTommy, 2019-09-01:
I thought I would open up an issue tracker for anyone who also owns or has access an iMac19,1.

Here are some issues that I experienced with this machine:

- Sometimes Linux [fails to startup](https://user-images.githubusercontent.com/3105794/64079606-6604cb00-cc9e-11e9-9aea-21a80da16db5.jpg) (I have to physically unplug the machine to get it working again).
- Built-in/Headphone audio doesn't work; however, audio through an HDMI dongle does work.
- Wifi doesn't work.

Here are the results after I ran the `get-info.sh` script: [iMac19,1.zip](https://github.com/Dunedan/mbp-2016-linux/files/3563723/iMac19.1.zip)

Dunedan, 2019-09-01:
Please consider opening a PR with the results from the `get-info.sh` script, so we can add it to the repository and make it easier searchable for everybody.

Regarding the other issues I suggest checking if the already existing issues apply (e.g. it's likely that Apple used the same wifi chip for multiple products) and if not to open separate issues for each problem, so we can avoid getting a single issue with hundreds of comments for a bunch of unrelated problems (as happened with #71).

CuriousTommy, 2019-09-01:
@Dunedan I can definitely open a PR.

> the results from the get-info.sh script

On a side note, am I suppose to use `sudo` when I run the script, or is it okay to run the script as a normal user?

> Regarding the other issues I suggest checking if the already existing issues apply (e.g. it's likely that Apple used the same wifi chip for multiple products) and if not to open separate issues for each problem,

Okay, I'll do that.

CuriousTommy, 2019-09-01:
I submitted the PR.

closed 2019-09-01 by CuriousTommy

aunali1, 2019-09-02:
@CuriousTommy For wifi u'll need to use a patched kernel with BCM4364 support. The specific patch is  [this](https://github.com/aunali1/linux-mbp-arch/blob/master/wifi.patch).

CuriousTommy, 2019-09-03:
@aunali1 Thanks for linking to the wifi patch. I'll keep an eye on this patch.

mrLSD, 2020-05-23:
@Dunedan I have problem with iMac 2019
`get-info.sh` for Ubuntu 16.04 dump:
[iMac19.1.zip](https://github.com/Dunedan/mbp-2016-linux/files/4671652/iMac19.1.zip)

It's not working:
* Build-in Audio
* WiFi
* Bluetooth

I can't imagine how to fix that.
Especially built-in Audio.

Should I create a new Issue?

It will be great to advise.