Dunedan/mbp-2016-linux #118: Instructions on how to build kernel image with drivers and patches MBP 15,x and MBA 8,x

YungRaj, 2019-09-16:
This thread is dedicated to providing instructions on how to build a kernel image with the drivers and patches needed to run linux on the machines specified in the title. The instructions will include what it is and isn't working, and what is currently being worked on. 

Will provide complete instructions once I myself have gotten a build running. Right now a lot of threads point in the right direction, but are far from complete.

If there are any developers that want to collaborate on reverse engineering Apple's drivers and chip implementations, I am also open to that as well.

Thank you all for your previous work

adamretter, 2019-09-27:
I have a new MBP 15,3. I just tried `Fedora-Workstation-Live-x86_64-Rawhide-20190923.n.1.iso` which has a 5.4.0 Linux kernel build. Unfortunately whilst it boots into the live USB environment, the trackpad and keyboard are unresponsive.

I had hoped that the patch below might work (assuming its included in the Kernel that I have?):
https://github.com/torvalds/linux/commit/038b1a05eae6666b731920e46f47d8e2332e07ff

Any thoughts on how I might get Linux booting on that machine?

GoncaloHit, 2019-09-30:
@adamretter im waiting too, currently on windows, but i think you need to built the modules into the kernel in order to work, im still waiting for when it goes upstream

leifliddy, 2019-09-30:
@adamretter 
Try performing the steps I've outlined here:
https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7#gistcomment-3028336

Please let me know if you run into any issues, and I'll try and replicate them.

yoeldovidcohen, 2019-12-31:
I have a 15.3 macbook pro.
Some live usbs boot without recognizing either the hard drive or the keyboard and mouse
With an arch with a different kernel it douse recoginze it

Dunedan, 2020-02-01:
> Will provide complete instructions once I myself have gotten a build running.

@YungRaj: What's your status with this. So far this issue doesn't provide much benefit.

Dunedan, 2020-07-22:
I'm closing this now for two reasons. First of all how to build the kernel or OS images isn't in the scope of this repository. If you don't know how to achieve that for your Linux distribution, you should it look up somewhere else. Secondly this issue doesn't provide any benefit, because it actually doesn't even include such instructions.

closed 2020-07-22 by Dunedan