Dunedan/mbp-2016-linux #119: 14,3 Battery on/off Charging under heavy load

marc-git, 2019-09-23:
When doing intensive work drawing heavy power on the laptop I get signals that the battery is discharging and then charging again on and off.  I think this can't be too good for the battery.  What is a good way to regulate that better?

Dunedan, 2020-07-22:
It's obvious that the power supply is undersized for the power which can be drawn by your MacBook Pro. It could be that this is just an issue when running Linux as macOS might have measures in place to limit the maximum power draw. Limiting the amount of power drawn is something I'd try to achieve under Linux as well if discharging under load happens frequently. Have you already tried to play around with CPU governor settings to limit the maximum power the CPU can draw?

marc-git, 2020-07-23:
I"m not so sure because it is one of the beefy power supplies and it then charges to full without the computer slowing down. I get repeated messages discharging, fully charged, discharging etc.  I would actually guess that the controller governing charging is responsible. It doesn't happen on Mac but then maybe MacOS just doesn't tell you. I could try by checking the actual power draw with a metered plug.

Dunedan, 2020-07-23:
I assume with "beefy" you hint at the bundled 87W power supply. Even with that power supply I'd expect the MacBook Pro to draw more power under peak load from time to time, as the CPU might consume way more power than its TDP suggests. There is a pretty good article from Anandtech going into the gory details of power consumption of Intel CPUs: https://www.anandtech.com/show/13544/why-intel-processors-draw-more-power-than-expected-tdp-turbo

Regarding the behavior under macOS: Does macOS even indicate the battery is discharging if there is a power supply connected or does it just show a charging battery as soon there is power coming from the power supply? Even if it does, Apple's engineers might have considered that case and might simply suppress messages about minor discharging.

Starting with Catalina, there is also Apple's [Battery Health Management](https://support.apple.com/en-us/HT211094), which will let the battery discharge down to 80% once it reached 100%, to avoid battery wear.

marc-git, 2020-07-23:
Yes that  was the beef I was talking about.  :) 
Still confused if it is related to max charging power, why does it reach full charge again even with the CPU load still maxed out?

Dunedan, 2020-07-23:
I suggest you read the Anandtech article I referenced above, as it explains that whole topic way better than I ever could. :slightly_smiling_face: 

But in short: The frequencies CPUs run at are based on calculations involving various factors on modern CPUs. So the decision whether to boost to a certain frequency is influenced by factors like available power and available thermal headroom. So when running a CPU under full load, it'll jump between different frequencies and therefore power consumption over time, based on its current temperature, especially in a system which is as thermally constraint as the MacBook Pro's.

marc-git, 2020-07-23:
I'm going to see if that is actually happening. I can monitor the clock speeds during a job and have a look if the increasing state of charge only comes when the clock speeds come down.

marc-git, 2020-08-11:
OK you're right. The CPU frequencies jump around even at full load, governor was set to `powersave`. Heat also appears to be related. So answer is that the power draw is simply very very high. Closing.

closed 2020-08-11 by marc-git