Dunedan/mbp-2016-linux #120: 14,3 MBP USB devices not present at boot do not appear in lsusb if inserted later.

marc-git, 2019-09-24:
Running debian from a USB on 5.2 kernel, I can't access a usb drives plugged in after boot. They don't appear in lsusb at all. If I insert at boot, they appear but if unplugged and replugged, no worky.

ClashTheBunny, 2019-09-25:
I experience something similar.  I have a USB-C GigE device that I have to run
`sudo modprobe -r xhci_pci; sudo modprobe xhci_pci` to get working from time to time.  I don't remember when it started, but I think it was somewhere in the 4.19->5->5.2 era.

marc-git, 2019-09-26:
I'm not sure it is a good idea for me to do that, because I'm booting from a usb device. 
Should I raise a kernel bug? Or someone who has an idea about what they are doing?

jboyens, 2019-09-28:
I have a USB-C hub that does this from time to time. I end up running `lsusb -v` as root to get it working again. There seems to be a bug in enumerating the devices on the bus, but I haven't gotten anywhere in trying to diagnose it.

monkcs, 2019-09-28:
Is the hub a usb 3.0 device?

Maybe not related, but I had problem with the usb 3.0 controllers to
autosuspend when on battery power, that resulted in devices not to turn
up until I run "lsusb -v" as root.

I solved that issue by adding a kernel parameter that prevents the usb
controller from autosuspend. This is sure not the best way because of
extra power draw, but it seems to work.

I am running Fedora 30 on a MackbookPro14,1 so it might not work for
you, but maybe a try!

Append or add in /etc/default/grub:
usbcore.autosuspend=-1

Then regenerate grub2 config:
grub2-mkconfig -o /boot/grub2/grub.cfg

marc-git, 2019-09-28:
@jboyens thanks for the tip will check that out.

Dunedan, 2020-07-22:
Any update on this? I never had problems with hotplugging USB-devices.

goa111, 2020-07-22:
Same on kernel 5.7, either usbcore.autosuspend=-1 or pci=nommconf as kernel parameters helps.
With kernel 4.19 it was OK without it.

goa111, 2020-07-23:
It seems that the problem is since 4.20-rc1. 
It happens only on usb super speed devices.
constantly happens while on initramfs shell, in order to enter to initramfs shell you can put break=mount as kernel parameter.

marc-git, 2021-08-11:
`sudo sed -i 's/GRUB_CMDLINE_LINUX_DEFAULT="/&usbcore.autosuspend=-1 /' /etc/default/grub`

disables usb autosuspend and fixes this, or at least is a satisfactory workaround.  I would leave this as closed now.

closed 2021-08-11 by marc-git