Dunedan/mbp-2016-linux #121: Catalina

jhabdas, 2019-10-08:
I'm working on an Arch dual-boot on a DMCtypt partition with Catalina once the download finishes. Assuming T2 disabled etc. I'm interested to hear now that Catalina "runs in its own dedicated system volume" ([as reported](https://www.youtube.com/watch?v=qxdkIcjn260) by 9to5 Mac, 9:20) how we can make Linux on MBP a thing. I'll be attempting this on a 2019.

ClashTheBunny, 2019-10-10:
Apfs volumes are different from partitions.  It's more like btrfs and zfs than a classic FS.

jhabdas, 2019-10-10:
That's fine, I just threw on an ExFat temporarily so I can try booting from USB-C, hopefully it goes off without a hitch like it did on the LUKS partition I used DMCrypt to create on my MacBomb https://habd.as/post/dual-boot-manjaro-mojave-macbook-pro/

Helios747, 2019-10-11:
This repo appears to be about the 2016/7 macbooks, so I'm not sure how relevant running Linux on a 2019 model is here.

If it's anything like the current state of 2016/7 models though, probably pretty miserable. The issue isn't getting it installed and booting.

jhabdas, 2019-10-12:
Found a workaround because I needed Arch. Will post here shortly and will close this issue with my solution.

julian-alarcon, 2019-10-16:
What was your solution @jhabdas ?

jhabdas, 2019-10-17:
Docker inside Hyper. CLI sneak preview info here: https://t.me/afterdarkhugo/13928

jhabdas, 2019-10-18:
@julian-alarcon wip solution https://git.habd.as/comfusion/archuro/src/branch/master/bin/archuro

closed 2019-10-18 by jhabdas