Dunedan/mbp-2016-linux #122: iMac19,1 has very low success rate during boot

CuriousTommy, 2019-10-12:
My Parent's 2019 iMac has a very high failure rate when I try to boot Fedora 30 from my portable drive. It often gets stucks on the loading screen, after I enter my encryption password.

I tried to update the boot arguments to remove quiet mode, but it doesn't seem to provide any useful information (I have to press ESC to get any useful information). 

Here is a screenshot of the issue. It seems to always fail after setting up the backlight.
[image]

CuriousTommy, 2019-10-12:
On rare occasions, the iMac may freeze after boot (fans tend to spin up), but that normally doesn't happen.

Dunedan, 2020-07-22:
Do you have any updates on this issue?

CuriousTommy, 2020-07-23:
From recent testing, it seems to boot fine now (or at least the boot error rate is very low). I'll close this issue.

closed 2020-07-23 by CuriousTommy