Dunedan/mbp-2016-linux #125: Keyboard in Arch Linux intramfs

arteal, 2019-11-16:
Hello,

first of all, thank for great work, I finally managed to try Linux on MBP thanks to this repo!

I have installed Arch Linux with full system encryption, so I need to enter password in initramfs. I understand, that driver for keyboard is `applespi` module, from running system I see : 
`applespi               45056  0
spi_pxa2xx_platform    32768  0
crc16                  16384  3 bluetooth,ext4,applespi`

So I have pulled all 3 modules to initramfs image and ensured it is here : 
`# lsinitcpio /boot/initramfs-linux.img  | grep crc16
usr/lib/modules/5.3.11-arch1-1/kernel/crc16.ko

# lsinitcpio /boot/initramfs-linux.img  | grep spi
usr/lib/modules/5.3.11-arch1-1/kernel/applespi.ko
usr/lib/modules/5.3.11-arch1-1/kernel/spi-pxa2xx-platform.ko
`

But keyboard still works only after booting whole system, there goes interesting parts of my dmesg : 

`[    2.205215] usb 1-2: new full-speed USB device number 2 using xhci_hcd
[    2.359610] usb 1-2: config 1 interface 1 altsetting 0 endpoint 0x3 has wMaxPacketSize 0, skipping
[    2.359614] usb 1-2: config 1 interface 1 altsetting 0 endpoint 0x83 has wMaxPacketSize 0, skipping
[    2.362190] usb 1-2: New USB device found, idVendor=0a12, idProduct=0001, bcdDevice=88.91
[    2.362195] usb 1-2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[    2.362197] usb 1-2: Product: CSR8510 A10
[   18.785195] usb 1-8: new high-speed USB device number 3 using xhci_hcd
[   18.932145] usb 1-8: New USB device found, idVendor=2109, idProduct=2813, bcdDevice=90.11
[   18.932150] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   18.932153] usb 1-8: Product: USB2.0 Hub
[   18.932156] usb 1-8: Manufacturer: VIA Labs, Inc.
[   18.933241] hub 1-8:1.0: USB hub found
[   18.934020] hub 1-8:1.0: 4 ports detected
[   19.251930] usb 1-8.2: new high-speed USB device number 4 using xhci_hcd
[   19.372776] usb 1-8.2: New USB device found, idVendor=0bda, idProduct=5411, bcdDevice= 1.21
[   19.372780] usb 1-8.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   19.372784] usb 1-8.2: Product: 4-Port USB 2.0 Hub
[   19.372786] usb 1-8.2: Manufacturer: Generic
[   19.374431] hub 1-8.2:1.0: USB hub found
[   19.375436] hub 1-8.2:1.0: 4 ports detected
[   19.675240] usb 1-8.2.2: new high-speed USB device number 5 using xhci_hcd
[   19.774728] usb 1-8.2.2: New USB device found, idVendor=05ac, idProduct=1006, bcdDevice=96.15
[   19.774732] usb 1-8.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   19.774735] usb 1-8.2.2: Product: Keyboard Hub
[   19.774738] usb 1-8.2.2: Manufacturer: Apple, Inc.
[   19.774740] usb 1-8.2.2: SerialNumber: 000000000000
[   19.775845] hub 1-8.2.2:1.0: USB hub found
[   19.775934] hub 1-8.2.2:1.0: 3 ports detected
[   19.865188] usb 1-8.2.3: new high-speed USB device number 6 using xhci_hcd
[   19.955970] usb 1-8.2.3: New USB device found, idVendor=0451, idProduct=8043, bcdDevice= 1.00
[   19.955975] usb 1-8.2.3: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[   19.955977] usb 1-8.2.3: SerialNumber: 9C000089BF29
[   19.956970] hub 1-8.2.3:1.0: USB hub found
[   19.957064] hub 1-8.2.3:1.0: 4 ports detected
[   20.055219] usb 1-8.2.2.2: new low-speed USB device number 7 using xhci_hcd
[   20.153052] usb 1-8.2.2.2: New USB device found, idVendor=05ac, idProduct=0250, bcdDevice= 0.74
[   20.153057] usb 1-8.2.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   20.153059] usb 1-8.2.2.2: Product: Apple Keyboard
[   20.153061] usb 1-8.2.2.2: Manufacturer: Apple Inc.
[   20.166848] hidraw: raw HID events driver (C) Jiri Kosina
[   20.173665] usbcore: registered new interface driver usbhid
[   20.173667] usbhid: USB HID core driver
[   20.175809] input: Apple Inc. Apple Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2.2/1-8.2.2.2/1-8.2.2.2:1.0/0003:05AC:0250.0001/input/input4
[   20.232030] apple 0003:05AC:0250.0001: input,hidraw0: USB HID v1.11 Keyboard [Apple Inc. Apple Keyboard] on usb-0000:00:14.0-8.2.2.2/input0
[   20.232404] input: Apple Inc. Apple Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.2/1-8.2.2/1-8.2.2.2/1-8.2.2.2:1.1/0003:05AC:0250.0002/input/input5
[   20.288822] apple 0003:05AC:0250.0002: input,hidraw1: USB HID v1.11 Device [Apple Inc. Apple Keyboard] on usb-0000:00:14.0-8.2.2.2/input1
[   22.788550] usb 1-8.1: new high-speed USB device number 8 using xhci_hcd
[   22.910200] random: fast init done
[   22.910243] usb 1-8.1: New USB device found, idVendor=2109, idProduct=d101, bcdDevice= 3.01
[   22.910248] usb 1-8.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   22.910251] usb 1-8.1: Product: USB Keyboard           
[   22.910254] usb 1-8.1: Manufacturer: VIA Labs, Inc.         
[   22.910257] usb 1-8.1: SerialNumber: 0000000000000001
[   22.918609] hid-generic 0003:2109:D101.0003: hiddev0,hidraw2: USB HID v1.10 Device [VIA Labs, Inc.          USB Keyboard           ] on usb-0000:00:14.0-8.1/input0
[   30.217980] random: crng init done
[   32.257266] EXT4-fs (dm-0): mounted filesystem with ordered data mode. Opts: (null)
[   32.461962] systemd[1]: Successfully credited entropy passed from boot loader.
[   32.463898] systemd[1]: systemd 243.78-2-arch running in system mode. (+PAM +AUDIT -SELINUX -IMA -APPARMOR +SMACK -SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=hybrid)
[   32.478546] systemd[1]: Detected architecture x86-64.
[   32.481260] systemd[1]: Set hostname to <thijs>.
[   32.729739] systemd[1]: Created slice system-getty.slice.
[   32.730321] systemd[1]: Created slice system-systemd\x2dfsck.slice.
[   32.730792] systemd[1]: Created slice User and Session Slice.
[   32.731039] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.
[   32.731258] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[   32.731836] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[   32.732036] systemd[1]: Reached target Local Encrypted Volumes.
[   32.768161] EXT4-fs (dm-0): re-mounted. Opts: (null)
[   32.769827] mc: Linux media interface: v0.10
[   32.788432] videodev: Linux video capture interface: v2.00
[   32.808982] facetimehd: loading out-of-tree module taints kernel.
[   32.809018] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[   32.809771] Found FaceTime HD camera with device id: 1570
[   32.809862] facetimehd 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[   33.025328] audit: type=1130 audit(1573896979.823:2): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-journald comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   33.031990] systemd-journald[404]: Received client request to flush runtime journal.
[   33.050384] audit: type=1130 audit(1573896979.847:3): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-udevd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   33.128237] audit: type=1130 audit(1573896979.923:4): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-journal-flush comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   33.130484] ACPI: AC Adapter [ADP1] (on-line)
[   33.142811] i801_smbus 0000:00:1f.4: SPD Write Disable is set
[   33.142978] i801_smbus 0000:00:1f.4: SMBus using PCI interrupt
[   33.148725] intel-lpss 0000:00:15.0: enabling device (0000 -> 0002)
[   33.159110] idma64 idma64.0: Found Intel integrated DMA 64-bit
[   33.175078] intel-lpss 0000:00:15.2: enabling device (0000 -> 0002)
[   33.175558] idma64 idma64.1: Found Intel integrated DMA 64-bit
[   33.191771] intel-lpss 0000:00:1e.0: enabling device (0000 -> 0002)
[   33.192609] idma64 idma64.2: Found Intel integrated DMA 64-bit
[   33.193203] Linux agpgart interface v0.103
[   33.208373] intel-lpss 0000:00:1e.3: enabling device (0000 -> 0002)
[   33.208738] idma64 idma64.3: Found Intel integrated DMA 64-bit
[   33.212402] input: Apple SPI Keyboard as /devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.3/spi_master/spi2/spi-APP000D:00/input/input6
[   33.217186] input: Apple SPI Touchpad as /devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.3/spi_master/spi2/spi-APP000D:00/input/input7
[   33.220064] applespi spi-APP000D:00: modeswitch done.
`

USB is recognized at right place, then I have attached USB keyboard at 18.785195, I gets correctly recognized at 20.175809, then I have entered password from this keyboard and booting continues (30.217980), but Apple keyboard is initialized on 33.212402.

Do I miss something there? Can somebody help me to analyze which module should be loaded and why? 

Hopefully there is someone with better knowledge than me :) I think this is more kernel related, then Arch Linux related, but if you say it is latter, I will write on Arch Linux forums.

Anyway, thanks for help!

roadrunner2, 2019-11-16:
You are missing the `intel_lpss_pci` module: see the README at https://github.com/roadrunner2/macbook12-spi-driver for details.

arteal, 2019-11-17:
That made a trick! Hope this thread helps somebody else. Thanks for help!

closed 2019-11-17 by arteal

Dunedan, 2019-11-18:
@arteal To improve the situation for future Arch users you could get Arch to include the (nowadays upstreamed) `applespi` driver in the initramfs by default.