Dunedan/mbp-2016-linux #127: Catalina Dual Boot

marc-git, 2019-11-28:
Can anybody report success or failures ? Can we add something to the README about that?

Dunedan, 2019-11-29:
There was already another issue here recently which turned out to be caused by the Catalina upgrade: #126

marc-git, 2019-11-30:
Yes purpose of this thread would be to track that issue.  It isn't clear to me whether Catalina upgrade just breaks the linux setup, or if it is incompatible with linux completely.  My MacOS is constantly reminding me to upgrade. I guess I would like to get an idea of what models this is a problem on, if not all.

devalanche, 2019-12-16:
I started dualbooting with catalina already installed. updated catalina to 15.2 and that didn’t brick anything, even the default start volume kept being the linux partition.
dunno if that helps tho...

marc-git, 2019-12-22:
> I started dualbooting with catalina already installed. updated catalina to 15.2 and that didn’t brick anything, even the default start volume kept being the linux partition.
> dunno if that helps tho...

Yes that helps. Will also try and install it then and see what happens. I run linux of a usb ssd

marc-git, 2019-12-23:
Installed Catalina. Didn't affect my USB drive install of debian sid. 

Answer is Catalina Dual Boot works on 14,3 but install corrupts the installation on the NVMe.

AnilRedshift, 2020-03-15:
I have so far been unable to get any live usb to boot on a 2018 MBP running Catalina. Currently have tried Fedora, Mint, and Ubuntu.

cristianmiranda, 2020-05-11:
> I have so far been unable to get any live usb to boot on a 2018 MBP running Catalina. Currently have tried Fedora, Mint, and Ubuntu.

@AnilRedshift what MBP model do you have exactly. Here are some guides:
* https://github.com/mikeeq/mbp-fedora
* https://github.com/marcosfad/mbp-ubuntu
* https://gist.github.com/TRPB/437f663b545d23cc8a2073253c774be3
* https://geeks-world.github.io/articles/472106/index.html

Dunedan, 2020-07-22:
Is it still up-to-date information that an upgrade to Catalina corrupts an existing Linux installation on the internal SSD? If yes, I'd like to understand how that corruption manifests so we can add an appropriate note to the README.

divad42, 2020-07-22:
The only problem I have when Catalina updates is that I have to reinstall rEFInd.  Otherwise, I don't have any problems, and I'm on the latest version of macOS.  I'm on a mbp 14,3.

marc-git, 2020-07-23:
I am also since running Catalina on 14,3. But I am running linux from a portable drive. I think the original issue was with NVMe drives.  Like @divad42 I have to reinstall ReFind every once in a while. 

I would close this issue now because it has no concrete info.

closed 2020-07-23 by marc-git

Dunedan, 2020-07-23:
The necessity to reinstall rEFInd is probably because macOS ensures it's the first entry in the boot order as part of applying updates. So after applying updates, instead of starting rEFInd, macOS boots instead. While I didn't use rEFInd, I had the same problem with the grub-efi not being the first entry in the boot order anymore after each macOS update. I had to use `efibootmgr` to set grub-efi as default boot entry again in such cases.

While that's annoying, it's technically not a bug, as it's intentional behavior of macOS updates. Just as explanation in case somebody wonders what's happening there.