Dunedan/mbp-2016-linux #128: webcam not properly recognized by any app

devalanche, 2019-12-08:
Macbook Pro 14,3 Ubuntu 19.10 Kernel 5.3.0.24
Cheese says "No Device found", Skype finds a cam but delivers black screen while vlc and guvcview recognize the built-in webcam like they should.
Does anyone have a clue why that is, and how I can solve this issue?

$ v4l2-ctl --list-devices
iBridge: FaceTime HD Camera (Bu (usb-0000:00:14.0-3):
	/dev/video0
	/dev/video1

$ v4l2-ctl --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : iBridge: FaceTime HD Camera (Bu
	Bus info         : usb-0000:00:14.0-3
	Driver version   : 5.3.10
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/720
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 0
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 720
	Default     : Left 0, Top 0, Width 1280, Height 720
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags: 
Selection: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: invalid (1/0)
	Read buffers     : 0

marc-git, 2019-12-10:
you could change the title of your issue, since you say it works like a charm with vlc and guvcview. 
For me it works well with webcamoid. The device being used is /dev/video0 . I think it is a bug with Cheese because even when running `cheese -d /dev/video0` it doesn't work and doesn't say anything about the error
I don't have skype so can't test that. I guess you don't actually need cheese, but you would like your webcam just for skype?

devalanche, 2019-12-10:
yea I would need the cam in skype but also on mattermost/zoom later. so a way to define which device is being used for any app would be perfect. cheese -d /dev/video0 didnt work for me neither.

Dunedan, 2019-12-10:
Zoom worked fine for me with a MacBookPro13,2 the past years.

devalanche, 2019-12-16:
zoom is working as well on mine, got skype managed now with a preload script.
thanx anyone

closed 2019-12-16 by devalanche