Dunedan/mbp-2016-linux #131: BCM43602 is working with MBP14,3 now

jiegec, 2020-01-06:
The bug [193121](https://bugzilla.kernel.org/show_bug.cgi?id=193121#c52) has a new solution for firmware issues. It works in my MBP 14,3 now and seems to be a better solution than hacking the fields of binary blob.

Dunedan, 2020-07-22:
Please note that all workarounds mentioned in the kernel bug report only work for a subset of users, so it's no generic workaround.

marc-git, 2020-07-23:
Can we document that? Which subset is it not working for? At the moment the README says what it has always said:

> 
> The MacBook Pro models without Touch Bar come with a Broadcom Limited BCM4350 802.11ac Wireless Network Adapter which works fine out of the box using the brcmfmac driver (ensure you got the matching firmware package installed).
> 
> The MacBook Pro models with Touch Bar come with a Broadcom Limited BCM43602 802.11ac Wireless LAN SoC (rev 02) which is also supported by brcmfmac, but has several issues rendering it unusable, caused by the available firmware. The issues are caused by failing country detection and are:
> 
>     Only 2.4Ghz APs are shown
>     Connecting to an AP barely works or fails directly
> 
> According to Broadcom releasing a fixed firmware would require verification to ensure that it complies with regulatory limits, which is very unlikely to happen as it wouldn't provide enough return on investment for them (see https://bugzilla.kernel.org/show_bug.cgi?id=193121 for details).
> 
> The MacBookPro16,1 comes with a Broadcom Inc. and subsidiaries BCM4364 802.11ac Wireless Network Adapter (rev 04) which is detected with wifi patches from here but doesn't work as no appropriate firmware is available. Catalina firmware doesn't seem to work here and no pre-Catalina firmware is available for this card.
> 

but on my 14,3 I have had trouble free wifi 5GHz and 2.4Ghz for months. Seems strange not to mention the easy fix people can try quite easily.

Dunedan, 2020-07-23:
> Can we document that? Which subset is it not working for?

For that we'd need to know what subset it is and that's still completely unclear. For some people some workarounds work, for some people they don't. I suggest to continue further discussions about that in #47.

andyholst, 2020-07-26:
> > Can we document that? Which subset is it not working for?
> 
> For that we'd need to know what subset it is and that's still completely unclear. For some people some workarounds work, for some people they don't. I suggest to continue further discussions about that in #47.

We should probably rename title for #47 saying 'WI-FI BCM43602 partially working (MBP models 13,3; 14,3)' since alot of progress has been made over there to avoid confusions and close all other related BCM43602 WI-FI issues.

Dunedan, 2020-07-29:
Agreed. Therefore closing this now.

closed 2020-07-29 by Dunedan