Dunedan/mbp-2016-linux #133: Apple NVMe low-power self-refresh mode

07151129, 2020-01-20:
Some Apple NVMe controllers support low-power self-refresh mode, LPSR. The following information describes the usage of LPSR by IONVMeFamily. Note that it is not derived from any Apple documentation and may be inaccurate. In addition to LPSR, there's also QoS with power budgets features, which is not considered here.

ACPI property "nvme-self-refresh" for the PCI controller indicates whether LPSR is supported for
that device. It appears that LPSR is not supported by S3ELab.

LPSR is entered when the controller has been idle for some time. The idleness period is determined
by the LPSR counter. Let `lpsrCount` denote the current LPSR counter value, `lpsrTimeout` the idle
timer period in seconds. Then, whenever the timer expires, the values are updated as follows:

```
if (lpsrCount >= 9500000) {
    lpsrTimeout = 0;
} else {
    if (RunningOnExternalPower)
        lpsrTimeout = 15;
    else if (lpsrCount <= 6999999)
        lpsrTimeout = 3;
    else if (lpsrCount <= 7999999)
        lpsrTimeout = 5;
    else
        lpsrTimeout = 8;
}

setIdleTimerPeriod(lpsrTimeout);
```

LPSR is prohibited in ACPI S3, S4 states if ACPI property for the controller
`nvme-LPSR-during-S3-S4 != 1`.

Value `lpsrCount` is initialised whenever the Identify command has been completed for the
controller. In order to fetch `lpsrCount`, the following command is used:

```
command.features.opcode = 0xc2; // DebugServiceRead
command.features.dword12 = 0;
command.features.dword13 = 848;

lpsrCount = IssueCommand(command) << 32;

command.features.dword13 = 849;
lpsrCount |= IssueCommand(command);
```

When the timer expires, LPSR entry is initiated:
1. `lpsrCount` is incremented.
2. Submission and completion queues are frozen, and the shutdown notification is sent by writing
`0b11` to `CC.SHN` controller register.
3. CSTS is then polled to ensure completion of shutdown.

Note that for LPSR the reserved bit of `CC.SHN` is used. Otherwise, for a normal shutdown, `CC.SHN`
is set to `0b01`.

IONVMeFamily hardcodes the response timeout for the above procedure to be 30 s for S3ELab, 60 s for S3X, and 5 s otherwise.

leifliddy, 2020-01-21:
There was some discussion regarding LPSR and how it was needed to put the NVMecontroller in sleep mode. However, no solution was ever found. It's believed that this is the reason that suspend/resume does not work on the macbook 12-inch models (8,1 + 9,1 and 10,1)

https://github.com/cb22/macbook12-spi-driver/pull/2
https://github.com/Dunedan/mbp-2016-linux/issues/133

07151129, 2020-01-21:
@leifliddy you're right about feature 0xcc being issued before shutdown, but it is only needed for S3X.

leifliddy, 2020-02-04:
@07151129 I'm not a kernel developer and don't know enough about acpi or nvme to know to implement that. There is however, a lot of interest in getting resume/suspend working on the 
Macbook 12 inch models: 8,1 + 9,1 + 10,1 and Macbook Pro models: 13,1 and 14,1
It sounds like this nvme LPSR problem is at the heart of the issue.
I was thinking about starting a bounty to either solve the suspend/resume issue or to implement the nvme LPSR mode. I just need think about the best approach to entice someone to work on this....