Dunedan/mbp-2016-linux #134: German Keyboard swapped ^ and < keys

marc-git, 2020-01-27:
Anybody have a fix for this? Is there an existing profile? I tried changing the layout from default 105 to Apple Aluminium ISO, also tried changing the variant to German (Macintosh). Something I am missing?

marc-git, 2020-01-27:
some info found here : https://wiki.debian.org/MacBook#Keyboard

christophgysin, 2020-01-27:
Try setting the driver to use the ISO layout:

```
$ echo Y | sudo tee /sys/module/applespi/parameters/iso_layout
```

If that fixes it, you can make it permanent by setting the module parameter in:

```
$ cat /etc/modprobe.d/hid_apple.conf
options hid_apple iso_layout=0
```

roadrunner2, 2020-02-24:
To make the change permanent, I think you really want this:
```sh
$ cat /etc/modprobe.d/applespi.conf
options applespi iso_layout=1
```

marc-git, 2020-02-24:
I actually have no applespi.conf file there. Should I?

Dunedan, 2020-02-24:
The name of the configuration file doesn't matter. What matters is that you apply the `iso_layout=1` option for the `applespi` driver as noted by @roadrunner2 and **not** to the `hid_apple` driver as noted by @christophgysin, if you're running a MacBook Pro with the keyboard attached via SPI.

christophgysin, 2020-02-24:
Correct, my mistake. I actually have them both, hid_apple is used for external USB apple keyboards.

Dunedan, 2020-06-21:
I believe this configuration issue got sorted out, therefore I'm closing this now.

closed 2020-06-21 by Dunedan