Dunedan/mbp-2016-linux #136: MacbookPro14,1 Audio working with workaround

awkimball, 2020-02-08:
github user leifliddy's fix worked for me on 14,1: [](https://github.com/leifliddy/snd_hda_macbookpro)

Arch Linux
Kernel 5.5.2

Volume even works with hardware buttons using KDE Plasma and plasma-pa package on Arch.

alexanderroidlhhs, 2020-02-17:
works for me as well

aranega, 2020-02-17:
@awkimball How did you compile it under Arch? The `install.cirrus.driver.sh` doesn't work out of the box.

alexanderroidlhhs, 2020-02-17:
@aranega did you install all the dependencies, maybe you can provide a log / error message?

aranega, 2020-02-17:
@alexanderroidlhhs Thanks for the answer! The issue is not located in the compilation process (I had to modify the source like it has been proposed here https://github.com/leifliddy/snd_hda_macbookpro/pull/1/files), but the deployment process doesn't work as intended because the `/lib/modules/xxxx/updates` folder doesn't exist on my system. As the deployment script had been written for Ubuntu/Fedora, I'm unsure I can use the same mechanism for Arch.

awkimball, 2020-02-17:
@aranega I had to make sure to install the Arch package `linux-headers` because the README doesn't have instructions for Arch: `pacman -S make wget gcc linux-headers` and then the install script worked for me

aranega, 2020-02-17:
@awkimball Strange, I already have those packages installed on my system. Did you run the script as root? (well probably for the deployment part, but I was unsure about the `/lib/modules/xxxxx/updates`). I will try as root then.

awkimball, 2020-02-17:
@aranega I believe I did run it as root. I think for any directories that didn't exist I manually created them and re-ran the script.

leifliddy, 2020-02-17:
```
update_dir="/lib/modules/$(uname -r)/updates"
...
[[ ! -d $update_dir ]] && mkdir $update_dir
```
If you look at the install script, it will create ```/lib/modules/$(uname -r)/updates``` if it doesn't exist. 
You'll need to run ```install.cirrus.driver.sh``` as ```root``` or with ```sudo```

aranega, 2020-02-17:
@awkimball @leifliddy ... I totally missed that in the script... I was not feeling secure to run the script as root... Thanks a lot!

fiowro, 2020-02-19:
@awkimball @aranega  do you have stereo sound? I have the same distro and the same mbp model, and I've used the same installation script - but there is something wrong with channels in my case. When I check sound in gnome-control-center, the "front left" button gives me sound from the right speaker, and when I click any other buttons ("front right", etc.) I get no sound.

So, is your stereo ok? And if so, haven't you make any tweaks in pulseaudio?

aranega, 2020-02-20:
@fiowro On my side, I did nothing more. I use pulsemixer, don't know if this could simply come from the tool.

awkimball, 2020-02-26:
@fiowro I have the same issues with the sound test on Plasma using the plasma-pa package for pulse audio integration. I'll try test now with pure L and R signals to see if my stereo is working regardless of the sound test and I'll update this comment with the results. Again, I'm on Arch & linux 5.5.2 running KDE Plasma on a MBP 14,1.

comport3, 2020-04-03:
Hmm, I had this working yesterday on Ubuntu 18.04 (and derivative Linux Mint 19.3) yet today after installing Ubuntu 20.04 it doesn't work at all. Are there any debug logs to review to determine the cause? syslog, dmesg, any others ?

awkimball, 2020-04-15:
@comport3 did you reinstall the patch after installing 20.04? A major version update of a distro would overwrite this patch, meaning you would need to re-patch.

Bad-Mother-Fucker, 2020-04-30:
It's working for me too, i'm running kali 2020 on MBP 13'' 2017 no touchbar. Please update status.

MrAsimov, 2020-05-17:
I can confirm this patch works as well for MacBookPro13,3

MrAsimov, 2020-05-17:
Don't forget todo follow ups about your findings at https://bugzilla.kernel.org/show_bug.cgi?id=195671

MrAsimov, 2020-06-08:
The patch has been merged to https://github.com/davidjo/snd_hda_macbookpro and the internal microphone is working as well at least for kernel v5.5. I am running MacBookPro13,3.

MrAsimov, 2020-06-21:
@awkimball @alexanderroidlhhs @leifliddy can you guys provide me the dmesg and hwinfo information for any of the MBP models you run of 14.1, 14.3, 13.1 and MBP 13.2 (13.3 is already provided); I need this so I can update the pull request https://github.com/Dunedan/mbp-2016-linux/pull/149 with related dmesg and hwinfo information for the MBP models 13.1, 13.2, 14.1, 14.3 and any other MBP models that supports the Cirrus 8409 HDA chips by the applied https://github.com/davidjo/snd_hda_macbookpro patch.

andyholst, 2020-07-23:
@awkimball l since the internal audio output  is working for MBP 14,1 and is documented in the README section, time to close this issue?

awkimball, 2020-07-24:
@andyholst Agreed! Sorry for the delay, closing now. Thanks everyone.

closed 2020-07-24 by awkimball