Dunedan/mbp-2016-linux #139: WiFi on 14,3

mzanetti, 2020-03-04:
Hey, just wanted to let you know that *finally* WiFi on the 14,3 seems to work very reliably for me by following the latest instructions on the linked bugzilla ticket.

Just to be clear, I had tried all the things with redducing txpower, modifying the binary firmware and while those things would have an effect, none of them made it actually working good enough for a daily usage except the latest one: https://bugzilla.kernel.org/show_bug.cgi?id=193121#c62

That is:
* download the text file brcmfmac43602-pcie.txt from https://bugzilla.kernel.org/attachment.cgi?id=285753 and open it in a editor
* find out your wifi MAC (ifconfig) and update the entry `macaddr=xx:xx:xx:xx:xx:xx` with the mac
* Set `ccode=X3` to `ccode=0`
* Set `regrev=15` to `regrev=1`

copy the file to /lib/firmware/brcm/ and reboot.

Using this now for a week without a single connection drop, on 5GHz, with copying data at about 13MB/s...

Maybe it's worth updating the content with this...

marc-git, 2020-03-05:
I thought this issue existed already somewhere and was closed.  For info, I did not update the macaddr, i don't think it is used. It works fine for me. Also confused about regrev because I have no problems and did not change anything there. 

The main thing that is confusing me is that although a loooot of people report great results with this, the main page says:

> 

> The MacBook Pro models with Touch Bar come with a Broadcom Limited BCM43602 802.11ac Wireless LAN SoC (rev 02) which is also supported by brcmfmac, but has several issues rendering it unusable, caused by the available firmware. The issues are caused by failing country detection and are:
> 
>     Only 2.4Ghz APs are shown
>     Connecting to an AP barely works or fails directly
> 

Which does not really reflect the state of things.

marc-git, 2020-03-13:
Powertop shows around 14W of power to the Wifi when I am just idling. Any ideas what can be edited to reduce power consumption in that text file? 
iwconfig power settings return` SET failed on device`

andyholst, 2020-07-26:
This issue has made alot of progress over at #47 for MBP models 13,3 and 14,3 involves the wireless BCM43602 firmware.

I recommend that this and other related  BCM43602 issues gets closed to avoid confusion with duplicated issues.

Dunedan, 2020-07-29:
I agree with @andyholst. The discussion regarding the WiFi issues should continue in #47. I'll change its title to make make it clear that it's not just an issue with a single model.

closed 2020-07-29 by Dunedan