Dunedan/mbp-2016-linux #140: NVME on Macbook16,1

lelik76, 2020-03-15:
I'm trying to install ubuntu on a Macbook pro. I can't get even the first step - nvme is not visible at all. Kernel 5.3. So much for "all models working"...

Apple Inc. MacBookPro16,1/Mac-E1008331FDC96864, BIOS 1037.80.53.0.0 (iBridge: 17.16.13050.0.0,0) 01/17/2020

root@ubuntu:/home/ubuntu# lspci -nn
04:00.0 Mass storage controller [0180]: Apple Inc. ANS2 NVMe Controller [106b:2005] (rev 01)

echo 106b 2005 > /sys/bus/pci/drivers/nvme/new_id

[  490.736586] nvme nvme0: pci function 0000:04:00.0
[  490.841673] nvme nvme0: IO queues not created

root@ubuntu:/home/ubuntu# lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0    7:0    0   1.9G  1 loop /rofs
loop1    7:1    0  89.1M  1 loop /snap/core/7917
loop2    7:2    0  54.5M  1 loop /snap/core18/1223
loop3    7:3    0 149.9M  1 loop /snap/gnome-3-28-1804/71
loop4    7:4    0   4.2M  1 loop /snap/gnome-calculator/501
loop5    7:5    0  14.8M  1 loop /snap/gnome-characters/317
loop6    7:6    0   956K  1 loop /snap/gnome-logs/81
loop7    7:7    0  44.2M  1 loop /snap/gtk-common-themes/1353
sda      8:0    1  14.6G  0 disk 
├─sda1   8:1    1   2.3G  0 part /cdrom
├─sda2   8:2    1   3.9M  0 part 
└─sda3   8:3    1  12.3G  0 part /var/crash

sznaider, 2020-03-15:
I think you're gonna need a newer kernel version. Not sure which one exactly, but I myself have been able to see NVME (and install Arch) with kernel 5.5.x. You may be interested in this repository https://github.com/aunali1/linux-mbp-arch, although it's for Arch, not Ubuntu.

lelik76, 2020-03-21:
Thanks a lot for the pointers! I've made some progress, here is the summary if anyoune wants to go this way:

Support for 2005 nvme pci dev was added in 5.4, but that kernel crashes early in boot phase. Stock 5.5.10 "works"! Important part from your link - one has to boot with "nomodeset"  arg.
So I've took Ubuntu 18.4 LTS iso, fused the 5.5 kernel with JLiveCD and got a bootable livecd with gui and nvme! Nothing really works on stock 5.5.10 - besides graphics, nvme and USB 2.0 (ehci) which do work. USB3.0 (xhci), keyboard, hdmi, sound, wifi - nothing works. 
But. Having a working nvme, USB adapter, usb kb & mouse, and usb 2.0 (not 3.0!) network adapter gives a working environment. Ubiquity (ubuntu installer) can be launched, we have kb, mouse, and netk. Grub installation predictably fails, leaving EFI partition in a limbo state - so make sure to backup it before. After install, chroot is available so packages can be customized/updated from livecd.

My next step is to build stock ubuntu 5.5-generic kernel with linux-mbp-arch patches, and backport dkms driver packages to ubuntu. This will probably happen once next ubuntu LTS is released (end of march). Hopefully by this time things will get better by themselves.

Dunedan, 2020-07-22:
I just pushed a commit, clarifying that at least Linux 5.4 is required for NVMe to work in models released since 2018. Therefore I'm closing this issue now.

closed 2020-07-22 by Dunedan