Dunedan/mbp-2016-linux #142: Built-in display doesn't work with AMDGPU on Linux 5.6 Macbook (MacBookPro13,3)

dmarinuswoodwing, 2020-04-29:
When I update to the latest kernel (5.6.6) using Fedora 31 my built-in display shows a (backlighted) black screen.

Could someone please share their experience with a AMDGPU Macbook and linux-5.6?

roadrunner2, 2020-04-29:
Yes, the amdgpu driver apparently got broken in the 5.6 kernel - I noticed because the `/sys/kernel/debug/vgaswitcheroo/switch` entry is missing (which means only the iGPU is being registered - the entry only shows up if there are at least 2 GPU's registered) and therefore I can't power down the dGPU anymore. (Note that I boot with the iGPU enabled and active, so the system would still be usable, just sucking way more power than I'd like).

I haven't had time to file a bug report, but it would be a good idea to do so.

fannullone, 2020-05-01:
I am facing same problem on fc31 after last dnf upgrade which installed kernel-5.6
I can open the bug, but please can you give me some directions as I have never done it before.
I can provide log and such to facilitate the fix

MrAsimov, 2020-05-05:
@fannullone I have same issues, file a bug at https://bugzilla.kernel.org/enter_bug.cgi?product=Drivers and pin point to linux kernel 5.6.x (whatever 'uname -r' gives) for Laptop System: Apple product: MacBookPro13,3 and graphics vendor: Apple driver: amdgpu v: kernel bus ID: 01:00.0 chip ID: 1002:67ef etc

MrAsimov, 2020-05-05:
Linux kernel 5.7.0-050700rc3-generic

My eDP (builtin display) has no modes for 5.6+

xrandr --query | grep '\bconnected\b'

```bash
xrandr: Output eDP is not disconnected but has no modes
eDP connected (normal left inverted right x axis y axis)
DisplayPort-0 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 521mm x 293mm
```

fannullone, 2020-05-05:
opened bug: https://bugzilla.kernel.org/show_bug.cgi?id=207589

meeuw, 2020-05-06:
I've also opened a bug for Fedora: https://bugzilla.redhat.com/show_bug.cgi?id=1829174

meeuw, 2020-05-08:
I've posted a question at amd-gfx@lists.freedesktop.org and got the following reply:
https://lists.freedesktop.org/archives/amd-gfx/2020-May/049363.html

So I'll try to do a bisect in the next days...

meeuw, 2020-05-09:
I've bisected the problem to:

b9f1246df179522bc28fda50b720553c845863db is the first bad commit
commit b9f1246df179522bc28fda50b720553c845863db
Author: Noah Abradjian <noah.abradjian@amd.com>
Date:   Fri Nov 22 16:07:24 2019 -0500

    drm/amd/display: Collapse resource arrays when pipe is disabled
    
    [Why]
    Currently, pipe resources are assigned to an index that matches the pipe position.
    However, if pipe 1 or 2 is disabled, there will be a gap in the arrays which causes a crash when iterating based on pipe_count.
    
    [How]
    Fix resource construct to assign resources to minimum available array index.
    
    Signed-off-by: Noah Abradjian <noah.abradjian@amd.com>
    Reviewed-by: Yongqiang Sun <yongqiang.sun@amd.com>
    Acked-by: Rodrigo Siqueira <Rodrigo.Siqueira@amd.com>
    Signed-off-by: Alex Deucher <alexander.deucher@amd.com>

 .../gpu/drm/amd/display/dc/dcn21/dcn21_resource.c    | 20 ++++++++++----------
 1 file changed, 10 insertions(+), 10 deletions(-)

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=b9f1246df179522bc28fda50b720553c845863db

aunali1, 2020-05-10:
You guys might have luck with [this](https://github.com/aunali1/linux-mbp-arch/blob/master/2001-drm-amd-display-Force-link_rate-as-LINK_RATE_RBR2.patch) patch. Its a quick and dirty hack to force all DPCD link rate values to whats expected by Retina panels. Currently, this will prevent externally attached displays from working.

meeuw, 2020-05-10:
> You guys might have luck with [this](https://github.com/aunali1/linux-mbp-arch/blob/master/2001-drm-amd-display-Force-link_rate-as-LINK_RATE_RBR2.patch) patch. Its a quick and dirty hack to force [...]

Thanks I'll try that! Could you please report your findings at:
https://bugzilla.kernel.org/show_bug.cgi?id=207589

MrAsimov, 2020-05-11:
I have reported my findings with the bisected kernel commits between v5.5.0 and v5.6.0 at https://bugzilla.kernel.org/show_bug.cgi?id=207589 with the related Xorg and dmesg logs.

dmarinuswoodwing, 2020-07-21:
Hi @aunali1 thanks for posting your patch, that seems to fix it on my system!

Updated URL:
https://github.com/aunali1/linux-mbp-arch/blob/master/2001-drm-amd-display-Force-link_rate-as-LINK_RATE_RBR2-fo.patch

Dunedan, 2020-07-22:
Is somebody currently working to get a fix for that upstreamed? What's the status there?

aunali1, 2020-07-22:
@dmarinuswoodwing Thanks for testing! Great to see that it works for other models aswell.

@Dunedan Unfortunately, this patch is not upstream ready since it causes issues with 16inch MBPs. Additional testing is also needed for other models.

dmarinuswoodwing, 2020-07-22:
@aunali1 is there an overview of models that need to be tested? What kind of issues do you have with 16" MBP's?
Is there somewhere a discussion/bug that I can track?

aunali1, 2020-07-22:
It seems that just an hour after I made the last comment, @marcosfad tested it again on his 16inch model and it works without issue!

@dmarinuswoodwing Regarding other models, I would prob say the MacBookPro14,x series.

andyholst, 2020-07-22:
The patch https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=dec9de2ada523b344eb2428abfedf9d6cd0a0029 is included in the upstream as commit 69d3e5a5a66bb59c39f36dcb9cf4e9a4239aa8cd. I am checking out the latest commit from Linux kernel master branch and compiling it and will test it on my Linux Mint distro.

andyholst, 2020-07-22:
The upstream patch included in the 69d3e5a5a66bb59c39f36dcb9cf4e9a4239aa8cd commit didn't work for me for kernel tag version v5.7.0, still none working built-in display for my MBP 13,3. Will retest with the applied patch from https://github.com/aunali1/linux-mbp-arch/blob/master/2001-drm-amd-display-Force-link_rate-as-LINK_RATE_RBR2-fo.patch mentioned in the kernel bug report  https://bugzilla.kernel.org/show_bug.cgi?id=207589

andyholst, 2020-07-23:
I can confirm the built-in display is working for kernel version 5.7.10.

andyholst, 2020-07-23:
@dmarinuswoodwing time to close the issue? :)

andyholst, 2020-07-23:
I have verified that the MBP 13,3 built-in display is working for Linux version 5.8.0-rc6+ from Linux master branch with commit 8c26c87b05323a7ccdc632820b85253e0bf47fd9 (HEAD -> master, origin/master, origin/HEAD)

andyholst, 2020-08-09:
This issue is still open, it should be closed since the fix is in the upstream master branch since kernel v5.7.10.

dmarinuswoodwing, 2020-08-10:
It might make sense to atleast give some notice about what kernels are impacted on https://github.com/Dunedan/mbp-2016-linux#amd before closing this issue.

andyholst, 2020-08-10:
@dmarinuswoodwing after bisecting the main source tree, the display stopped working for me on commit 4a8ca46bae8affba063aabac85a0b1401ba810a3

So just need to figure out what kernel versions are tagged between this bad commit and the good commit dec9de2ada523b344eb2428abfedf9d6cd0a0029 

`git describe --contains 4a8ca46bae8affba063aabac85a0b1401ba810a3` gives me kernel version **v5.6-rc1~34^2~16^2~161**
`git describe --contains 473e3f7720f652716eb7a5b8f49628d6f66b1e87`gives me kernel version **v5.7-rc1~23^2~16^2~39**

So built-in display stopped working for kernel version 5.6 and started to work again for version 5.7 (since they include the release candidates tags)

Dunedan, 2020-08-10:
Does one of you want to open a PR for that? If not I can do it as well.
I suggest not to add that information to the graphics section, as it only affected a single kernel version which is already outdated again, but to the kernel history section: https://github.com/Dunedan/mbp-2016-linux#history-of-improvements-per-kernel-version

andyholst, 2020-08-10:
I can do it to keep track of history kernel changes that have made impact of the built-in display for MBP model 13,3 unless anyone is up for it. Just be sure to involve as reviewers before merge.

andyholst, 2020-08-12:
I will test Linux kernel v5.7 from upstream and report my findings for commit 3d77e6a8804abcc0504c904bd6e5cdf3a5cf8162 (HEAD, tag: v5.7) and if it is working a new PR if none else has done it and update the README.md accordingly.

andyholst, 2020-08-12:
Ok, after a few testing, the kernel version 5.7.8 was the one where the built-in display started to work again for MBP model 13,3. Making a PR.

andyholst, 2020-08-13:
It seems like commit 639e0db2d70fb84833d96e782cc4a01825e03b13 is actually the commit that fixes the built-in display issue for MBP model 13,3.

`git tag --contains 639e0db2d70fb84833d96e782cc4a01825e03b13` gives 

v5.8
v5.8-rc1
v5.8-rc2
v5.8-rc3
v5.8-rc4
v5.8-rc5
v5.8-rc6
v5.8-rc7

`commit 639e0db2d70fb84833d96e782cc4a01825e03b13
Author: Mario Kleiner <mario.kleiner.de@gmail.com>
Date:   Mon Mar 16 05:23:40 2020 +0100

    drm/i915/dp: Add dpcd link_rate quirk for Apple 15" MBP 2017 (v3)
    
    This fixes a problem found on the MacBookPro 2017 Retina panel.
    
    The panel reports 10 bpc color depth in its EDID, and the
    firmware chooses link settings at boot which support enough
    bandwidth for 10 bpc (324000 kbit/sec = multiplier 0xc),
    but the DP_MAX_LINK_RATE dpcd register only reports
    2.7 Gbps (multiplier value 0xa) as possible, in direct
    contradiction of what the firmware successfully set up.
    
    This restricts the panel to 8 bpc, not providing the full
    color depth of the panel.
    
    This patch adds a quirk specific to the MBP 2017 15" Retina
    panel to add the additiional 324000 kbps link rate during
    edp setup.
    
    Link to previous discussion of a different attempted fix
    with Ville and Jani:
    
    https://patchwork.kernel.org/patch/11325935/
    
    v2: Follow Jani's proposal of defining quirk_rates[] instead
        of just appending 324000. This for better clarity.
    
    v3: Rebased onto current drm-tip, as of 16-March-2020. Adapt
        to new edid_quirks parameter of drm_dp_has_quirk().
    
    Signed-off-by: Mario Kleiner <mario.kleiner.de@gmail.com>
    Tested-by: Mario Kleiner <mario.kleiner.de@gmail.com>
    Cc: Jani Nikula <jani.nikula@intel.com>
    Signed-off-by: Jani Nikula <jani.nikula@intel.com>
    Link: https://patchwork.freedesktop.org/patch/msgid/20200316042340.4783-1-mario.kleiner.de@gmail.com
`

So the kernel version 5.7.10 I am currently running must include that patch somehow.

Will try compiling that specific commit and the one before.

andyholst, 2020-08-13:
@Dunedan I created a PR https://github.com/Dunedan/mbp-2016-linux/pull/157 with proper README.md documentation about the MBP 13,3 built-in display kernel issue.

andyholst, 2020-08-13:
@fannullone you should close the bug report at https://bugzilla.kernel.org/show_bug.cgi?id=207589

cristianmiranda, 2020-08-15:
I'm on a MacBook Pro 13,3 and can confirm that built-in display works fine on `5.8.0-050800-generic`.

@andyholst quick question, I can no longer run `sudo modprobe amdgpu`. I got the following when installing the new kernel (I was on 5.0.0 and amdgpu worked fine. I loaded it to switch the dedicated gpu off using `echo OFF > /sys/kernel/debug/vgaswitcheroo/switch`, but apparently on 5.8 there's no vgaswitcheroo anymore).

I don't understand what `Possible missing firmware /lib/firmware/` means. Any ideas?.
Thank you for all the hard work so far!

```
DKMS: install completed.
   ...done.
Setting up linux-modules-5.8.0-050800-generic (5.8.0-050800.202008022230) ...
Setting up linux-image-unsigned-5.8.0-050800-generic (5.8.0-050800.202008022230) ...
I: /vmlinuz.old is now a symlink to boot/vmlinuz-5.0.0-32-generic
I: /vmlinuz is now a symlink to boot/vmlinuz-5.8.0-050800-generic
I: /initrd.img is now a symlink to boot/initrd.img-5.8.0-050800-generic
Processing triggers for linux-image-unsigned-5.8.0-050800-generic (5.8.0-050800.202008022230) ...
/etc/kernel/postinst.d/dkms:
 * dkms: running auto installation service for kernel 5.8.0-050800-generic
   ...done.
/etc/kernel/postinst.d/initramfs-tools:
update-initramfs: Generating /boot/initrd.img-5.8.0-050800-generic
W: Possible missing firmware /lib/firmware/amdgpu/navi12_gpu_info.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_gpu_info.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi10_gpu_info.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_gpu_info.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/arcturus_gpu_info.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/raven_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/raven2_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/picasso_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/arcturus_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/arcturus_asd.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/arcturus_sos.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_asd.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_sos.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_asd.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_sos.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi10_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi10_asd.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi10_sos.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/vega20_ta.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_asd.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_mec2.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_me.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_pfp.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/renoir_ce.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/arcturus_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/arcturus_mec2.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/arcturus_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/raven_kicker_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_mec2.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_me.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_pfp.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi12_ce.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_rlc.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_mec2.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_mec.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_me.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_pfp.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_ce.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_mec2_wks.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_mec_wks.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_me_wks.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_pfp_wks.bin for module amdgpu
W: Possible missing firmware /lib/firmware/amdgpu/navi14_ce_wks.bin for module amdgpu
[... 25 lines cut ...]
[... cut ...]
W: Possible missing firmware /lib/firmware/i915/ehl_guc_33.0.4.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_huc_7.0.12.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_guc_35.2.0.bin for module i915
W: Possible missing firmware /lib/firmware/i915/icl_dmc_ver1_09.bin for module i915
W: Possible missing firmware /lib/firmware/i915/tgl_dmc_ver2_06.bin for module i915
I: The initramfs will attempt to resume from /dev/nvme0n1p3
I: (UUID=d6cab994-b144-43a3-a412-100ce59aa599)
I: Set the RESUME variable to override this.
```

andyholst, 2020-08-15:
@cristianmiranda create a new issue of the details you are having problem with the amdgpu driver. I am currently running Linux kernel version 5.7.14.

cristianmiranda, 2020-08-15:
@andyholst You're right. I've opened a new issue for this. Thanks. https://github.com/Dunedan/mbp-2016-linux/issues/159

andyholst, 2020-09-06:
@Dunedan time to finally close this issue?

Dunedan, 2020-09-07:
Yes :+1:

closed 2020-09-07 by Dunedan