Dunedan/mbp-2016-linux #143: WIFI is extremely slow when Bluetooth is on Ubuntu 20.04

amerhaddadeh, 2020-05-16:
Hello Everyone,

since i have upgraded to Ubuntu 20.04 i started facing an issue with the WIFI when the Bluetooth in on, i can hear some chopping seems like both are working in the same channel or smth.

i have tried to do set the co-existence to N using this:

`echo "options iwlwifi bt_coex_active=N" | sudo tee -a /etc/modprobe.d/iwlwifi.conf`

now this is the output when i `cat` that `iwlwifi.conf` file:

_`options iwlwifi 11n_disable=1 bt_coex_active=1 power_save=0 auto_agg=0 swcrypto=1`_

lastly, i tried to remove the `backport-iwlwifi-dkms` package using this:

`sudo apt remove backport-iwlwifi-dkms`

**info** **might** **help**:

amer@amer:~$ inxi -Nn

Network:
  Device-1: Realtek RTL8723BE PCIe Wireless Network Adapter 
  driver: rtl8723be 
  IF: wlo1 state: up mac: 38:b1:db:3f:ad:73 
  Device-2: Realtek RTL810xE PCI Express Fast Ethernet 
  driver: r8169 
  IF: eno1 state: down mac: 8c:dc:d4:72:f0:02 

**kernel** **version**: 

amer@amer:~$ uname -sr
Linux 5.6.13-050613-generic

Any thoughts or suggestions?

jackweller, 2020-05-29:
Wow, I remember a couple years ago I spent weeks, and weeks, and weeks, trying to work out how to fix wifi on an old HP laptop with the same wireless card as you. Here's what I found worked every time:

`sudo modprobe -r rtl8723be`
`sudo modprobe rtl8723be ant_sel=2`
`echo "options rtl8723be ant_sel=2" | sudo tee /etc/modprobe.d/rtl8723-ant-sel.confecho `

If this doesn't work, replace ant_sel=2 with ant_sel=1, I hope this was even at all helpful.

Dunedan, 2020-07-19:
Please discuss problems related to specific Linux distributions somewhere more fitting for the distribution. This repository isn't for distribution specific advice, but rather to get the kernel and basic userland utilities to work with the hardware.

closed 2020-07-19 by Dunedan

Maskla-Dev, 2022-07-21:
if someone else are using an intel wireless device (iwlwifi) might works: `sudo iwlwifi ant_sel=2`. Doing some research, it only works when wireless device has double band, first google the product name specs from `lshw -c network`if you are not sure. 
_PD: For nooby linux users (like me), dont worry if you are using kali or ubuntu, debian based distros works simirlarly._