Dunedan/mbp-2016-linux #144: RFC: I can't figure out if MBP 15,4 has bluetooth hardware or there's a SW bug

atodorov, 2020-05-24:
@Dunedan I've previously contributed my HW listing here:
https://github.com/Dunedan/mbp-2016-linux/tree/master/MacBookPro15%2C4

but I don't see bluetooth hardware listed. This website says MacBook Pro 13 inch, 2019 has bluetooth hardware: https://everymac.com/systems/by_capability/macs-with-bluetooth-different-bluetooth-capabilities.html (all models do).

and in README you say:
`Works out of the box ... except for the models without Touch Bar, which still suffers from a bug and needs an additional patch `.

Could it be that I don't see any BT hardware because it isn't detected due to this bug you mention? Or am I having a rare breed of machine which happens to not have BT at all (that would be a first) ? Any ideas how to figure that out ?

Dunedan, 2020-12-08:
There isn't a single MacBook (no matter if regular one, Air or Pro) which doesn't ship without Bluetooth. And I mean that literally: All MacBooks since the very first ones from 2006 contain Bluetooth and Wifi chips. So yours definitely got Bluetooth hardware.

Aside from that I have no idea what could be the issue there. I also couldn't find any mention of the Bluetooth chip in your hardware listing.

Did you for any chance tried using Bluetooth with macOS to verify that it's not a hardware defect?

Dunedan, 2021-05-01:
@atodorov Any update on this?

atodorov, 2021-05-01:
I haven't been digging into this at all. I never got WiFi firmware to work so I started using a USB tethering to my phone for wifi. 

The only piece of info that somebody told me is that in modern hardware all radio is in 1 chip which probably means BT and WiFi are bundled together. But like I said I found a workaround and stopped digging into the matter.

Dunedan, 2021-05-01:
All right, thanks for your response. As nobody else has encountered such a problem with Bluetooth, I'm going to close this issue.

closed 2021-05-01 by Dunedan