Dunedan/mbp-2016-linux #145: MacBookPro13,3 graphics don't work properly in any distro unless 'nomodeset' flag is set

connervieira, 2020-06-13:
(Integrated graphics: Intel HD Graphics 530, Dedicated graphics: Radeon Pro 460)
I've tested in the following distros:
1. Manjaro GNOME
2. Manjaro KDE
3. Pop!_OS
4. Fedora
5. elementaryOS
6. Ubuntu

After booting into the OS or installer, there's a period of anywhere from 10 seconds to 15 minutes where everything works fine. However, after that period, the screen will suddenly become all one color, and the system will crash after 30 seconds or so. The color seems to be somewhat random, but is dependent on what colors are currently on the screen. For example, in Ubuntu, the color is frequently orange, due to the default wallpaper being orange, but can also be black or grey because of the menus.
I've tried downloading/installing drivers through each distro's repositories, via the command line, directly through the manufacturers website, and through various other methods I've found online tutorials. The problem still occurs, and the only way to remedy it is to set the 'nomodeset' flag in GRUB

Dunedan, 2021-05-01:
@connervieira Any updates regarding this issue? Have you figured out what was wrong there?

connervieira, 2021-05-01:
Unfortunately, I was never able to figure out what was causing it. I ended up just biting the bullet and getting a different laptop that was more Linux friendly.

Dunedan, 2021-05-01:
All right. Thanks for reporting back. As you were the only one experiencing this problem, I'm going to close this issue now.

closed 2021-05-01 by Dunedan