Dunedan/mbp-2016-linux #147: Issue running FaceTime HD through the uvcvideo driver for kernel v5.5+?

MrAsimov, 2020-06-14:
FaceTime HD through the uvcvideo driver seems not to be running out of box for kernel v5.5+ for MacBook Pro 13,3. Can anyone else confirm it?

FSMaxB-dooshop, 2020-06-15:
Yes, I can confirm on Kernel 5.6. Although the  iBridge is detected and the uvcvideo module is loaded there still aren't any video devices.

MrAsimov, 2020-06-15:
Any related link how to set it up properly?

MrAsimov, 2020-06-20:
I found out I didn't have the proper APPLE EFI firmware files installed. I followed the steps how to properly copy the APPLE EFI firmware files from https://github.com/Dunedan/mbp-2016-linux/issues/52#issuecomment-364876533

closed 2020-06-20 by MrAsimov

Dunedan, 2020-06-21:
This is also covered in the README itself in the iBridge section: https://github.com/Dunedan/mbp-2016-linux#ibridge

FSMaxB-dooshop, 2020-06-22:
On my system (Kerrnel 5.6.19) iBridge is detected but there is no FacetimeHD camera.

```
$ lsusb | grep iBridge
Bus 001 Device 002: ID 05ac:8600 Apple, Inc. iBridge
```

```
# usb-devices
...

T:  Bus=01 Lev=01 Prnt=01 Port=02 Cnt=02 Dev#=  2 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  3
P:  Vendor=05ac ProdID=8600 Rev=01.01
S:  Manufacturer=Apple Inc.
S:  Product=iBridge
C:  #Ifs= 0 Cfg#= 0 Atr= MxPwr=
I:  If#=0x Alt= 0 #EPs= 0 Cls=() Sub= Prot= Driver=

...
```

```
$ lsmod | grep uvcvideo
uvcvideo              118784  0
videobuf2_vmalloc      20480  1 uvcvideo
videobuf2_v4l2         32768  1 uvcvideo
videobuf2_common       57344  2 videobuf2_v4l2,uvcvideo
videodev              294912  3 videobuf2_v4l2,uvcvideo,videobuf2_common
mc                     65536  5 videodev,snd_usb_audio,videobuf2_v4l2,uvcvideo,videobuf2_common
```

```
$ lsmod | grep apple
applesmc               24576  0
apple_ib_als           20480  0
industrialio_triggered_buffer    16384  1 apple_ib_als
industrialio          102400  3 industrialio_triggered_buffer,kfifo_buf,apple_ib_als
apple_gmux             20480  0
apple_bl               20480  1 apple_gmux
apple_ib_tb            28672  0
apple_ibridge          28672  2 apple_ib_als,apple_ib_tb
hid                   143360  4 usbhid,apple_ib_als,hid_generic,apple_ibridge
applespi               45056  0
crc16                  16384  3 bluetooth,ext4,applespi
```

```
# v4l2-ctl --list-devices
Cannot open device /dev/video0, exiting.
```

FSMaxB-dooshop, 2020-06-22:
Strangely enough, the touchbar works in the initramfs when entering the hard disk password, but the display content vanishes once the system boots.

And I did copy the firmware files.

MrAsimov, 2020-06-22:
@FSMaxB-dooshop I actually had to apply the macbook12 spi driver for it to work for my Debian distro system following the steps at https://github.com/cb22/macbook12-spi-driver#dkms-module-debian--co

FSMaxB-dooshop, 2020-06-25:
That's what I did and the iBride is detected, but Touchbar and Camera don't seem to be detected as devices attached to it.

MrAsimov, 2020-06-26:
@FSMaxB-dooshop  How does your efi APPLE structure look like?

My structure is:

`root@asimov-MacBookPro:/boot/efi/EFI/APPLE# ls -lhR
.:
total 16K
drwx------ 2 root root 4,0K Jun 20 09:59 EMBEDDEDOS
drwx------ 2 root root 4,0K Jun 20 08:52 EXTENSIONS
drwx------ 2 root root 4,0K Jun 20 07:36 FIRMWARE
drwx------ 3 root root 4,0K Jun 20 07:36 UPDATERS

./EMBEDDEDOS:
total 30M
-rwx------ 1 root root  30M Jun 20 09:59 combined.memboot
-rwx------ 1 root root 205K Jun 20 09:59 FDRData
-rwx------ 1 root root  360 Jun 20 09:59 version.plist

./EXTENSIONS:
total 16M
-rwx------ 1 root root 16M Jun 20 08:52 Firmware.scap

./FIRMWARE:
total 8,0M
-rwx------ 1 root root 8,0M Jun 20 07:36 MBP133.fd

./UPDATERS:
total 4,0K
drwx------ 2 root root 4,0K Jun 20 07:36 MULTIUPDATER

./UPDATERS/MULTIUPDATER:
total 932K
-rwx------ 1 root root 108K Jun 20 07:36 flasher_base.smc
-rwx------ 1 root root  52K Jun 20 07:36 flasher_update.smc
-rwx------ 1 root root  47K Jun 20 07:36 Mac-A5C67F76ED83108C.epm
-rwx------ 1 root root 451K Jun 20 07:36 Mac-A5C67F76ED83108C.smc
-rwx------ 1 root root  83K Jun 20 07:36 MultiUpdater.efi
-rwx------ 1 root root 185K Jun 20 07:36 SmcFlasher.efi
root@asimov-MacBookPro:/boot/efi/EFI/APPLE# 
`

FSMaxB-dooshop, 2020-06-26:
```
APPLE/
├── EMBEDDEDOS
│   ├── combined.memboot
│   ├── FDRData
│   └── version.plist
├── EXTENSIONS
│   └── Firmware.scap
├── FIRMWARE
│   └── MBP133.fd
└── UPDATERS
    └── MULTIUPDATER
        ├── Mac-A5C67F76ED83108C-B0_13.36.0-C0_41.1-R0.bin
        ├── Mac-A5C67F76ED83108C-B0_13.36.0-C0_41.1-R1.bin
        ├── MultiUpdater.efi
        └── ThorUtil.efi
```

MrAsimov, 2020-06-26:
@FSMaxB-dooshop the only thing I can think of is that you are running to 'old' firmware which is causing some hiccups when the ibridge drivers on the kernel is talking to the ibridge device.

Dunno how easy it is for to 'update' your firmwares, but I am willing to share my firmware if you want to test them out.

FSMaxB-dooshop, 2020-06-26:
It's worth a try. But I still have macOS on there, maybe that can be used for the update?

MrAsimov, 2020-06-26:
@FSMaxB-dooshop try updating your firmware with latest Catalina OS version to be able to get the latest firmware files for MBP 13,3 and you should have the same list of files like I have in /dev/nvme0n1p1 partition for the /efi directory and copy thos to your /boot/efi/EFI structure.