Dunedan/mbp-2016-linux #152: Audio works with workaround on MacBookPro14,3

vxcamiloxv, 2020-06-19:
Audio works in speakers (with a little noise after in a high volume  level) and headphones (perfect) on Debian with kernel ´5.6.0-2-amd64´ and this patch ttps://github.com/davidjo/snd_hda_macbookpro on MacBookPro14,3 (2018)

treymerkley, 2020-06-19:
Try restarting pulse, that's worked for me for getting crackling fixed on Manjaro with KDE5

MrAsimov, 2020-06-21:
@treymerkley @vxcamiloxv  can you guys provide me the hwinfo and dmesg and arecord information of MBP14,3 model so I can update the pull request over at https://github.com/Dunedan/mbp-2016-linux/pull/149 ?

Thanks

vxcamiloxv, 2020-06-23:
mine [dmesg](https://termbin.com/jvazl) and [hwinfo](https://termbin.com/nlb2)

MrAsimov, 2020-06-24:
@vxcamiloxv  Thank you, it has already been added over at the pull request https://github.com/Dunedan/mbp-2016-linux/pull/149

andyholst, 2020-07-20:
@vxcamiloxv you should close this issue since the documentation of the README.md has been updated for the audio section.

Dunedan, 2020-07-22:
@vxcamiloxv Did you manage to get the microphone to work in the meanwhile? If not, we should probably adjust the README, as it currently states that audio input works for the MacBookPro14,3.

vxcamiloxv, 2020-07-22:
let me check the mic again

andyholst, 2020-07-23:
@vxcamiloxv it has been verified that the mic is working for MBP model 14,4 at https://github.com/davidjo/snd_hda_macbookpro/issues/12 for kernel version 5.5+.

vxcamiloxv, 2020-10-27:
I confirm that the audio/internal mic are working fine with debian 10 - linux kernel 5.9 but with headphones only work audio

andyholst, 2020-10-27:
@vxcamiloxv Right,the audio input/output works just fine for MBP 14,3. You should again close this issue.

closed 2020-10-27 by vxcamiloxv