Dunedan/mbp-2016-linux #155: Bluetooth Problem with Macbook Pro 16,2

kami83, 2020-07-10:
Hi,

i am using a MBP 2020 with Linux.

- Bluetooth is also not working. Has anyone tested this?
   

Thanks a lot.

Cu kami

bepstein, 2020-07-29:
Hi Cu,

I'm also trying to get it working on a MacBookPro 16,2.  Were you able to get the keyboard and trackpad working?  I haven't even gotten to Bluetooth or Wi-Fi at this point.

Thanks,
Brian

kami83, 2020-07-30:
Hi,
yes the keyboard and the trackpad is working with the bcme driver. BT and Wifi ist still open.

Tell me if you find something new.

BG kami

bepstein, 2020-08-03:
Thanks, do you have a link to that bcme driver?  Also, which distribution of Linux are you using?Thanks,Brian

kami83, 2020-08-04:
Hi,

sorry i mean the bce driver from here : https://github.com/MCMrARM/mbp2018-bridge-drv

Cu kami

Dunedan, 2020-09-10:
Any updates regarding Bluetooth on the MacBookPro16,2? Did you manage to get it working?

bepstein, 2020-09-10:
I haven't had a chance to look into it at this point.  Might be a while.

Thanks,
ep

https://github.com/Dunedan/mbp-2016-linux/issues/155#issuecomment-690606674

kami83, 2020-09-11:
Hi, for me, too. Nothing new. Any Help??
Cu kami

snorez, 2020-11-28:
Any updates on MacBookPro16.2?

Redecorating, 2021-10-02:
This patch will fix it https://lore.kernel.org/lkml/20211002032810.3729-1-redecorating@protonmail.com/T/#u

Hopefully it'll make it's way into the upstream kernel soon.

Redecorating, 2021-11-19:
That patch is in 5.16, but it seems like the bluetooth controller is sending junk and deformed packets (could be an issue with the uart), after the first few commands are sent to it.

If you want to see the bluetooth trace, download this, `gunzip` it and use `btmon -r` on it.
[MBP162.btsnoop.gz](https://github.com/Dunedan/mbp-2016-linux/files/7570188/MBP162.btsnoop.gz)

There's thousands of `= Vendor Diagnostic (len 0)` and then various invalid packets, i.e.

```
> ACL Data RX: Handle 16 flags 0x08 dlen 256                             
        invalid packet flags (0x08)
        02 10 80 00 00 00 00 00 c0 0a 50 2e 00 01 02 10  ..........P.....
        80 00 01 02 10 80 00 00 00 00 00 80 08 0a 12 80  ................
        00 00 00 00 b8 51 88 36 00 30 12 f2 1d 80 70 12  .....Q.6.0....p.
        72 07 30 ee 16 f2 07 30 fe 16 fa 07 30 ff 12 fa  r.0....0....0...
        07 10 ff 14 f2 07 10 ff 14 f2 07 30 ff 16 fa 03  ...........0....
        10 ff 14 c2 ca 16 fd 90 0a 08 49 67 fd 33 bf d6  ..........Ig.3..
        1c 80 00 e0 80 1f fa 02 10 d1 04 02 16 80 60 06  ..............`.
        02 17 80 07 c7 c7 c7 50 80 c0 01 08 40 00 00 00  .......P....@...
        39 7c 02 10 80 78 c3 c7 c7 d7 d9 ec 20 02 10 c7  9|...x...... ...
        c3 c3 b3 4f ff ff 00 00 00 00 00 00 6f 8a 06 10  ...O........o...
        80 f8 c3 c3 c3 33 94 50 83 19 0a 40 00 00 00 7c  .....3.P...@...|
        0b 10 b8 00 01 02 10 80 00 01 02 10 80 00 08 fa  ................
        e8 37 3a 10 fc 59 3a 10 fa 00 00 80 29 02 11 80  .7:..Y:.....)...
        00 00 00 00 81 71 02 10 80 78 c3 c7 c7 33 94 50  .....q...x...3.P
        83 1d 0a 40 00 00 00 40 02 10 b8 00 00 00 00 00  ...@...@........
        ff 00 00 00 00 00 00 00 3c 02 10 98 00 80 02 10  ........<.......
```

Redecorating, 2021-11-20:
It was a uart issue, the clock speed was wrong, hence the random data. https://lore.kernel.org/lkml/20211120083312.41230-1-redecorating@protonmail.com/T/#u

Now it's failing to power on the controller:
```
@ MGMT Command: Set Powered (0x0005) plen 1                    
        Powered: Enabled (0x01)
@ MGMT Event: Command Status (0x0002) plen 3                   
      Set Powered (0x0005)
        Status: RFKilled (0x12)
= bluetoothd: Failed to set mode: Blocked through rfkill (0x12)
```

I feel like it might be hard or soft blocked in `rfkill` (can someone check this?). Maybe those acpi methods for bluetooth aren't being used?

Redecorating, 2021-11-20:
I'm told it works with the correct clock speed for uart after unblocking with rfkill and restarting the bluetooth service, so I think this
> ```
> @ MGMT Command: Set Powered (0x0005) plen 1                    
>         Powered: Enabled (0x01)
> @ MGMT Event: Command Status (0x0002) plen 3                   
>       Set Powered (0x0005)
>         Status: RFKilled (0x12)
> = bluetoothd: Failed to set mode: Blocked through rfkill (0x12)
> ```
was just because it was rfkill blocked