Dunedan/mbp-2016-linux #160: Keyboard function during debian installation

marc-git, 2020-08-26:
A couple of bugs seem to be preventing me from reinstalling.  

Grub runs from the usb drive. Keyboard working. 
I can't get enter any options in the startup menu because the options to boot are F10 or ctrl-x but neither of those work. 
Whatever I select from the Grub menu I lose keyboard and touchpad support once I am in.  Kernel is 5.7.0-2
Possibly related to the USB device bug, I can't run a usb keyboard either. No input is detected. 

Anybody have a tip for getting this going?

Sorry for posting an installation related question here but it seemed to me that this was in the spirit of the repository since it is a hardware compatibility bug.

risen, 2020-08-26:
Debian kernels don't enable CONFIG_KEYBOARD_APPLESPI yet.

See: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=943337

marc-git, 2020-08-26:
OK didn't realise that was still open... how silly.

marc-git, 2020-08-26:
Possible work around would be to make a custom live usb with `live-build `

Dunedan, 2020-08-26:
You could try to open a merge request for the Debian linux kernel repository at https://salsa.debian.org/kernel-team/linux to get it included. That way this could get solved once and for all for Debian users.

marc-git, 2020-08-26:
> You could try to open a merge request for the Debian linux kernel repository at https://salsa.debian.org/kernel-team/linux to get it included. That way this could get solved once and for all for Debian users.

Any idea which of these configs they would put it in?
https://salsa.debian.org/kernel-team/linux/-/tree/master/debian/config

marc-git, 2020-08-26:
I guess options would be  here:
https://salsa.debian.org/kernel-team/linux/-/blob/master/debian/config/config
or 
here https://salsa.debian.org/kernel-team/linux/-/blob/master/debian/config/amd64/config
since it is strictly restricted to amd64 machines.

marc-git, 2020-08-26:
https://salsa.debian.org/kernel-team/linux/-/merge_requests/265

marc-git, 2020-09-09:
This will be fixed in 5.8.7-2 
https://salsa.debian.org/kernel-team/linux/-/commit/830bc6d579b9645318ee4605d4aed3d0d39cd58a

Dunedan, 2020-09-10:
@marc-git: Thank you for handling this. This is going to help a lot of people.

marc-git, 2020-09-10:
A bit surprised myself!

marc-git, 2020-10-23:
Might have to reopen this. I tried installing with a daily snapshot of testing and despite the right kernel version, still no keyboard after booting into the installer.

Dunedan, 2020-10-24:
Is `applespi` included in the initial ramdisk? If so, are all other necessary modules as well?

marc-git, 2020-10-25:
there is a trick to check this from within another linux install right?

Dunedan, 2020-10-25:
I did a quick check booting the latest daily snapshot of the installer for Debian testing (from https://cdimage.debian.org/cdimage/daily-builds/daily/arch-latest/amd64/iso-cd/) as a VM to check whether proper kernel and `applespi` are available.
Turns out kernel version is 5.9.1-1 and `applespi` is included as well. So this should be fine. However the `spi_pxa2xx_platform` and `intel_lpss_pci` modules (and `spi_pxa2xx_pci` for the 2015 MacBook) don't seem to be available (see note in README of https://github.com/roadrunner2/macbook12-spi-driver). Those modules are available on an installed system, but apparently not in the installer. As I don't own a compatible MacBook anymore, I can't further test this.

What you can do to verify: Can you complete the installation with an external keyboard and check whether or not the keyboard works afterwards? If it doesn't, does `modprobe applespi` get it to work?

My guess is that the mentioned modules are either not enabled in the kernel config for the installer or aren't included in the initrd of the installer. I'm not sure where to get this information from, but that would probably the next step when the keyboard does work after installation, but not during the installer.

marc-git, 2020-10-26:
cheers. can check again but there is the other problem related to the usb devices which were not being enabled in the installer, so an external keyboard wouldn't work.  
Strangely the other two modules were already included in the kernel before I submitted the merge request to include applespi. I guess they try to reduce the size of the initrd as much as possible. 
Do you know what script is responsible for the daily snapshot build?

Dunedan, 2020-10-26:
> Do you know what script is responsible for the daily snapshot build?

Unfortunately not, otherwise I would've already checked how the config used there looks like.

marc-git, 2020-10-26:
only got this far. Might have to ask on mailing list...
https://salsa.debian.org/installer-team/d-i/-/tree/master/scripts/kernel

marc-git, 2020-11-10:
> What you can do to verify: Can you complete the installation with an external keyboard and check whether or not the keyboard works afterwards? If it doesn't, does `modprobe applespi` get it to work?

external keyboard finally works with this kernel so install was possible and then keyboard/mouse work out of box.

I've mailed the mailing list

Dunedan, 2020-12-08:
Any updates on this?

For reference I believe the mentioned post to a mailing list is this one: https://lists.debian.org/debian-boot/2020/11/msg00142.html

Two days ago the Debian Installer Bullseye Alpha 3 got released (https://lists.debian.org/debian-devel-announce/2020/12/msg00001.html). It'd be great if somebody could check out, if this version of the Debian installer still faces the same problem.

marc-git, 2020-12-14:
> Any updates on this?
>
> For reference I believe the mentioned post to a mailing list is this one: https://lists.debian.org/debian-boot/2020/11/msg00142.html
>
> Two days ago the Debian Installer Bullseye [...]

will check it out. thanks for the heads up

marc-git, 2021-01-20:
Still no go with the latest version...

akda5id, 2021-04-16:
The [live image I tried](https://cdimage.debian.org/cdimage/unofficial/non-free/images-including-firmware/weekly-live-builds/amd64/iso-hybrid/debian-live-testing-amd64-kde+nonfree.iso) from 2021-04-12, keyboard didn't work on my MacBookAir9,1

Now, your page doesn't list anything about MacBookAir9,1, as far as I can see, so perhaps this isn't supported anywhere yet?

I am a linux newb (well, a returnee from so long ago, I am again a newb), but fairly tech savvy otherwise, so if I can help out chasing down info, let me know. External keyboard does work, so I can poke around if you tell me what to look for :)