Dunedan/mbp-2016-linux #162: Macbook 14.1 no output (speakers) sound

plop2020, 2020-09-22:
Hi,

in redme it says: "With the MacBookPro14,1 the internal audio output is working, however the internal audio input is not working"  

I tried so many distros and the speakers are broken on all of them. I tried the latest: Ubuntu, Fedora, pop! OS, Elementary OS, Zorin OS, Linux Lite.. and few other I forgot already. what I'm doing wrong?

marc-git, 2020-09-22:
Hi Plop. Maybe you could indicate if you applied the driver patches for those distros? It sounds like you read the relevant section of the readme that indicates to do that. 
It is not working out of the box for anyone.

plop2020, 2020-09-22:
> Hi Plop. Maybe you could indicate if you applied the driver patches for those distros? It sounds like you read the relevant section of the readme that indicates to do that.
> It is not working out of the box for anyone.

I really don't know why I assumed you are working and patching the main linux kernel so I assumed it should be included!!!! Im tried of my dumb brain at this point! since Im an ignorant person I should I execute the proper command in this link: https://github.com/davidjo/snd_hda_macbookpro ?

marc-git, 2020-09-23:
yes exactly. Maybe close the issue now too?

closed 2020-09-23 by plop2020