Dunedan/mbp-2016-linux #163: Caps Lock LED (and keyboard brightness) stuck after a while (MacBook Pro 14,1)

LapisLazulis, 2020-10-30:
For some reason I can't find any info on this bug, but whatever distro/DE/WM I'm using, for some reason, when booting my computer it's actually working, but after some moment (can be a few minutes or a few hours) the caps lock LED/keyboard brightness do not want to change their setting. For example, about the caps lock LED, even setting  `echo 0 | sudo tee /sys/class/leds/input5\:\:capslock/brightness ` or echo 1 (on both input5/input13) does not work: the values are set but the LED will remain turned on no matter what.

MacBook Pro 14,1, using Manjaro here (but as I said I've had this problem in Fedora too so it's not distro-dependant), i3wm (but this happens in KDE/GNOME too so not WM/DE dependant)

Dunedan, 2020-10-31:
You should probably open an issue for this in issue in the repository related to the input device drivers: https://github.com/roadrunner2/macbook12-spi-driver

LapisLazulis, 2020-10-31:
I had thought of it at first! But it happens even when that driver is not installed via AUR (since the keyboard works fine without it anyway) so it shouldn't be that driver causing the issue, right?

Dunedan, 2020-11-01:
Nowadays the driver is part of the mainline Linux kernel, so there is no need to install it separately. You can check that it's in use by checking if the `applespi` module is loaded. Even though it's in the mainline kernel I believe @roadrunner2 does still maintain it and handles bugs via the mentioned repository above.

LapisLazulis, 2020-11-01:
Oh, I feel dumb suddenly... Well I'm gonna ask it there too, thanks! Should I close the issue then?

Dunedan, 2020-11-05:
Yes, let's close it here for now. If necessary we can re-open this issue later on.

For reference here is a link to the issue in the `applespi` repository: https://github.com/roadrunner2/macbook12-spi-driver/issues/53

closed 2020-11-05 by Dunedan