Dunedan/mbp-2016-linux #166: Any simpler tutorial on applying the quirks?

jaxjexjox, 2020-12-24:
Based on the information located here:

https://gist.github.com/peterychuang/5cf9bf527bc26adef47d714c758a5509

It seems applying the quirks may be currently broken, furthermore it looks kind of difficult to do for mid skill users.
Is there any kind of information on how to do this and if it works currently?

I'm on a 13,1 MacBookPro 2017 A1708 (no Touchbar) but I'm still frequently, accidentally hitting the trackpad.   Would like to stop hitting it mid typing.

I've actually made 3 mistakes typing this post alone due to the touchpad puicking up accidental hits!

Dunedan, 2020-12-30:
There is a simple solution to completely avoid having to apply quirks and that's to upstream them, so they are available out of the box. That's already done for several components (e.g. I did it for the NVMe and Webcam drivers) and also the motivation why I created this repository. For several other components upstreaming necessary quirks isn't done yet, so somebody has to take responsibility and do it. Unfortunately very few people seem to be willing to do that.

marc-git, 2021-01-09:
Do we specifically mean .quirks files here? If so, can we start a list and aim to get them checked off one-by-one ?

jaxjexjox, 2021-01-09:
Sorry it's still not clear, how do I even fix the touch pad issues? Are quirks files working?