Dunedan/mbp-2016-linux #167: MacBookPro 16,1 Touchpad recognised as pointer device

NoaHimesaka1873, 2020-12-30:
I currently installed Artix Linux with many tries, and installing all kernel modules mentioned here makes almost everything working! But I have a one big problem: my touchpad is recognised as pointer device. 
[image]

Dunedan, 2021-05-01:
@NoaHimesaka1873 any updates regarding this problem? Have you managed to get your Touchpad to work correctly?

Redecorating, 2021-05-05:
This patch in linux-mbp-arch fixes that for me. https://github.com/aunali1/linux-mbp-arch/blob/master/4008-HID-apple-Add-support-for-MacBookPro16-1-keyboard-tr.patch