Dunedan/mbp-2016-linux #172: Potential WiFi Fix

sallen881, 2021-05-06:
Hello,
Let me first say thanks for all of the valuable information compiled here. I was doing further digging into the WiFi issue and I came across a potential solution. I am not familiar with github to submit a pull request suggesting changes to this page, but I believe you may be interested in the solution in comment 62 of the bug report you have shared: 
https://bugzilla.kernel.org/show_bug.cgi?id=193121#c62
Someone took the time to investigate the Windows drivers for the BCM43602 and found a text file which modifies the firmware configuration. It is not distributed with the linux firmware, but it appears to function as expected if it is present. This solution is working reliably on my 13,2 MBP running Ubuntu 20.04! 
Hope this helps,
-Shane

marc-git, 2021-05-08:
Hi Sallen, This was also my experience. Major improvement. Looking closer though I realised it doesn't boost both channels. Can you report your experience with networks that are uniquely 2.4GHz and uniquely 5GHz please? I also thought everything was working smoothly but it turned out one of the channels was not working correctly. 
@Dunedan will probably close this because it is already part of an existing issue. Have a look down the list.

marc-git, 2021-05-08:
See here
https://github.com/Dunedan/mbp-2016-linux/issues/47

mindforrest, 2022-05-02:
working on 14,2 as well! 300Mbps down, wow.