Dunedan/mbp-2016-linux #175: 13,2: CPU Throttling

binarious, 2021-05-28:
I suffer from very bad CPU throttling on my MacBook Pro 13,2. When under high load (Jitsi video call, multiple windows open, etc.) then `grep MHz /proc/cpuinfo` outputs the following:
``` 
cpu MHz		: 400.000
cpu MHz		: 399.999
cpu MHz		: 399.999
cpu MHz		: 399.998
``` 

Usually this is at 3500.00. The temperature is still at around 40 °C and therefore no fan is active. The system is unusable at this point. 
I've already tried the following without success:
- https://wiki.archlinux.org/title/CPU_frequency_scaling#intel_pstate
- https://wiki.archlinux.org/title/CPU_frequency_scaling#acpi-cpufreq
- Set `scaling_governor` to `performance` for all CPUs
- Increase minimum MHz via `cpupower-gui`

Setup:
- MacBook Pro 2016 (13,2)
- Linux Mint 20.1
- Kernel 5.4.0-73-generic
- `mbpfan` installed and running

Any ideas?

binarious, 2021-06-02:
Thanks, will try that when the issue comes back. Seems to be working now with https://wiki.archlinux.org/title/CPU_frequency_scaling#x86_energy_perf_policy and by updating thermald from 1.9.1 to 2.4.5.

closed 2021-06-02 by binarious

binarious, 2021-06-23:
Actually my problem reappeared and the `BD PROCHOT` script fixed the issue. Any idea why this is necessary, @Redecorating?

Redecorating, 2021-06-23:
Something triggers prochot (processor hot), and to prevent overheating it forces the cpu frequency to the minimum, but it doesn't reset for some reason, so it gets stuck