Dunedan/mbp-2016-linux #176: CIRRUS CS8409 introduced on kernel 5.13.rc5

sbe-arg, 2021-06-08:
https://github.com/torvalds/linux/commit/6cc7e93f46a5ce9f65ad3c6c6f645f1d831a8fa4
To be tested.

LapisLazulis, 2021-07-05:
So since no one wanted to make the test, I've tried. I've free'd up some space in my MBP, installed once again Manjaro, updated the kernel (to 5.13.0-1, which SHOULD feature the commit in 5.13-rc5), but and it seems that audio still doesn't work. :/ I **didn't have any audio detected even with this kernel.** 

Though, **_I did notice one thing that I don't know if it's linked to this kernel or not._**
After applying the classic leifliddy audio patch (https://github.com/leifliddy/macbook12-audio-driver/issues), of course it worked (because that was the point of the patch, even though it's "subpar quality" apparently but I don't notice it much). However, what's weird is that my microphone actually works now! I have a MBP 14,1, which meant that it was supposed to not work. 
But the major problem is that after hibernation (using the d3cold method in the mbp-2016-linux README) audio output still doesn't work, microphone still detects my voice apparently but audio output, while still detecting an audio device (apparently) can't output sound.

sbe-arg, 2021-10-09:
Another update https://git.kernel.org/pub/scm/linux/kernel/git/tiwai/sound.git/log/?h=for-next&qt=grep&q=cs8409 for 5.15

closed 2022-04-06 by sbe-arg