Dunedan/mbp-2016-linux #178: USB 3 regression

stefand, 2021-07-31:
USB 3 devices are broken for me since kernel commit 90d28fb53d4a51299ff324dede015d5cb11b88a2 [usb: core: reduce power-on-good delay time of root hub] (or rather, 972b4c19f55da91a42682c74634d36bf5ef3061c in the 5.10 stable tree). USB 2 devices work OK. USB 3 devices work OK only if they are connected via a USB 2 wire.

The symptom is that the devices aren't detected at all. Nothing in lsusb, nothing in dmesg. My device is a MacBookPro14,3.

Reverting the above patch makes USB 3 work on Linux 5.10 and 5.13 (and presumably everything in between), at least if they are attached at boot time. I still seem to have problems with USB 3 devices plugged directly into the USB C ports after boot. I still have to investigate if this is a regression too.

Does anyone else see this issue? I cannot rule out a configuration issue on my side at the moment. I did not find similar regression reports caused by this patch via Google.

marc-git, 2021-08-01:
I am also on a 14,3.  I boot from a usb3 drive and suddenly couldn't boot anymore (no crypto drive found). Thanks for localising this!

marc-git, 2021-08-01:
For me the problem with devices after booting required running `sudo lsusb -v `

risen, 2021-08-02:
Are your devices detected after running the following script?

```bash
#!/bin/bash

for i in /sys/bus/usb/devices/usb*
do
  echo "on" > "$i/power/control"
done
```

stefand, 2021-08-03:
Indeed this helps, at least for devices plugged in after boot. And so does usbcore.autosuspend=-1 . I am going to test if this also makes my boot hard disk work without reverting the mentioned kernel patch.

stefand, 2021-08-03:
Yeah disabling autosuspend also fixes boot time device detection with modern kernels. Any idea how to make autosuspend work again?

In the "bad" state the USB port is powered though. Attached hard disks spin up, but aren't detected. The power setting of the device itself doesn't seem to matter, I think it is some hub that matters. If I run your script before attaching the device the device is detected fine when it is attached later.

fortroche, 2022-11-13:
> For me the problem with devices after booting required running `sudo lsusb -v `

It works for me on 13,1

F1nn, 2023-03-15:
ive got the same problem. i was able to get sudo lsusb -v to work once but it doesn't do it anymore. if i leave the usb drive connected and reboot the laptop, it works until the device is removed and re-inserted and then it never detects again until a reboot. have tried a few options for power cycling usb hub devices, etc but can't find anything that works. did anyone ever find a fix for this regression?

chrism9876, 2024-08-21:
I have this same issue 14,1 usb3 device only works in UsB 2 ports. USB 2 devices work in usb 3 ports