Dunedan/mbp-2016-linux #181: HOW DO I NOW IF MY MAC SUPPORTS IT

jpslpy, 2021-09-23:


jpslpy, 2021-09-23:
MY MAC IS A MACBOOK PRO 13 2017

MacBookPro14,2

Apple Internal Keyboard / Trackpad:

  Product ID:	0x0277
  Vendor ID:	0x05ac (Apple Inc.)
  ST Version:	8.96
  MT Version:	4.69
  Serial Number:	FM772320795GJCQAS+ERP
  Location ID:	0x01000000

WIFI FIRMAWARE 
  Firmware Version:	Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.8)

boafur, 2021-09-27:
Follow any normal Linux install guide (use google. the option key is used to access the boot menu). 

Then apply any needed patches for your specific model, looks like you would need to apply the hibernation, and touch bar patch. 

Also keep in mind that touch id, Wi-Fi, your built-in speakers, and microphone all **will not work**.

Dunedan, 2022-10-23:
This repository isn't meant for end user support, therefore I'm closing this issue now.

closed 2022-10-23 by Dunedan