Dunedan/mbp-2016-linux #183: Battery charge/full charge/full charge design issue MBP 14.1

aranega, 2021-11-25:
Hi,

I'm having trouble with the life of my battery in a strange way. I'm currently under kernel 5.15.4 in archlinux, but I had issues like this since few versions of the kernel.

The thing is that `tlp-stat` and `upower` report a full battery while echoing values from the device directly says something a little bit different. For example, I can have 99% battery reported, but in fact it's more close to 53% directly looking at infos in `BAT0` directory.

Currently on my machine:

`tlp-stat` reports this: (notive the `Charge` and `Capacity` which are really not in sync, I don't know if they need to be, but it would be logical to me)
```
+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = bq20z451
/sys/class/power_supply/BAT0/cycle_count                    =    183
/sys/class/power_supply/BAT0/charge_full_design             =   4790 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   2550 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2550 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_control_start_threshold = (not available) 
/sys/class/power_supply/BAT0/charge_control_end_threshold   = (not available) 

Charge                                                      =  100.0 [%]
Capacity                                                    =   53.2 [%]
``` 

`upower` report:
```
  native-path:          BAT0
  vendor:               SMP
  model:                bq20z451
  power supply:         yes
  updated:              jeu. 25 nov. 2021 08:30:29 (1 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               fully-charged
    warning-level:       none
    energy:              29,121 Wh
    energy-empty:        0 Wh
    energy-full:         29,121 Wh
    energy-full-design:  54,7018 Wh
    energy-rate:         3,11766 W
    voltage:             12,594 V
    percentage:          100%
    temperature:         29,3 degrees C
    capacity:            53,2359%
    technology:          lithium-ion
    icon-name:          'battery-full-charged-symbolic'
  History (charge):
    1637825424	100,000	fully-charged
  History (rate):
    1637825424	3,118	fully-charged
```

and the information from the `BAT0` directory:

```
$ cat /sys/class/power_supply/BAT0/capacity
54

$ cat /sys/class/power_supply/BAT0/charge_full 
2550000

$ cat /sys/class/power_supply/BAT0/charge_full_design 
4790000

$ /sys/class/power_supply/BAT0/charge_now
2550000
```

I have the impression that the battery doesn't run at its full design capacity, but I don't know if that's normal. It is possible to force the `charge_full` and `charge_full_design` to be the same? Or it's not a good thing?

Thanks for your help!

sainathmurali, 2022-09-11:
Hi! I have the same model. I'm using ubuntu 22.04 by erasing macosX completely as I have a black screen issue in macosX, but I couldn't get the audio working. I tried the https://github.com/davidjo/snd_hda_macbookpro stuff yet no luck. Is it working in archlinux or is there a workaround to make it work?

WOLFRAZOR, 2023-05-09:
Your battery may need to be replaced (irregular charging and discharging can also lead to battery damage).If the battery still appears after replacement, it may be a system problem.Also try disabling Turbo Boost, as Apple's machine often triggers Turbo Boost on Linux.