Dunedan/mbp-2016-linux #184: Can't get touchbar to work with Ubuntu 21.04 & 21.10 on macbook pro 14,3

ajmaidak, 2021-12-30:
I've been able to get everything reported to work (touchbar, wifi etc) to functional on Ubuntu 20.04 but I've been unable to get the touchbar to work on ubuntu 21.04 or 21.10.  I've been using https://github.com/PatrickVerner/macbook12-spi-driver touchbar-driver-hid-driver branch.  I'm able to get the touchpad driver to build and load into the kernel but none of the sub devices are available.

Has anyone gotten the touchbar to work with later Ubuntus?  Any suggestions on how to troubleshoot the driver?

ajmaidak, 2021-12-31:
Here's the usb-devices output from 21.10

```
T:  Bus=01 Lev=01 Prnt=01 Port=02 Cnt=02 Dev#=  3 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=ef(misc ) Sub=02 Prot=01 MxPS=64 #Cfgs=  3
P:  Vendor=05ac ProdID=8600 Rev=01.01
S:  Manufacturer=Apple Inc.
S:  Product=iBridge
C:  #Ifs= 0 Cfg#= 0 Atr= MxPwr=
cat: '/sys/bus/usb/devices/usb1/1-3/1-*:?.*/bInterfaceNumber': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-3/1-*:?.*/bAlternateSetting': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-3/1-*:?.*/bNumEndpoints': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-3/1-*:?.*/bInterfaceClass': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-3/1-*:?.*/bInterfaceSubClass': No such file or directory
cat: '/sys/bus/usb/devices/usb1/1-3/1-*:?.*/bInterfaceProtocol': No such file or directory
/usr/bin/usb-devices: 79: printf: (none): expected numeric value
I:  If#=0x Alt= 0 #EPs= 0 Cls=() Sub= Prot= Driver=
```
The driver has loaded:

```
root@mbpro:/sys/bus/usb/devices/usb1/1-3# lsmod|grep apple
apple_ib_als           20480  0
industrialio_triggered_buffer    16384  1 apple_ib_als
industrialio           81920  3 industrialio_triggered_buffer,kfifo_buf,apple_ib_als
apple_ib_tb            24576  0
apple_ibridge          24576  2 apple_ib_als,apple_ib_tb
applesmc               20480  0
hid                   135168  3 usbhid,apple_ib_als,apple_ibridge
apple_mfi_fastcharge    20480  0
apple_gmux             20480  0
apple_bl               20480  1 apple_gmux
applespi               49152  0
video                  53248  1 apple_gmux
root@mbpro:/sys/bus/usb/devices/usb1/1-3# modinfo apple_ibridge
filename:       /lib/modules/5.13.0-22-generic/updates/dkms/apple-ibridge.ko
license:        GPL v2
description:    Apple iBridge driver
author:         Ronald Tschalär
srcversion:     77D4887FFE639B57F8F3390
alias:          acpi*:APP7777:*
depends:        hid
retpoline:      Y
name:           apple_ibridge
vermagic:       5.13.0-22-generic SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         mbpro Secure Boot Module Signature key
sig_key:        0F:AE:B3:D8:81:98:48:72:D5:DB:EB:D5:82:9D:E7:40:23:2F:3A:B0
sig_hashalgo:   sha512
signature:      48:98:33:7A:94:6F:95:0C:A6:5A:B3:EF:54:B7:0A:6B:01:BA:E5:26:
		16:15:82:64:C1:D6:EE:19:79:8C:8C:28:CC:CF:13:5F:70:99:39:EE:
		F6:D6:8F:1E:71:24:54:78:7F:41:E7:0C:47:8C:79:F2:9D:4A:EB:A6:
		C1:98:62:D4:45:61:B4:E4:FA:1E:6A:D9:A1:6B:8E:7E:BD:5D:66:B5:
		F7:DC:5F:60:84:FE:0D:47:85:27:19:22:B4:82:E6:B4:86:F3:D3:85:
		3E:D5:E7:76:DD:0D:81:7D:BC:E2:BF:A8:3B:3F:15:EB:11:62:90:23:
		04:A1:33:49:F0:82:4A:B1:DF:0A:F4:08:A3:4A:35:A6:D9:96:65:A9:
		5E:08:AA:2A:49:E1:50:29:B1:DC:76:D1:49:DF:C2:95:E3:89:93:33:
		0F:8A:58:E8:CF:14:4D:DC:45:79:3E:86:92:0B:E1:38:15:D7:E2:2A:
		1F:DA:52:CA:F1:32:6E:11:64:7F:09:CF:6B:EC:89:7D:F9:9B:C8:7C:
		B9:F8:F0:09:EA:A3:8C:34:9E:54:AA:6A:48:8C:61:B5:17:0A:85:5A:
		87:E7:B0:9E:C3:60:AD:47:BD:B7:4F:47:FD:7B:9A:20:DA:43:D4:CB:
		C0:F7:DB:0D:6F:D9:E2:C1:67:3C:EA:19:6D:47:C9:04
root@mbpro:/sys/bus/usb/devices/usb1/1-3# modinfo apple_ib_tb
filename:       /lib/modules/5.13.0-22-generic/updates/dkms/apple-ib-tb.ko
license:        GPL v2
description:    MacBookPro Touch Bar driver
author:         Ronald Tschalär
srcversion:     406D9309CBD9E25BE36EC5F
alias:          platform:apple-ib-tb
depends:        apple-ibridge
retpoline:      Y
name:           apple_ib_tb
vermagic:       5.13.0-22-generic SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         mbpro Secure Boot Module Signature key
sig_key:        0F:AE:B3:D8:81:98:48:72:D5:DB:EB:D5:82:9D:E7:40:23:2F:3A:B0
sig_hashalgo:   sha512
signature:      23:2F:24:EA:87:EF:30:D1:3C:B6:E6:20:41:70:69:BE:36:13:6C:E4:
		CB:60:B5:5E:AD:5F:86:A1:77:55:5F:91:69:ED:B1:A2:47:22:C2:F3:
		37:A3:A0:10:64:25:DE:B2:94:5E:9B:03:C2:77:7D:99:3E:66:A3:BF:
		D3:13:99:CA:49:9E:B0:19:3C:60:59:AC:15:E6:4A:94:10:41:26:40:
		C1:C7:CC:57:22:80:65:1C:91:A4:91:01:85:AA:3F:6B:10:5A:ED:97:
		0D:EA:9B:3E:29:C4:5E:52:F9:20:84:F0:82:C4:0D:24:17:BE:FE:B6:
		17:CC:EA:EA:26:05:3E:94:5D:E3:E9:09:DF:52:BF:1D:F5:75:34:3B:
		77:CF:DA:49:2B:21:27:8F:2E:E0:0D:D1:01:C8:CA:D2:DD:A6:F6:CF:
		21:5C:18:00:2D:6C:3A:3B:73:35:80:5D:C1:91:D0:7C:4E:5F:3E:7F:
		9D:AA:C0:E0:41:13:86:BF:06:DF:B9:59:89:9B:4D:5B:86:EA:96:0D:
		FA:A1:8F:8C:05:8D:B4:FA:6C:57:AB:6C:CF:F7:FA:79:6E:C1:B5:BD:
		23:5D:6D:2F:78:4F:6F:C9:FC:EE:3C:62:E1:18:24:88:A1:55:8D:C1:
		94:79:63:99:7A:B3:7A:1A:19:98:2E:04:BF:5D:B3:A3
parm:           idle_timeout:Default touch bar idle timeout:
    >0 - turn touch bar display off after no keyboard, trackpad, or touch bar input has been received for this many seconds;
         the display will be turned back on as soon as new input is received
     0 - turn touch bar display off (input does not turn it on again)
    -1 - turn touch bar display on (does not turn off automatically)
    -2 - disable touch bar completely (int)
parm:           dim_timeout:Default touch bar dim timeout:
    >0 - dim touch bar display after no keyboard, trackpad, or touch bar input has been received for this many seconds
         the display will be returned to full brightness as soon as new input is received
     0 - dim touch bar display (input does not return it to full brightness)
    -1 - disable timeout (touch bar never dimmed)
    [-2] - calculate timeout based on idle-timeout (int)
parm:           fnmode:Default Fn key mode:
    0 - function-keys only
    [1] - fn key switches from special to function-keys
    2 - inverse of 1
    3 - special keys only (int)
```

truonggiang0710, 2022-03-22:
I have the same issue. and the output of "usb-device" does not any "Vendor=05ac ProdID=8600"

sebastianhaberey, 2022-04-15:
@ajmaidak I have exactly the same issue. I'm on Ubuntu 21.10 (5.13.0.22-generic). Im using the Patrick Verner fork and `lsmod` shows `apple_ibridge`, `apple_ib_tb` and `apple_ib_als` just as yours. Also my `usb-devices` shows the same output for iBridge.

Did you manage to get the touch bar to work in the end?

ajmaidak, 2022-04-18:
Sadly no, I gave up and stayed with the 20.04 Ubuntu.

BrianValente, 2022-04-18:
I fixed it by removing the three occurences of `|5ac/8600/*` in `/usr/lib/udev/rules.d/39-usbmuxd.rules`
The file should be like this:
```bash
# usbmuxd (Apple Mobile Device Muxer listening on /var/run/usbmuxd)

# systemd should receive all events relating to device
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", TAG+="systemd"

# Initialize iOS devices into "deactivated" USB configuration state and activate usbmuxd
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", ENV{SYSTEMD_WANTS}="usbmuxd.service"

# Make sure properties don't get lost when bind action is called
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", ACTION=="bind", ENV{USBMUX_SUPPORTED}="1", OWNER="usbmux", ENV{SYSTEMD_WANTS}="usbmuxd.service"

# Exit usbmuxd when the last device is removed
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*", ACTION=="remove", RUN+="/usr/bin/usbmuxd -x"
```

sebastianhaberey, 2022-04-18:
@BrianValente holy sh*t that did it, thank you so much!

I'd be interested in the reasoning behind it - I am guessing this is excluding the iBridge (05ac/8600) device from being handled by usbmuxd?

ajmaidak, 2022-08-05:
@BrianValente fix works on 21.10 and 22.04.

closed 2022-08-05 by ajmaidak