Dunedan/mbp-2016-linux #189: Patched touchbar driver not working in kernel 6.0

marc-git, 2022-12-20:
The patched touchbar [driver](https://github.com/PatrickVerner/macbook12-spi-driver)  from @PatrickVerner  is no longer working on the latest kernel.  The compile error I get is 
`
applespi.c:1810:38: error: invalid application of 'sizeof' to incomplete type 'struct efivar_entry'`

Confused why this suddenly became a problem. Anyone want to hazard a guess at a solution?

Sjd-Risca, 2023-03-30:
Yes, I can confirm the provided error but it is possible to easily overcome it as described by the following comment: https://github.com/roadrunner2/macbook12-spi-driver/issues/67#issuecomment-1315273486

Still to be published a proper fix upstream to the repository, but at least it is possible quite easily to make it work.

marc-git, 2023-04-10:
yeah that's right. I forked it to remove the saving functionality. [https://github.com/marc-git/macbook12-spi-driver](see here)

seedneus, 2023-07-04:
@marc-git I still can't compile it under 6.3.8-200.fc38...
"Deprecated feature: REMAKE_INITRD (/var/lib/dkms/applespi/0.1/source/dkms.conf)
Creating symlink /var/lib/dkms/applespi/0.1/source -> /usr/src/applespi-0.1

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.3.8-200.fc38.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.3.8-200.fc38.x86_64 (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.

gjvanderheiden, 2023-11-15:
I used the [macbook12-spi-driver-dkms](https://aur.archlinux.org/packages/macbook12-spi-driver-dkms) from the AUR. It uses a patch.

marc-git, 2024-01-28:
> Deprecated feature: REMAKE_INITRD (/var/lib/dkms/applespi/0.1/source/dkms.conf)

I can delete that from dkms.conf but it still won't work.

marc-git, 2024-01-28:
> I used the [macbook12-spi-driver-dkms](https://aur.archlinux.org/packages/macbook12-spi-driver-dkms) from the AUR. It uses a patch.

that patch is not working for me in 6.6.13 
what about you ?

radbirb, 2024-02-04:
patch doesn't work for me on Fedora 39 (6.6.14) or openSUSE Tumbleweed (6.7.2).
make log:
```
Sign command: /lib/modules/6.7.2-1-default/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/apple-ibridge/0.1/source/dkms.conf)

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.7.2-1-default...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.7.2-1-default (x86_64)
Consult /var/lib/dkms/apple-ibridge/0.1/build/make.log for more information.
rad@susietosh:~/macbook12-spi-driver> sudo dkms install -m applespi -v 0.1Sign command: /lib/modules/6.7.2-1-default/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Deprecated feature: REMAKE_INITRD (/var/lib/dkms/applespi/0.1/source/dkms.conf)

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.7.2-1-default...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.7.2-1-default (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.
rad@susietosh:~/macbook12-spi-driver> cat /var/lib/dkms/applespi/0.1/build/make.log 
DKMS make.log for applespi-0.1 for kernel 6.7.2-1-default (x86_64)
Sun Feb  4 04:24:31 AM +03 2024
make -C /lib/modules/6.7.2-1-default/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-6.7.2-1-obj/x86_64/default'
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c: In function ‘appletb_fill_report_info’:
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c:949:9: warning: enumeration value ‘HID_REPORT_TYPES’ not handled in switch [-Wswitch]
  949 |         switch (field->report->type) {
      |         ^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:905:35: error: initialization of ‘void (*)(struct acpi_device *)’ from incompatible pointer type ‘int (*)(struct acpi_device *)’ [-Werror=incompatible-pointer-types]
  905 |                 .remove         = appleib_remove,
      |                                   ^~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:905:35: note: (near initialization for ‘appleib_driver.ops.remove’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-6.7.2-1/scripts/Makefile.build:244: /var/lib/dkms/applespi/0.1/build/apple-ibridge.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-6.7.2-1/Makefile:1928: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: *** [../../../linux-6.7.2-1/Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-6.7.2-1-obj/x86_64/default'
make: *** [Makefile:16: all] Error 2
```
perhaps someone smarter than me would know a fix? in a last ditch effort I tried the t2linux fork (https://github.com/t2linux/apple-ib-drv), and I ran into the same issue.

radbirb, 2024-02-04:
> The build error with .remove is trivial to fix, but for me the touch bar doesn't work with 6.6 regardless. The driver detects it, registers the input device, but the touch bar just stays black. It worked ok on Linux 6.1 with the same driver.
> 
> 
> 
> It randomly worked twice, but I don't understand why and could not reproduce this.
> 
> 
> 
> [fix.txt](https://github.com/Dunedan/mbp-2016-linux/files/14157847/fix.txt)
> 
> 

Could this be due to `usbmuxd` binding to the chip? I've managed to get the touchbar working (thank you for the fix! the AUR package seems to work also) on 6.6.15 (Arch Linux LTS) and 6.7.3 (Arch), but it didn't work until I did the this workaround (see https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-609749027), ran into the same thing on 6.1.0 on Debian... so I'm not sure why it's not working for you, though I've yet to test on Fedora/OpenSUSE.

stefand, 2024-02-04:
I don't have usbmuxd running at all. I am now building a 6.1 kernel again to see if something else changed (like accidental deletion of the firmware from the efi boot partition). If things still work with the old kernel I'll try my luck with a bisect.

stefand, 2024-02-06:
With Linux 6.1.76 the touch bar is working with @marc-git 's driver. With 6.4 and 6.6 it doesn't work most of the time. So I have a baseline for bisecting.

A problem is that in "bad" kernels it is randomly starting up, and I haven't figured out why - I estimate that it works 10% of the time. So if the touch bar starts up I have to reboot a few times to make sure the kernel revision is really working and not randomly (un)lucky.

gjvanderheiden, 2024-02-06:
> Could this be due to `usbmuxd` binding to the chip?

I patched /usr/lib/udev/rules.d/39-usbmuxd.rules according to:
https://github.com/libimobiledevice/usbmuxd/issues/138

Instead of using a new script to fix this issue. That works also. Without the patch it was not lighting up.

> I used the [macbook12-spi-driver-dkms](https://aur.archlinux.org/packages/macbook12-spi-driver-dkms) from the AUR. It uses a patch.
> that patch is not working for me in 6.6.13
> what about you ?

I run Arch Linux, which is shipping the latest kernel version.It's been through server 6.6.x versions. It's on 6.7 by now and haven't had any issues with the TouchBar.

marc-git, 2024-02-06:
> I run Arch Linux, which is shipping the latest kernel version.It's been through server 6.6.x versions. It's on 6.7 by now and haven't had any issues with the TouchBar.

Are you running with just different usbmuxd rules? No home brew spi drivers?

gjvanderheiden, 2024-02-07:
> Are you running with just different usbmuxd rules?
yes.
I used the file content @KiLLeRRaT posted  in https://github.com/libimobiledevice/usbmuxd/issues/138
This patch is supposed to be shipped in a new version of usbmuxd, but the last version is from 2020.

> No home brew spi drivers?
just the one from the AUR package, which has a patch build in. So just `yay -S macbook12-spi-driver-dkms`

I deleted the generated stuff from the aur package and did a makepkg again, no errors. It is up to date, I update my system regularly.

So in order to get it to work I do:

- update file `/usr/lib/udev/rules.d/39-usbmuxd.rules`
- `yay -S macbook12-spi-driver-dkms`
- add the modules to `/etc/mkinitcpio.conf`: `MODULES=(intel_lpss_pci spi_pxa2xx_platform applespi apple-ib-tb)`
- run sudo mkinitcpio -P

gjvanderheiden, 2024-02-08:
@stefand that took some effort 💪already

stefand, 2024-02-10:
So my previous bisect ended up with a non-sensical commit because the touch bar luckily worked 3 times with a "bad" kernel.

On a second attempt I ended up with this commit as breaking the touch bar on my system:

```
e04955db6a7c3fc4a1e6978649b61a6f5f8028e3 is the first bad commit
commit e04955db6a7c3fc4a1e6978649b61a6f5f8028e3
Author: Ronald Tschalär <ronald@innovation.ch>
Date:   Wed Jan 11 10:07:36 2023 +0000

    HID: Recognize sensors with application collections
```

This makes more sense - it talks about T2 macbooks pro. My MacBookPro14,3 has a touch bar, but not a T2 chip. That would also explain why the touch bar driver works fine for some people here.

stefand, 2024-02-12:
Reverting e04955db on a 6.6 kernel makes the touch bar come up reliably. I guess the problem only affects pre T2 macs. Sadly I don't know the USB technical details to figure out how to fix this properly.

Since the touch bar driver still lives out-of-tree it will be tricky to get the kernel devs to fix it. I'll try to get the attention of the t2linux people.

gjvanderheiden, 2024-02-12:
@stefand I have  MacBook 14,3 (A1707). In Mac OS System information, under hardware->Controller I see Apple T1 Security-chip. Firmware 14Y910. The AUR patch I use hasn't been updated, checked that too. 

I do run with the I act like I'm Mac OS trick. Found [here](https://github.com/Dunedan/mbp-2016-linux/issues/6). I use rEFInd to be precise, shouldn't matter. Maybe that matters. 

I boot into Mac OS (Ventura 13.6.4) from time to time too and run the updates in Mac OS. This sometimes updates firmware, could be something too. Though Ventura doesn't receive all the updates anymore.

I might be missing something completely here, but I just don't get why it's working on mine, except I'm on 6.7.4-arch-1 kernel version.

stefand, 2024-02-12:
@gjvanderheiden the touch bar works for me occasionally with an unmodified 6.6 kernel. I don't know if this is runtime luck or has something to do with how the kernel and/or driver are built.

I tried updating to 6.7, but it doesn't change the behavior.

Maybe the presence of usbmuxd influences things in a positive way, at least when the workaround discussed in this bug is used.

marc-git, 2024-02-18:
>  I'll try to get the attention of the t2linux people.
@stefand Any luck with them?

stefand, 2024-02-18:
@marc-git There is no response yet to the bug I filed.

almas, 2024-05-11:
You can try https://github.com/almas/macbook12-spi-driver/tree/touchbar-driver-hid-driver
I have successfully installed this driver with kernel 6.5.0-28-generic.

stefand, 2024-05-11:
@almas I don't think this will help on pre-T2 macs. The changes you made to the module are the already discussed build fix, but I don't see anything that would fix the regression introdced by kernel commit e04955db6a.

almas, 2024-05-11:
@stefand I don't know much about it.
Did you see the BrianValente commits? 
At least Touchbar is turned on and working without problems for me.

marc-git, 2024-06-02:
> @marc-git There is no response yet to the bug I filed.

@stefand Still nothin?

marc-git, 2024-06-02:
> [fix.txt](https://github.com/Dunedan/mbp-2016-linux/files/14157847/fix.txt)

updated my fork again to keep it compiling. Still not working, even with suggestions from @gjvanderheiden to change the `usbmuxd` rules

marc-git, 2024-06-19:
@stefand still no answer on t2linux. I wonder if @roadrunner2 would know what is causing this. My MBP is running slower and slower on the newer OSes so I am switching to Linux more often and its getting on my nerves now. 

What are the options here? I guess:
 - start a t1linux repo and try to draw in efforts there
 - try to merge some fix to the t2linux kernel that detects t1 and leaves it alone? 
 - suffer the pain
 - any more ideas?

stefand, 2024-06-19:
No answer. Sadly I don't have the time to follow up. The only thing I checked is if a 6.8 kernel magically fixed the bug. The result is as expected no, the dark touch bar is not fixed. The commit that causes the regression still reverts cleanly though.

Since the macbook is just a tool to an end I've reverted the commit locally.

radbirb, 2024-07-24:
> @stefand still no answer on t2linux. I wonder if @roadrunner2 would know what is causing this. My MBP is running slower and slower on the newer OSes so I am switching to Linux more often and its [...]

I think a t1linux project would be a good idea solely to document stuff (like audio now *mostly* working with davidjo's driver on touchbar macs), not sure if we'd be able to pull in support like t2linux unfortunately :l, the 2016/2017 macbooks are at worst a year (or two) away from losing any form of official support on both proprietary operating systems. (Monterey/Windows 10 in mid 2025, Ventura in 2026), and if you have a touchbar macbook you're basically damned to a far more hassled experience on Linux (moving aside the touchbar, the bcm43602 STILL still half functions with the .txt file, and the .txt file isn't even read on Fedora/Fedora-based distros for some reason, and drowning in suspend or having everything break in said resuming with the "fix" is still a major issue on these laptops regardless of touchbar or not). If we want these laptops to remain usable/up-to-date within the next 3 years, we'd probably have to figure out something soon, so I think starting such a project now wouldn't be a bad idea.

I wish I had the skillset to contribute in a meaningful manner, but I don't see myself having this laptop (MBP2016/13in+TB) around for much longer, especially as my butterfly keyboard deteriorates further and further.

cctfubeet, 2024-08-15:
On my 14,2 Macbook Pro the touchbar wasn't lighting up either on the latest kernels, so i had to use stock Fedora 38 for some time, but it suddenly started working on newer kernels and other distros (Void and Artix). The only remarkable change i made on my system was installing unsupported macOS Sonoma through OpenCore Legacy Patcher. Could it be firmware related?

stefand, 2024-08-16:
On which Linux kernel version(s) is it working? I tested an unmodified 6.9 and it still needs the revert. I am running 6.10 now, but with e04955db reverted. I didn't (yet) test 6.10 without this change.

It is possible that a firmware update improved something, but I find it unlikely, though not impossible, that an officially unsupported OSX version would bring a firmware update for this hardware.

Also even with "bad" kernel versions the touch bar started up randomly. My gut feeling says it works 10% of boots, so if you booted 3 or 4 times a lucky streak is certainly possible.

cctfubeet, 2024-08-18:
I'm running latest stock kernel on Void Linux (6.6.46_1).
For testing i made a fresh install, updated it, and installed the driver without early module loading, which made it not light up, but setting it up on dracut made it work again.
I also tested it in Artix with kernel 6.4 with the same results. 
Cannot remember the exact kernel versions in which it worked, but stock Fedora 38 and Fedora 40 also worked.
Seems to work every time i boot, as i haven't experienced the touch bar not lighting up except for my past experiences.

stefand, 2024-08-22:
Fwiw it is still broken for me on 6.10. I have to revert the mentioned commit.

cctfubeet, 2024-08-22:
Tried it again using Gentoo's kernel binary version 6.6.47 and the issue came back again, could only make it work once.

stefand, 2024-08-22:
So that rules out that MacOS sonoma did anything?

This might be a race condition. I am not familiar with Dracut, but I think what it changes for this particular case is that it puts the touch bar driver into the initrd. In my case the apple-ibridge and apple-ib-tb modules are *not* in the initrd and are loaded in the boot process when udev (or rather systemd-udev) loads a bunch of drivers. That is likely to change the module load order and might change the order in which USB devices are initialized.

skbtwiz, 2024-09-25:
How hard would it be to just write a bunch of dkms modules that take care of the wifi .txt fixes as well as reverting the commit for the touchbar? In my humble opinion the largest barrier to entry for working on linux on T1 macbooks is how everyone has to first read through this entire thread, as well as the wifi thread etc. Roadrunner's informational page is long outdated, and there is no updated, well documented source of all the latest information for getting linux working on T1 macbooks. I have a MBP 13,2 sitting unused ready for active development. If there's enough interest I wouldn't be opposed to starting the repo myself. We can focus on first tackling this barrier to entry, and then hopefully we will see more interested people. T1 macbooks are near death in terms of official OS's, now is the time to get the linux effort going. I will most definitely need help though.

foice, 2024-09-29:
> With Linux 6.1.76 the touch bar is working with @marc-git 's driver. With 6.4 and 6.6 it doesn't work most of the time. So I have a baseline for bisecting.
>
> A problem is that in "bad" kernels it is [...]

I am happy to read that it works for you on 6.1, but I am not able to compile it on 6.1.0-25 from Debian 12 fresh install.

```
DKMS make.log for applespi-0.1 for kernel 6.1.0-25-amd64 (x86_64)
dom 29 set 2024, 15:56:36, CEST
make -C /lib/modules/6.1.0-25-amd64/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-25-amd64'
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c: In function ‘appletb_fill_report_info’:
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c:949:9: warning: enumeration value ‘HID_REPORT_TYPES’ not handled in switch [-Wswitch]
  949 |         switch (field->report->type) {
      |         ^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:905:35: error: initialization of ‘int (*)(struct acpi_device *)’ from incompatible pointer type ‘void (*)(struct acpi_device *)’ [-Werror=incompatible-pointer-types]
  905 |                 .remove         = appleib_remove,
      |                                   ^~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:905:35: note: (near initialization for ‘appleib_driver.ops.remove’)
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-25-common/scripts/Makefile.build:255: /var/lib/dkms/applespi/0.1/build/apple-ibridge.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.1.0-25-common/Makefile:2034: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-25-amd64'
make: *** [Makefile:16: all] Error 2
```
any clue what is this compilation error? I have experience on kernel module compilation, but I made sure I have the linux-header package at least.

foice, 2024-09-29:
> The build error with .remove is trivial to fix, but for me the touch bar doesn't work with 6.6 regardless. The driver detects it, registers the input device, but the touch bar just stays black. It [...]

Can you post how to fix that trivial error with the 
```
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:905:35: error: initialization of ‘int (*)(struct acpi_device *)’ from incompatible pointer type ‘void (*)(struct acpi_device *)’ [-Werror=incompatible-pointer-types]
  905 |                 .remove         = appleib_remove,
      |                                   ^~~~~~~~~~~~~~
/
```
It is not trivial at all here ;)

almas, 2024-09-29:
@foice 
Try:
https://github.com/almas/macbook12-spi-driver/tree/touchbar-driver-hid-driver

foice, 2024-09-29:
Thanks for the tip. Tried but no success 
````
 DKMS make.log for applespi-0.1 for kernel 6.1.0-25-amd64 (x86_64)
dom 29 set 2024, 16:57:25, CEST
make -C /lib/modules/6.1.0-25-amd64/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.0-25-amd64'
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c: In function ‘appletb_fill_report_info’:
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c:949:9: warning: enumeration value ‘HID_REPORT_TYPES’ not handled in switch [-Wswitch]
  949 |         switch (field->report->type) {
      |         ^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c: In function ‘appleals_config_iio’:
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:36: warning: passing argument 1 of ‘iio_device_alloc’ makes pointer from integer without a cast [-Wint-conversion]
  463 |         iio_dev = iio_device_alloc(sizeof(als_dev));
      |                                    ^~~~~~~~~~~~~~~
      |                                    |
      |                                    long unsigned int
In file included from /usr/src/linux-headers-6.1.0-25-common/include/linux/iio/buffer.h:10,
                 from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:32:
/usr/src/linux-headers-6.1.0-25-common/include/linux/iio/iio.h:724:49: note: expected ‘struct device *’ but argument is of type ‘long unsigned int’
  724 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                                  ~~~~~~~~~~~~~~~^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:19: error: too few arguments to function ‘iio_device_alloc’
  463 |         iio_dev = iio_device_alloc(sizeof(als_dev));
      |                   ^~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.1.0-25-common/include/linux/iio/iio.h:724:17: note: declared here
  724 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                 ^~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:36:
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:485:72: error: ‘struct iio_dev’ has no member named ‘id’
  485 |         iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
      |                                                                        ^~
/usr/src/linux-headers-6.1.0-25-common/include/linux/iio/trigger.h:166:61: note: in definition of macro ‘iio_trigger_alloc’
  166 |         __iio_trigger_alloc((parent), THIS_MODULE, (fmt), ##__VA_ARGS__)
      |                                                             ^~~~~~~~~~~
/usr/src/linux-headers-6.1.0-25-common/include/linux/iio/trigger.h:166:29: error: passing argument 1 of ‘__iio_trigger_alloc’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  166 |         __iio_trigger_alloc((parent), THIS_MODULE, (fmt), ##__VA_ARGS__)
      |                             ^~~~~~~~
      |                             |
      |                             char *
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:485:20: note: in expansion of macro ‘iio_trigger_alloc’
  485 |         iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
      |                    ^~~~~~~~~~~~~~~~~
/usr/src/linux-headers-6.1.0-25-common/include/linux/iio/trigger.h:169:56: note: expected ‘struct device *’ but argument is of type ‘char *’
  169 | struct iio_trigger *__iio_trigger_alloc(struct device *parent,
      |                                         ~~~~~~~~~~~~~~~^~~~~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-25-common/scripts/Makefile.build:255: /var/lib/dkms/applespi/0.1/build/apple-ib-als.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_setup_read_txfrs’:
/var/lib/dkms/applespi/0.1/build/applespi.c:590:13: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  590 |         dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
      |             ^~
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_setup_write_txfrs’:
/var/lib/dkms/applespi/0.1/build/applespi.c:619:13: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  619 |         wt_t->delay_usecs = SPI_RW_CHG_DELAY_US;
      |             ^~
/var/lib/dkms/applespi/0.1/build/applespi.c:622:13: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  622 |         dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
      |             ^~
/var/lib/dkms/applespi/0.1/build/applespi.c:626:13: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  626 |         wr_t->delay_usecs = SPI_RW_CHG_DELAY_US;
      |             ^~
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_get_saved_bl_level’:
/var/lib/dkms/applespi/0.1/build/applespi.c:1794:38: error: invalid application of ‘sizeof’ to incomplete type ‘struct efivar_entry’
 1794 |         efivar_entry = kmalloc(sizeof(*efivar_entry), GFP_KERNEL);
      |                                      ^
[... 44 lines cut ...]
 1798 |         memcpy(efivar_entry->var.VariableName, EFI_BL_LEVEL_NAME,
      |         ^~~~~~
[... cut ...]
/var/lib/dkms/applespi/0.1/build/applespi.c:2228:27: note: (near initialization for ‘applespi_driver.remove’)
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.1.0-25-common/scripts/Makefile.build:255: /var/lib/dkms/applespi/0.1/build/applespi.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.1.0-25-common/Makefile:2034: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.0-25-amd64'
make: *** [Makefile:16: all] Error 2

````

stefand, 2024-09-29:
foice, when I called the fix "trivial" I meant it for someone who is proficient in the C programming language. If you don't know the language it isn't easy of course.

I don't know why Almas' code fails for you, but I can try to describe what you need to change to fix your original error:

in apple-ibridge.c change
`static int appleib_remove(struct acpi_device *acpi)
to
`static void appleib_remove(struct acpi_device *acpi)`

i.e., change the "int" to "void".

Then at the end of the function change
`return 0;`
to
`return;`

https://github.com/almas/macbook12-spi-driver/commit/3355ca7a9e3efb48660edd5624be010adc0e871b#diff-984f3aa56499b4f3f2430ce4ff6da90aa14c970c7cf0acbc11790f566f619926 shows a diff that you can try to download and apply instead of hand editing.

foice, 2024-10-12:
doing the opposite change of what you wrote did the trick, i.e. it works if I use the function as int. Then I did `modprobe intel_lpss_pci spi_pxa2xx_platform applespi apple-ib-tb`, it gave no errors, and rebooted. Now the touchbar works.

marc-git, 2024-10-17:
> doing the opposite change of what you wrote did the trick, i.e. it works if I use the function as int. Then I did `modprobe intel_lpss_pci spi_pxa2xx_platform applespi apple-ib-tb`, it gave no errors, and rebooted. Now the touchbar works.

You're using Kernel 6.1.xx right? I think that is no longer relevant to the discussion here. We are looking at changes after the kernel modification that stefand rolls back.

woodsy900, 2024-12-28:
I am guessing no progress has been made on this. Just got my hands on a Macbook Pro late 2016  13,3 trying to run Nobara which uses kernal version 6.11.9-200.fsync.fc40.x86_64 (64-bit) its fedora based. 

I have some more work to do with the touchpad it works but has no palm rejection and stuff and the audio is working now but the wifi is trash but thats a known and almost unfixable issue. I would love to use the touchbar. The sucky thing is my skills are almost none when it comes to dev. I can provide a device for testing but thats about it.