Dunedan/mbp-2016-linux #196: WIFI on MBP 13,3

janithcooray, 2023-09-23:
MBP 13,3 - macOS Monetary 

arch / Linux 6.5.4

Update , WIFI is working although only for 2.4Ghz connection. sometimes connections fail but overall it is usable, 
I'm Seeing a bit of hope to get this working 1 way or another (at least a script to make it connect successfully).

According to some forums ,  they have got `Broadcom Corporation BCM4360` working, the one in the MBP 13,3 is 
`Broadcom Corporation BCM43602`

https://gist.github.com/torresashjian/e97d954c7f1554b6a017f07d69a66374

I'm opening this as an issue so if anyone has tried get it working or has some related findings. i will try to get something of working in the meantime.

janithcooray, 2023-09-28:
Update

most of drivers out there don't get compiled for linux 6.5 due to some header struct changes, i have started refactoring code.

will update once i get it running

gjvanderheiden, 2023-11-15:
I have a 14,3. Check this [this link](https://gist.github.com/rob-hills/9134b7352ee7471c4d4f4fbd6454c4b9)

Just the section on the file you have to put in usr/lib/firmware/brcm/brcmfmac43602-pcie.txt

I changed the mac adres to my card's mac adres (from macos) and works fine.

snoweuph, 2024-07-06:
newest Fedora KDE Spin on Macbook Pro 2017 13,3 Touchbar

Wifi Works out of the Box, after the Initial System update
for getting Ethernet running with my USB-C Hub, which worked before the initial update, I had to just add [this Udev rules](https://github.com/wget/realtek-r8152-linux/blob/master/50-usb-realtek-net.rules)