Dunedan/mbp-2016-linux #200: Touchbar Issue: PatrickVerner/macbook12-spi-driver Repo NOT FOUND

davidfuzju, 2024-07-10:
any one help offer PatrickVerner/macbook12-spi-driver source code ?

radbirb, 2024-07-24:
Use https://github.com/marc-git/macbook12-spi-driver, mbp-2016-linux is pretty outdated unfortunately, though do note that even marc-git's repo may not work in recent kernels, see #189 

(do note that I also believe some distro's versions of `usbmuxd` break the driver, you can work around that (https://github.com/Dunedan/mbp-2016-linux/issues/189#issuecomment-1925886611 / https://github.com/libimobiledevice/usbmuxd/issues/138) or use a git version of `usbmuxd`)

foice, 2024-10-12:
I used the module from [marc-git](https://github.com/marc-git/macbook12-spi-driver) following the instructions there and finishing it up with `modprobe intel_lpss_pci spi_pxa2xx_platform applespi apple-ib-tb`. After reboot it worked on my Debian 12 running `6.1.0-25-amd64` . Please note that  in `apple-ibridge.c` you need to change

```
static void appleib_remove(struct acpi_device *acpi)
```
to
```
static int appleib_remove(struct acpi_device *acpi)
```

and at the end of the function change
`return;`
to
`return 0;`  to comply with the function now being a `int` type.