Dunedan/mbp-2016-linux #202: Keyboard now generally not working on 14,3 with 6.12.5

marc-git, 2024-12-20:
I hadn't updated for a while and now 6.12.5 has neither the keyboard nor the touchpad (the touchbar either) working. Seemingly major regression. Anybody had a look into this?

risen, 2025-01-09:
Yes, I had the same problem on Debian's linux-image-6.12.x-amd64 packages.

For now I've worked around it with `iommu.passthrough=1` on the kernel command line. I haven't had time yet to narrow it down and file a proper bug report on the kernel bugzilla.

Are you using a vanilla kernel, or a distribution-provided kernel package?

marc-git, 2025-01-09:
I'm using debians kernel. where did you hear about iommu.passthrough? I'll give it a go

risen, 2025-01-10:
Reading the kernel log, I managed to infer something was going on with DMA remapping. Switching it (intel_iommu) off completely worked, but broke some other things, so I started narrowing it down to what seemed to be the least invasive option.

marc-git, 2025-01-16:
yeah ok. It works for me too. I wonder which change did this. Do you know which kernel version started it?

risen, 2025-01-16:
I'm not sure whether or not it's Debian-specific, but the 6.11.x packages were all working for me. None of the 6.12.x packages were OK.

Heratiki, 2025-01-29:
Fedora 41 6.12.10-200 and it's no longer working for me either.  I'm also unable to build it at all due to errors.  Because I didn't get it built before 6.12 I don't believe the iommu.passthrough will work for me.

Heratiki, 2025-01-30:
@risen I'll give that a shot later today after work and let you know.  I think it's possible that could work.  I also realized I could roll back to 6.11 but with so many other issues present I might just switch to Manjaro for the time being.  Trying to do things with a disappearing cursor is not all that fun!

I was able to build on Ubuntu Budgie using this - https://github.com/almas/macbook12-spi-driver/pull/1 so no clue if that has anything to do with it.

AdityaGarg8, 2025-05-07:
Is the issue with 6.12.4 as well?

joberkel, 2025-05-07:
FYI: bug report concerning this regression was filed: [https://lore.kernel.org/linux-input/PN3PR01MB9597D8E327CC7910673849D3B888A@PN3PR01MB9597.INDPRD01.PROD.OUTLOOK.COM/T/#t](https://lore.kernel.org/linux-input/PN3PR01MB9597D8E327CC7910673849D3B888A@PN3PR01MB9597.INDPRD01.PROD.OUTLOOK.COM/T/#t)

AdityaGarg8, 2025-05-18:
https://lore.kernel.org/lkml/20250514060523.2862195-1-baolu.lu@linux.intel.com/

This patch should fix this.

bbbrandyn, 2025-05-20:
Not working for me as well on 14,2 on Arch 6.14.4 even with `iommu.passthrough=1`