Dunedan/mbp-2016-linux #203: [Debian] Bluetooth doesn't wotk on 14,1

DoYouEvenSheesh, 2025-04-04:
Running Debian bookworm (Kernel Version: 6.1). Tried to switching to Debian unstable with kernel version 6.12.20 still no luck. Arch Linux had no problems with Bluetooth before is some sort of firmware not being loaded?

DoYouEvenSheesh, 2025-04-04:
systemctl log of bluetooth.service is this:

```
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support bap plugin
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support bass plugin
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support mcp plugin
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support vcp plugin
Apr 04 13:30:59 debian bluetoothd[735]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support micp plugin
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support ccp plugin
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support csip plugin
Apr 04 13:30:59 debian bluetoothd[735]: src/plugin.c:init_plugin() System does not support asha plugin
Apr 04 13:30:59 debian bluetoothd[735]: Bluetooth management interface 1.23 initialized
```

risen, 2025-04-04:
I used to use a workaround patch for the vanilla and debian kernels, which made it work but seemed like a hack - which is probably why it was never upstreamed. I haven't rebuilt a kernel in a while, so can't say for sure if it still works.

Which Arch kernel version did you use specifically? Curious if they carry a better patch.

risen, 2025-04-04:
This is the patch I used to use, btw:

```diff
diff --git a/drivers/bluetooth/hci_bcm.c b/drivers/bluetooth/hci_bcm.c
index 89d4c2224546f..c9ccddd7b66bf 100644
--- a/drivers/bluetooth/hci_bcm.c
+++ b/drivers/bluetooth/hci_bcm.c
@@ -281,10 +281,6 @@ static int bcm_gpio_set_power(struct bcm_device *dev, bool powered)
 	if (err)
 		goto err_txco_clk_disable;
 
-	err = dev->set_device_wakeup(dev, powered);
-	if (err)
-		goto err_revert_shutdown;
-
 	if (!powered && dev->res_enabled) {
 		clk_disable_unprepare(dev->txco_clk);
 		clk_disable_unprepare(dev->lpo_clk);
@@ -304,8 +300,6 @@ static int bcm_gpio_set_power(struct bcm_device *dev, bool powered)
 
 	return 0;
 
-err_revert_shutdown:
-	dev->set_shutdown(dev, !powered);
 err_txco_clk_disable:
 	if (powered && !dev->res_enabled)
 		clk_disable_unprepare(dev->txco_clk);
```

DoYouEvenSheesh, 2025-04-04:
> Which Arch kernel version did you use specifically? Curious if they carry a better patch.

I think it was 6.13.x Don't know the exact version I stopped using arch a few weeks ago.

DoYouEvenSheesh, 2025-04-04:
> This is the patch I used to use, btw:
>
> ```diff
> diff --git a/drivers/bluetooth/hci_bcm.c b/drivers/bluetooth/hci_bcm.c
> index 89d4c2224546f..c9ccddd7b66bf 100644
> --- a/drivers/bluetooth/hci_bcm.c
> +++ [...]

Interesting. I'll try it out.

DoYouEvenSheesh, 2025-04-07:
Fixed it with [leifliddy/macbook12-bluetooth-driver](https://github.com/leifliddy/macbook12-bluetooth-driver)